blueman-project/blueman #164: Close notification after PIN code pairing is completed

cschramm, 2015-01-05:
Follow-up to #120, @patrakov. The notification never closes as BlueZ does not call the Cancel method although [it should](https://git.kernel.org/cgit/bluetooth/bluez.git/tree/doc/agent-api.txt?id=793fb9ed655c80b43791b46874e7d4525c17eb92#n96).

[This](https://github.com/blueman-project/blueman/blob/bc58e5dc636a5fd3981727025e58b536f49e0ff0/blueman/gui/DeviceList.py#L176) should be the best place to close any left over pairing notification, but we need to keep track of it up to this point or retrieve it from the agent.

infirit, 2015-12-11:
I am in unfamiliar territory, how would I retrieve the ``BluezAgent``?  Once I get it I can call the Cencel method I guess..

cschramm, 2015-12-12:
I think the `BluezAgent` plugin should handle this itself. It define a handler in `DisplayPinCode` and `DisplayPasskey` like `device.connect_signal('property-changes', self._on_device_property_changed)` and then doing something like this in the handler:

````python
def _on_device_property_changed(self, device, key, value, path):
  if key == 'Paired' and value:
    self.Cancel()
````

However I really wonder why Cancel does not get called by BlueZ, also it's specified for the API...

infirit, 2015-12-12:
Ah, yes I was following your initial suggestions but that makes more sense.

The below works for me with my logitech mx5000 keyboard with the notification bubble. I'll try the dialog as well.

Assuming we work around this in blueman, do you want to keep this issue open?

````diff
diff --git a/blueman/main/applet/BluezAgent.py b/blueman/main/applet/BluezAgent.py
index 4fdf5df..ddc3e40 100644
--- a/blueman/main/applet/BluezAgent.py
+++ b/blueman/main/applet/BluezAgent.py
@@ -145,6 +145,11 @@ class BluezAgent(_GObjectAgent, Agent, GObject.GObject):
         self.dialog.connect("response", passkey_dialog_cb)
         self.dialog.present()
 
+    # Workaround BlueZ not calling the Cancel method, see #164
+    def _on_device_property_changed(self, device, key, value, path):
+        if key == "Paired" and value:
+            self.Cancel()
+
     @AgentMethod
     def Release(self):
         dprint("Agent.Release")
@@ -183,6 +188,8 @@ class BluezAgent(_GObjectAgent, Agent, GObject.GObject):
     @AgentMethod
     def DisplayPasskey(self, device, passkey, entered):
         dprint('DisplayPasskey (%s, %d)' % (device, passkey))
+        dev = Bluez.Device(device)
+        dev.connect_signal("property-changed", self._on_device_property_changed)
         notify_message = _("Pairing passkey for") + " %s: %s" % (self.get_device_alias(device), passkey)
         self.n = Notification("Bluetooth", notify_message, 0,
                               pixbuf=get_icon("blueman", 48), status_icon=self.status_icon)
@@ -190,6 +197,8 @@ class BluezAgent(_GObjectAgent, Agent, GObject.GObject):
     @AgentMethod
     def DisplayPinCode(self, device, pin_code):
         dprint('DisplayPinCode (%s, %s)' % (device, pin_code))
+        dev = Bluez.Device(device)
+        dev.connect_signal("property-changed", self._on_device_property_changed)
         notify_message = _("Pairing PIN code for") + " %s: %s" % (self.get_device_alias(device), pin_code)
         self.n = Notification("Bluetooth", notify_message, 0,
                               pixbuf=get_icon("blueman", 48), status_icon=self.status_icon)
````

cschramm, 2015-12-13:
We can close this then, although I'm really curious why BlueZ seems to not fulfill its contract.

infirit, 2015-12-14:
I pushed a commit that works-around this to master. Does it makes sense to also push it to 2-0-stable? If not we can close this one.

cschramm, 2015-12-14:
I would not really consider it a bug (at least not on our side), so I wouldn't pick it for 2-0-stable.

closed 2015-12-14 by cschramm