blueman-project/blueman #165: no bluetooth audio output option to xfce volume control

oldgranola, 2015-01-11:
No sound option to send audio to connected bluetooth headset (motorola s305)
Blueman apparently pairs, however there does not seem to be any way to send audio to headset either via the bluemon device app or via the xfce-4 indicator volume/mixer applet.
This same headset works in LM17.1 cinnamon, android, windows7

system: linux-mint 17 xfce with kernel 3.16.0-28 with blueman applet

Plaque-fcc, 2015-01-12:
You can try «refresh services» option, in previous versions of blueman
this helped me sometimes.

oldgranola, 2015-01-12:
Here's what I learned so far for an unsatisfactory but working fix. Works better in Cinnamon
Bluemon installed via the applications manager doesn't seem to work as it doesn't activate the bluetooth-discover module and activating via command line fails also.
Fix :
1. Optional purge all bluetooth
sudo apt-get purge pulseaudio-module-bluetooth bluetooth bluez-* bluez
2. reinstall bluemon and pulseaudio module for bt
sudo apt-get install blueman bluez pulseaudio-module-bluetooth
3. Load missing module
pactl load-module module-bluetooth-discover
4. open bluemon applet and "set up new device", allow to pair.
5. open bluemon aplplet >devices. in menue >devices, choose to connect to 'audio synce' and 'headset'. Im sure your setup may vary here.
6.also within bluemon applet>devices: menue >devices>profile choose >profile A2DP (or what ever you want)
Then you can play with the sound control applet in sound settings >configuration tab.
Problem is I have to reload 'module-bluetooth-discover' every time I boot and manually configure the divice as above each time I pair. Much more automatic in LM cinnamon using the cinnamon-BT applet. Can this be fixed for bluemon?

Plaque-fcc, 2015-01-12:
Then you have to add a line into PulseAudio config to enable it loading
the module for you on each start.

On my system here's all about bluetooth:

$ grep bluetooth /etc/pulse/default.pa 
.ifexists module-bluetooth-policy.so
load-module module-bluetooth-policy
.ifexists module-bluetooth-discover.so
load-module module-bluetooth-discover

Hope it helps.

oldgranola, 2015-01-13:
Hi Plaque FCC, Thanks.
I see the same lines in my 'default.pa' with both statements ended with
'.endif 'yet the needed module doesn't load automatically. So either those
lines aren't read/executed or are not sufficient. I still must manually
start by entering $ pactl load-module module-bluetooth-discover before
pairing or the pairing will not include audiosink with A2DP profile. So I
think that is the bug/issue. It is not read or executed in LM17.1 XFCE.

I have a LM17.1 Cinnamon desktop which also has the file default.pa for
pulse audo with those lines yet bluetooth audio sink A2DP works fine so
that makes me wonder if the problem is outside pulseaudio but in blueman.
Cinnamon uses cinnamon-bluetooth applet for managing bluetooth connections
which can be loaded from the software manager. I tried to load it for xfce
to test it out but didn't know how to access it. I looked for other
components of bluez between the two systems, looks like I have most of the
same stuff installed. I am using identical USB-BT dongles for both system.

The other part of this 'bug' that is important to keep in mind is that if I
install blueman from the LM software installer I can't even get the missing
module to load from the command line. Another reason I don't think the
problem lies with pulse. The loading of that module must be a part of
blueman.

Here is another discussion of this bug. I also read similar in a 'mate'
thread. Again, definitely a blueman problem. I hope the dev's can fix it.
https://bugs.launchpad.net/ubuntu/+source/pulseaudio/+bug/1274613

Thanks again, cheers.

Plaque-fcc, 2015-01-14:
I ran into another problem when switched to using blueman: it didn't
connect to audiosink although GNOME bluetooth applet (which they
removed, so I had to switch feeling lots of hate about new weird
«indicators»), and found that building blueman from current git master
branch removes this problem since the issue has been fixed (long ago).

Have you tried this way?

cschramm, 2015-01-14:
If I'm correct, Linux Mint 17 is based on Ubuntu trusty, which still has a version where blueman explicitly unloads the pulseaudio module. This was removed in the [utopic package](http://packages.ubuntu.com/utopic/blueman).

oldgranola, 2015-01-14:
Christopher Schramm, Yes LM17 is based on Ubuntu Trusty. Is there a way to
take advantage of utopic package fixes in LM xfce? Are you also saying that
the matter is considered fixed/closed because utopic is available?  I am
using kernel 3.16.0-28. In Cinnamon desktop the Cinnamon-bluetooth
controller app/applet workswith pulseaudio. But with XFCE ro MATEu using
blueman, I encounter this problem.

PlaqueFCC, You've gotten beyond me there. Are you saying to recompile a
different version of blueman?

Thanks, cheers

Plaque-fcc, 2015-01-14:
Either try using another version of package or try building it from
source. You can see yourself if it helps or not, although both of our
suggestions are based on the same thing.

If it helps, you should file LM package maintainer a bug report about
that and let them know that the problem has been fixed and the package
is outdated.

cschramm, 2015-01-14:
You can download the utopic deb file from the site and `dpkg --install` it.

The Ubuntu guys [talked about](https://bugs.launchpad.net/ubuntu/+bug/1283003/comments/106) bringing the fix into trusty as well, but either they did not even start that process or it got stuck somewhere. I guess that would be necessary to get it into LM17 as well, if that's even possible.

Anyway, yes, this is a packaging problem at the distribution level and fixed on our side for half a year now.

closed 2015-01-14 by cschramm

cschramm, 2015-01-15:
By the way, out ticket for this was #64.