blueman-project/blueman #167: Icon-name conflicts: settings versus panel applet

ochosi, 2015-01-15:
Hey everyone,

I'm maintaining an icon-theme for Xfce/Xubuntu (amongst others) and noticed that with the transition to the new blueman, the icon names for the settings-app and the panel-applet are identical (blueman, as opposed to the previously in use blueman-tray, if I'm not mistaken).

It'd be great if there could be a separate icon-name for the panel applet in blueman, e.g. "blueman-applet". You could ofc just create a symlink from "blueman" to it in your upstream icons, but it would enable me/us to have a monochrome icon in the panel again like we used to.

Thanks for this really nice piece of software!

cschramm, 2015-01-15:
Thanks for this really nice piece of feedback! :wink: 

@Teknocrat Do you want to take on this?

cschramm, 2015-02-05:
No @Teknocrat far and wide. :cry: 

I've created a PR. Does it fit your requirements?

ochosi, 2015-02-05:
Yup, looks good to me! Thanks a bunch!

commit 278b5563 (2015-02-07): Use name blueman-tray for tray icon

Closes #167
closed 2015-02-07 by cschramm