blueman-project/blueman #169: Playstation Move Motion Controllers Wont Stay Connected

jeottav, 2015-01-26:


cschramm, 2015-01-26:
https://github.com/blueman-project/blueman/wiki/Troubleshooting

jeottav, 2015-01-26:
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/blueman/plugins/applet/SerialManager.py", line 70, in on_device_property_changed
    d = Device(path)
  File "/usr/lib/python2.7/dist-packages/blueman/main/Device.py", line 39, in __init__
    self.Properties = self.Device.GetProperties()
  File "/usr/lib/python2.7/dist-packages/blueman/bluez/utils.py", line 28, in warp
    raise errors.parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetProperties" with signature "" on interface "org.bluez.Device" doesn't exist

cschramm, 2015-01-31:
That exception is out of context and most probably not the cause of your problem. Please try to produce some meaningful output. You can try the script from #173 to describe, what you're doing and what's happening while producing the log ("Connect controller", "Controller just disconnected", ...).

closed 2015-02-23 by cschramm