blueman-project/blueman #171: blueman-applet: newest git snapshot + dbus-1.8.14 = work no more

alium, 2015-01-30:
Hi,
after update to version 1.8.14 and compiling the latest git snapshot runs blueman-applet no more:

I recompiled all my package pygobject-devel-3.14.0 python-dbus-1.2.0 python-dbus-common-1.2.0  python-gobject-3.14.0 python-pybluez-0.20  python2-dbus-1.2.0  python2-gobject-3.14.0  python2-pybluez-0.20 and blueman, but without effect.

------
[alois@toshiba ~]$ blueman-applet 
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
Stale PID, overwriting
Using GConf config backend
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
['TransferService', 'StatusIcon', 'StandardItems', 'SerialManager', 'RecentConns', 'NMDUNSupport', 'PPPSupport', 'PowerManager', 'Networking', 'Menu', 'KillSwitch', 'Headset', 'NetUsage', 'DiscvManager', 'DBusService', 'NMPANSupport', 'DhcpClient', 'AuthAgent', 'ShowConnected', 'ExitItem', 'AppIndicator'] 
Using GConf config backend
_________
get_interface_version (/usr/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/blueman/plugins/BasePlugin.py", line 49, in _load
    self.on_load(parent)
  File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/DBusService.py", line 35, in on_load
    self.add_dbus_method(self.ServiceProxy, in_signature="sosas", async_callbacks=("ok", "err"))
  File "/usr/lib/python2.7/site-packages/blueman/plugins/AppletPlugin.py", line 60, in add_dbus_method
    self.Applet.DbusSvc.add_method(func, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/blueman/main/DbusService.py", line 41, in add_method
    dec = dbus.service.method(dbus_interface, in_signature, *args, **kwargs)(%(0)s)""" % {"0": func.__name__, "1": a}
  File "<string>", line 5
    ServiceProxy._orig_func = func
                                 ^
IndentationError: unindent does not match any outer indentation level
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
Failed to load DBusService
unindent does not match any outer indentation level (<string>, line 5)

alium, 2015-01-30:
more info:
[alois@toshiba ~]$ blueman-manager 
Loading configuration plugins
Using GConf config backend
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
['Services', 'PulseAudioProfile'] 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:333)
1 
blueman-manager version 1.99.alpha1 starting
_________
pa_context_event (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:333)
2 
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:105)
org.bluez owner changed to  :1.48 
Using GConf config backend
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
_________
get_interface_version (/usr/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
SetAdapter (/usr/lib/python2.7/site-packages/blueman/gui/DeviceList.py:264)
None 
_________
on_adapter_changed (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerToolbar.py:74)
toolbar adapter /org/bluez/hci0 
_________
__init__ (/usr/lib/python2.7/site-packages/blueman/main/Device.py:16)
caching initial properties 
_________
add_device (/usr/lib/python2.7/site-packages/blueman/gui/DeviceList.py:310)
adding new device 
_________
do_cache (/usr/lib/python2.7/site-packages/blueman/gui/DeviceList.py:489)
Caching new device 12:11:21:40:91:7A 
_________
row_update_event (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Fake False 
_________
row_update_event (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Trusted 1 
_________
row_update_event (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Paired 1 
_________
init_services (/usr/lib/python2.7/site-packages/blueman/main/Device.py:73)
Loading services 
_________
Generate (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:213)
610 
_________
pa_context_event (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:333)
3 
_________
pa_context_event (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:333)
4 
_________
on_pa_ready (/usr/lib/python2.7/site-packages/blueman/plugins/manager/PulseAudioProfile.py:24)
connected 
_________
<lambda> (/usr/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:345)
1 
[... 6 lines cut ...]
Generate (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:213)
610 
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
_________
set_op (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:76)
op: regenerating instance <ManagerDeviceMenu object at 0x7fd1acaae0a0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x25f8ab0)> 
_________
set_op (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:76)
op: regenerating instance <ManagerDeviceMenu object at 0x7fd1b28c78c0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x25f8d30)> 
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
ERROR:dbus.connection:Unable to set arguments ('org.bluez.Device1', dbus.ObjectPath('/org/bluez/hci0/dev_12_11_21_40_91_7A'), 'Connect', []) according to signature None: <type 'exceptions.ValueError'>: Unable to guess signature from an empty list
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 176, in on_connect
    reply_handler=success, error_handler=fail, timeout=200)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 137, in __call__
    **keywords)
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 584, in call_async
    message.append(signature=signature, *args)
ValueError: Unable to guess signature from an empty list
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message did not receive a reply (timeout by message bus)

cschramm, 2015-01-30:
I think this issue was present between revisions 1b587f3 (included) and 4797888 (excluded). Are you really using the latest master (4797888)?

alium, 2015-01-31:
i used befor everytime only https://github.com/blueman-project/blueman/archive/blueman-master.zip, but today i download again - this time another package (https://github.com/blueman-project/blueman/archive/4797888d83274186472ac0adc8e00ed44603117b.zip) and this works.

Thank you!

closed 2015-01-31 by alium