blueman-project/blueman #174: /etc/bluetooth/audio.conf is gone

gentoo-root, 2015-02-04:
https://git.kernel.org/cgit/bluetooth/bluez.git/commit/?id=8aee579387454bca9ecaf1b9d77aef79b5275f1f

/etc/bluetooth/audio.conf file is removed in 2013, but blueman still tries to write it and fails.

Steps to reproduce:
1. Start blueman-applet.
2. Right-click on tray icon, choose "Local Services...".
3. Check both checkboxes in Audio tab.
4. Click Apply.
5. Exception throws while trying to parse non-existing /etc/bluetooth/audio.conf:

```
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
Stale PID, overwriting
Using file config backend
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
['TransferService', 'StatusIcon', 'StandardItems', 'SerialManager', 'RecentConns', 'NMDUNSupport', 'PPPSupport', 'PowerManager', 'Networking', 'Menu', 'KillSwitch', 'Headset', 'NetUsage', 'DiscvManager', 'DBusService', 'NMPANSupport', 'DhcpClient', 'AuthAgent', 'ShowConnected', 'ExitItem', 'AppIndicator'] 
Using file config backend
_________
get_interface_version (/usr/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module NMPANSupport
cannot import name GConf 
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using file config backend
_________
update_menuitems (/usr/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:114)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
_________
on_load (/usr/lib/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using file config backend
_________
load_nap_settings (/usr/lib/python2.7/site-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
update_state (/usr/lib/python2.7/site-packages/blueman/main/KillSwitchNG.py:162)
State: True 
_________
on_switch_added (/usr/lib/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:45)
killswitch registered 0 
_________
on_obex_owner_changed (/usr/lib/python2.7/site-packages/blueman/plugins/applet/TransferService.py:70)
obex owner changed: :1.44 
Using file config backend
_________
update_state (/usr/lib/python2.7/site-packages/blueman/main/KillSwitchNG.py:162)
State: True 
_________
on_switch_added (/usr/lib/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:45)
killswitch registered 2 
_________
on_started (/usr/lib/python2.7/site-packages/blueman/main/applet/Transfer.py:72)
opp Started 
_________
on_started (/usr/lib/python2.7/site-packages/blueman/main/applet/Transfer.py:72)
ftp Started 
_________
on_dbus_name_owner_change (/usr/bin/blueman-applet:86)
org.bluez owner changed to :1.2 
_________
set_nap (/usr/lib/python2.7/site-packages/blueman/plugins/applet/Networking.py:61)
set nap False 
_________
register_agent (/usr/lib/python2.7/site-packages/blueman/plugins/applet/AuthAgent.py:57)
Registering agent 
_________
initialize (/usr/lib/python2.7/site-packages/blueman/plugins/applet/RecentConns.py:138)
rebuilding menu 
_________
enumerate_connections (/usr/lib/python2.7/site-packages/blueman/plugins/applet/ShowConnected.py:45)
Found 0 existing connections 
Loading configuration plugins
Using file config backend
blueman-services version 1.99.alpha1 starting
_________
load_plugins (/usr/bin/blueman-services:87)
['Audio', 'Network', 'Transfer'] 
_________
set_page (/usr/bin/blueman-services:134)
Set page Audio 
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
_________
on_apply (/usr/lib/python2.7/site-packages/blueman/plugins/services/Audio.py:75)
org.freedesktop.DBus.Python.ConfigParser.NoSectionError: Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "<string>", line 2, in SetBluezConfig
  File "/usr/lib/python2.7/site-packages/blueman/plugins/mechanism/Config.py", line 40, in SetBluezConfig
    c.set(section, key, value)
  File "/usr/lib/python2.7/site-packages/blueman/iniparse/compat.py", line 305, in set
    ConfigParser.set(self, section, option, value)
  File "/usr/lib/python2.7/site-packages/blueman/iniparse/compat.py", line 169, in set
    raise NoSectionError(section)
NoSectionError: No section: dbus.String(u'General')
 
^C
```

As a result, I can't enable audio features in blueman.

cschramm, 2015-02-04:
In [Audio.py](https://github.com/blueman-project/blueman/blob/22e653979ea16313d21f182559721007c36c4734/blueman/plugins/services/Audio.py) blueman is adding / removing values Sink, Source, and Gateway to / from the General section of audio.conf. At a quick look I could not find a way to do this now (starting with 5.8). Do you know?

gentoo-root, 2015-02-04:
I'm sorry, but unfortunately I never worked with bluez directly, so I don't know a way to fix this.

gentoo-root, 2015-02-04:
If you modify audio.conf to setup a2dp audio sink functionality, you may take a look at KDE's bluedevil sources, it has such functionality and does not modify audio.conf.

cschramm, 2015-02-04:
Thanks, I'll have a look.

cschramm, 2015-02-05:
I cannot find such a setting in the bluedevil GUI. Can you point me to that?

gentoo-root, 2015-02-05:
There are two ways of doing it. Of course, you need bluedevil and pulseaudio running. Let's suppose there are two paired and trusted devices - Android phone named "Tru Phone" and my KDE laptop named "ideapad".

The first way is initiating connection from phone.
Go to bluetooth menu on phone and press gear icon near ideapad:
[image]
Tick the checkbox near "Multimedia sound" ("Звук мультимедиа"):
[image]
Connection will be established immediately, pulseaudio loopback appears:
[image]

The second way is initiating connection from laptop.
Go to bluedevil's menu and press "Connect" in device's submenu:
[image]
Connection will be established immediately, pulseaudio loopback also appears.

These two ways are fully equvalent and lead to the same results. Second way is just less intuitive - it's not obvious that Connect button will establish a2dp audio connection.

P.S. Sorry for Russian screenshots, I hope you will understand what's going on anyway (by icons and by items placement). If you won't understand something, I can translate text or try to make English screenshots.

cschramm, 2015-02-05:
The second way should work in blueman as well. The notebook is connecting to an audio service of the phone. It should work independent of the settings in "Local services". The fact that this process is unintuitive is - unfortunately - a design concept of BlueZ 5 (see [this thread](http://ml.mate-desktop.org/pipermail/blueman/2015-January/000000.html) if interested).

I'm not sure what's going on in your first example, specifically if you're connecting to a service provided by the notebook or somehow triggering the notebook to connect to the same service of the phone as in the other process (this would be odd as I don't no a way to do such a "reverse connect"). Do you see the same things on the phone when trying this with blueman?

The most obvious explanation is that BlueZ provides an audio service by default and bluedevil does not do anything there. This is also the way it should work in blueman. The only additional feature blueman has (or had) is re-configuring BlueZ to enable and disable those services. It's just an extra so to say and if they are always enabled or enabled by default the easiest solution is to just remove the "Audio" tab from the "Local settings" dialog. If there is a new way to enable / disable the BlueZ features, the user then just needs to do it directly at the BlueZ level instead.

gentoo-root, 2015-02-05:
I've just tried to test a2dp audio sink feature with blueman on gentoo, and it worked! No configuration needed, so that checkbox in blueman settings is useless now. Before that, I only tried to test this feature on xubuntu 14.04 livecd, and it was very buggy. Maybe it's xubuntu-specific bugs, because on gentoo Advanced Audio Receiver with blueman works like a charm, so the corresponding checkbox in blueman settings dialog should be removed, and blueman should not try to modify audio.conf to enable this feature as it's enabled by default.

There is also second checkbox named "Headset Emulation" in blueman settings dialog, and I was unable to test this feature. When I pair my phone with xubuntu, I get two checkboxes in my phone's settings - one for multimedia sound, and one for in-call sound. And when I pair my phone with gentoo, I only get one checkbox for multimedia sound. I was unable to test Headset Emulation, because xubuntu is very buggy (it even can't connect in Advanced Audio Receiver mode), and my gentoo just don't expose such feature. Could you please tell me what should I turn on in bluez or/and pulseaudio settings so that Headset Emulation on my gentoo appears?

gentoo-root, 2015-02-06:
OK, I've managed to set up both A2DP and HSP/HFP profiles on gentoo using bluez5, ofono, pulseaudio and blueman. "Audio" tab in "Local Services" menu of blueman is no longer needed to get things working - they are working out of box if necessary software is installed. So I think usage of [Audio.py](https://github.com/blueman-project/blueman/blob/22e653979ea16313d21f182559721007c36c4734/blueman/plugins/services/Audio.py) and Audio tab in blueman settings could be safely removed if blueman is targeting new versions of bluez.

cschramm, 2015-02-06:
In my opinion it's also not necessary for older versions of BlueZ. The user can at least modify the audio.conf file himself. So I will just remove that.

cschramm, 2015-02-06:
Created a PR. Nice cleanup. :smirk_cat:

commit 93f4fce3 (2015-02-07): Remove settings for local audio services

audio.conf does not exist in BlueZ 5.8 and higher. The services should work without enabling them specifically. See #174.

Also remove iniparse library, BluezConfig class, and Config mechanism plugin. :)

Closes #174
Closes #175
Closes #180
commit 1ca34a14 (2015-02-11): Remove settings for local audio services

audio.conf does not exist in BlueZ 5.8 and higher. The services should work without enabling them specifically. See #174.

Also remove iniparse library, BluezConfig class, and Config mechanism plugin. :)

Closes #174
Closes #175
Closes #180
closed 2015-02-11 by cschramm