blueman-project/blueman #176: Strange chmod /dev/rfkill

gentoo-root, 2015-02-04:
https://github.com/blueman-project/blueman/blob/master/blueman/plugins/mechanism/RfKill.py#L18

    os.chmod("/dev/rfkill", 0o655)

This chmod sets 655 (rw-r-xr-x) on /dev/rfkill. Why do we need X bit on this device? It looks like something strange.

Also blueman should not touch /dev/rfkill permissions at all. They are managed by udev and system administrator. If system administrator allows users to read or write rfkill state, he will give them permission. If users are not allowed to do it, blueman must not try to get this permission, it should use some fallback. Also bluez D-Bus interface allows to get and set power state of adapters (org.bluez.Adapter1.Powered property).

cschramm, 2015-02-07:
The x permission definitely does not make sense, especially since the function is called if the file is missing the **read** permission flag for others.

The file is used by blueman to monitor the commands for added, removed, and changed switches, which is in terms used by the KillSwitch applet plugin to change the status (icon). This has the effect that bluetooth is disabled if a bluetooth killswitch is on.

I think that's useful functionality, but I also agree that we should not force the permission. Not sure what to do about that. A solution would be to shift the monitoring into blueman-mechanism which has root privileges and transmit the fetched signals through its dbus API, but that's a lot of ugly overhead. Another solution would be to just disable the monitoring in the unusual case that the user cannot read the file.

gentoo-root, 2015-02-07:
>Another solution would be to just disable the monitoring in the unusual case that the user cannot read the file.

This solution might be quite good. On most (especially desktop) systems /dev/rfkill is readable by all users (/lib/udev/rules.d/50-udev-default.rules, line `KERNEL=="rfkill", MODE="0664"`). If somewhere it is not readable, it is decision of local administrator to disable it, so blueman should not try to override this.

Also, we should not check if we have read permission by doing things like this: [KillSwitchNG.py:71](https://github.com/blueman-project/blueman/blob/22e653979ea16313d21f182559721007c36c4734/blueman/main/KillSwitchNG.py#L71). We should just try to open file and fallback if it fails, or use access(2) to check permissions (but it is racy). The reason is that maybe on some hypothetical system /dev/rfkill has permissions 0660 root:rfkill, so users in rfkill group will be able to access file.

cschramm, 2015-02-07:
Drafted a PR for that.

commit 881b255b (2015-02-09): Remove KillSwitchNG

Closes #176
Closes #182
commit dc2ce590 (2015-02-23): Remove KillSwitchNG

Closes #176
Closes #182
closed 2015-02-23 by cschramm