blueman-project/blueman #178: Use gtkinfobar instead of MessageArea

ochosi, 2015-02-05:
I know blueman has been around longer than Gtk3, but it'd be nice to use some of the new widgets (where they make sense), especially when the current widget seems to break with many themes.

For failed attempts at establishing a connection, a "blueman+gui+MessageArea+MessageArea" widget is used. It would be great to just use the appropriate infobar instead.

See the screenshots below (with the Numix gtk theme, same results with Ambiance or Greybird).

[image]
[image]

cschramm, 2015-02-05:
Definitely. PR? :grin:

infirit, 2017-09-11:
Merged the code from @cschramm with some modification. There is one thing to still, see #747. Closing this one, further comments can be done in #747

closed 2017-09-11 by infirit