blueman-project/blueman #183: Headset P311: Connection Failed: No such file or directory

orschiro, 2015-02-08:
Dear all,

Blueman looks very promising for plain window manager environments. Thank you very much for your efforts!

I have the following packages installed on my Arch Linux machine:

```
blueman-git 1.9.9.20150202.4797888-1
bluez 5.28-1
bluez-libs 5.28-1
bluez-utils 5.28-1
gnome-bluetooth 3.14.0-1
python2-pybluez 0.20-2
```

I started the manager and tried to connect to my Headset P311:

[image]

As you can see from the screenshot, the connection fails with `Connection Failed: No such file or directory`.

I ensured that the experimental audio features are enabled:

[image]

Is there something else that I have to take into account when connecting to a Bluetooth headset?

Thankfully,

Robert

orschiro, 2015-02-08:
To add, I just saw on the command line output, that I am not authorised to enable the experimental audio services:

```
on_apply (/usr/lib/python2.7/site-packages/blueman/plugins/services/Audio.py:55)
org.freedesktop.DBus.Python.dbus.exceptions.DBusException: Not authorized 
```

cschramm, 2015-02-08:
The audio settings in "Local services" are just for local audio services provided by BlueZ and do not work with versions 5.8 and later. We're just about to remove this in #180.

In general connecting to a remote audio service should work, but I don't know of any current results with headsets.

Please start blueman-applet from a terminal window and show all the output from trying to connect (start the manager from the applet to get its output as well).

orschiro, 2015-02-08:
@cschramm 

Thank you for your help.Very appreciated! Below the entire debug output:

```
❯ blueman-applet

** (blueman-applet:32275): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-dWt5lrByXM: Connection refused
Loading configuration plugins
blueman-applet version 1.99.alpha1 starting
Stale PID, overwriting
Using GConf config backend
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:54)
['StandardItems', 'RecentConns', 'Networking', 'StatusIcon', 'DiscvManager', 'SerialManager', 'NMDUNSupport', 'KillSwitch', 'AppIndicator', 'PPPSupport', 'NMPANSupport', 'NetUsage', 'DBusService', 'TransferService', 'Menu', 'ShowConnected', 'Headset', 'PowerManager', 'DhcpClient', 'AuthAgent', 'ExitItem'] 
Using GConf config backend
_________
get_interface_version (/usr/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using GConf config backend
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using GConf config backend
_________
load_nap_settings (/usr/lib/python2.7/site-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using GConf config backend
_________
on_load (/usr/lib/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using GConf config backend
_________
update_menuitems (/usr/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:114)
[... 304 lines cut ...]
<Device object at 0x7fd3b81142d0 (blueman+main+Device+Device at 0x188d5a0)> org.bluez.Device1 dbus.Array([], signature=dbus.Signature('s')) 
_________
__del__ (/usr/lib/python2.7/site-packages/blueman/main/Device.py:60)
deleting device /org/bluez/hci0/dev_00_1A_7D_B0_70_69 
_________
Destroy (/usr/lib/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/hci0/dev_00_1A_7D_B0_70_69 
_________
initialize (/usr/lib/python2.7/site-packages/blueman/plugins/applet/RecentConns.py:138)
rebuilding menu 
_________
unset_op (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:90)
op: regenerating instance <ManagerDeviceMenu object at 0x7fc16c0c2f50 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x372a9c0)> 
_________
Generate (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:213)
P311 
_________
unset_op (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:90)
op: regenerating instance <ManagerDeviceMenu object at 0x7fc160d0b4b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x372ac40)> 
_________
fail (/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:120)
fail (DBusException(dbus.String(u'No such file or directory'),),) 
/usr/lib/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 487 was not found when attempting to remove it
  return fn(*args, **kwargs)
```

cschramm, 2015-02-09:
Hmm. Nothing exposing there... What does the right click menu of the device look for you? Do you have

- a generic connect item?
- a connect item for the audio service?
- an item to choose the audio profile?

It may be possible that audio services do not work with BlueZ 5 and current master. You could try the services branch.

orschiro, 2015-02-12:
@cschramm 

I will currently pause investigating the issue due to a lack of time. Please keep this opened. I will report back as soon as I have found a solution to my P311 headset.

orschiro, 2015-02-20:
I switched to elementary OS whose default Bluetooth applet recognises my headset automatically. I will close this issue for now.

closed 2015-02-20 by orschiro

temporary321321, 2015-04-07:
Hi; exactly the same problem is still present in Debian (Jessie). I have two bluetooth audio devices and both have only the generic "Connect" option available; the error message in terminal output is basically the same. The bluetooth-related configuration should be clean from the Debian repository (purged & reinstalled).
Martin

da99, 2015-12-15:
I realized this is a closed issue, but in case anyone else is in a similar situation:
I was getting that "No such file..." error on Arch using the P311 and pulseaudio.
These instructions helped solved it: https://wiki.archlinux.org/index.php/Bluetooth_headset#Headset_via_Bluez5.2FPulseAudio

I did a `reboot` and finally connected to the headphones.