blueman-project/blueman #184: blueman segfaults when trying to pair

juanitotc, 2015-02-18:
bluez-1.0.1, blueman git as of 2015/02/18

blueman will scan and find bluetooth devices without problems, but segfaults when the "pair" icon is clicked.

terminal output below:

cschramm, 2015-02-18:
> terminal output below

Unfortunately not. :wink:

juanitotc, 2015-02-18:
Yeah - I had to switch to windows as this funky interface doesn't like my linux browser:(

$ blueman-manager
Loading configuration plugins
_________
<module> (/usr/local/lib/python2.7/site-packages/blueman/main/Config.py:1)
Skipping plugin Gconf
cannot import name GConf 
Using file config backend
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
['Services', 'PulseAudioProfile'] 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/local/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:333)
1 
_________
pa_context_event (/usr/local/lib/python2.7/site-packages/blueman/main/PulseAudioUtils.py:333)
5 
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:49: Warning: The property GtkSettings:gtk-button-images is deprecated and shouldn't be used anymore. It will be removed in a future version.
  self.b_more.set_image(im)
blueman-manager version 1.99.alpha1 starting
Stale PID, overwriting
_________
on_bluez_name_owner_changed (/usr/local/bin/blueman-manager:108)
org.bluez owner changed to  :1.0 
Using file config backend
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process /usr/local/bin/blueman-applet exited with status 1
_________
get_interface_version (/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 4 
_________
SetAdapter (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:264)
None 
_________
on_adapter_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerToolbar.py:75)
toolbar adapter /org/bluez/7605/hci0 
_________
on_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:148)
adapter propery changed Discovering 1 
_________
on_device_found (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:128)
Device discovered 48:5A:B6:3F:CE:FE 
_________
__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:16)
caching initial properties 
_________
add_device (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:310)
adding new device 
_________
do_cache (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:489)
Caching new device 48:5A:B6:3F:CE:FE 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Fake True 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Trusted 0 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Paired 0 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event RSSI -81 
('opacity', 147)
[... 32 lines cut ...]
Loading services 
_________
init_services (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:73)
Loading services 
_________
init_services (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:73)
Loading services 
_________
init_services (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:73)
Loading services 
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process /usr/local/bin/blueman-applet exited with status 1
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process /usr/local/bin/blueman-applet exited with status 1
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process /usr/local/bin/blueman-applet exited with status 1
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process /usr/local/bin/blueman-applet exited with status 1
ERROR:dbus.proxies:Introspect error on org.blueman.Applet:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process /usr/local/bin/blueman-applet exited with status 1
_________
err (/usr/local/bin/blueman-manager:249)
(DBusException(dbus.String(u'Process /usr/local/bin/blueman-applet exited with status 1'),),) 

(blueman-manager:11339): Gtk-CRITICAL **: gtk_style_context_get_property: assertion 'priv->widget != NULL || priv->widget_path != NULL' failed
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:153: Warning: gtype.c:4221: type id '0' is invalid
  self.show()
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:153: Warning: can't peek value table for type '<invalid>' which is not currently referenced
  self.show()
Segmentation fault

cschramm, 2015-02-18:
Does blueman-applet crash immediately when started from the terminal?

cschramm, 2015-02-18:
Wait,

> bluez-1.0.1

Seriously? This would be from the last millenium and probably not even have a dbus API.

juanitotc, 2015-02-18:
Sorry - bluez-4.101

juanitotc, 2015-02-18:
starting blueman-applet revealed a problem with libnotify that I've now fix.

blueman will pair with a device with an assumed pin 0000, and things work.

However with other devices, the blueman generated random pin does not appear on the remote device and blueman eventually segfaults waiting for something to happen with the same message:

(blueman-manager:11339): Gtk-CRITICAL **: gtk_style_context_get_property: assertion 'priv->widget != NULL || priv->widget_path != NULL' failed
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:153: Warning: gtype.c:4221: type id '0' is invalid
self.show()
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:153: Warning: can't peek value table for type '' which is not currently referenced
self.show()
Segmentation fault

cschramm, 2015-02-18:
Probably your theme is not compatible with your GTK+ 3 version, see #42.

closed 2015-02-18 by cschramm

juanitotc, 2015-02-18:
The gtk3 is already set to Adwaita...