blueman-project/blueman #187: Scripts contain hardcoded path to python

juanitotc, 2015-02-19:
The blueman python scripts installed to bin and libexec are hardcoded to /usr/bin/python.

Suggest that either /usr/bin/env python or the actual location of python is used - /usr/local/bin/python in my case

cschramm, 2015-02-19:
I think changing to /usr/bin/env is reasonable, yes.

rworkman, 2015-02-25:
As expected, this tests out fine locally. Perhaps make it use python2 so long as blueman is not fully python3 ready (or is it now?) though...

juanitotc, 2015-02-25:
It'd be good to have the choice of using python2 or python3

rworkman, 2015-02-25:
It looks pretty trivial to rename the relevant scripts to .in files and use autotools magic to sub in "python2" or "python3" -- but assuming the code does (or will before 2.0) work fine in either 2 or 3, then leaving it as is (/usr/bin/env python) would be my preference, as then the system default is what will be used.

cschramm, 2015-02-25:
Python 3 is not ready yet, but close. We need to get rid of the unicode type and fix some str vs. bytes issues in the cython functions, see https://github.com/blueman-project/blueman/issues/19#issuecomment-73652652.

commit 6be07be7 (2015-02-25): Use /usr/bin/env

Closes #187
Closes #192
PR #192 (2015-02-25): Use /usr/bin/env
Closes #187
Closes #192
closed 2015-02-25 by cschramm