blueman-project/blueman #189: Timing issue pairing with randon pin

juanitotc, 2015-02-19:
Using bluez-5.24

There appears to be a timing issue when trying to connect to an iPhone.

When ititiating pairing from blueman, by the time (a few seconds) the random pin pops up on the iPhone with the message "OK to pair", blueman has stopped waiting and reports a pairing failure.

Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
juanitotc iPhone 
_________
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 39, in on_request_menu_items
    manager_menu.Signals.Handle("bluez", device, manager_menu.service_property_changed, "PropertyChanged")
  File "/usr/local/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
    obj.handle_signal(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
    self._handle_signal(wrapper, 'PropertiesChanged', interface, self.get_object_path(), **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 33, in _handle_signal
    self.get_bus().add_signal_receiver(handler, signal, interface, 'org.bluez', obj_path, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 157, in add_signal_receiver
    watch = self.watch_name_owner(bus_name, callback)
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 373, in watch_name_owner
    return NameOwnerWatch(self, bus_name, callback)
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 80, in __init__
    arg0=bus_name)
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 160, in add_signal_receiver
    self.add_match_string(str(match))
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 398, in add_match_string
    BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.13" is not allowed to add more match rules (increase limits in configuration file if required)

cschramm, 2015-04-13:
@juanitotc @bjrosen Please run the [189 branch](https://github.com/blueman-project/blueman/tree/189) and check the output of blueman-applet (and manager). There should be one line that repeats a lot of times (ten thousands). Can you spot it?

juanitotc, 2015-04-14:
I compiled the 189 branch and tested by searching for and pairing with an iPhone.

Note that (and this happens with the other git branches) after blueman shows my laptop has paired with the iPhone, the iPhone shows "not connected". If I tap on the laptop name on the iPhone, this appears to initiate the "trust" dialogue and, if I click on "always accept" on the laptop pop-up notification, the iPhone shows "connected" and blueman displays the "trust symbol"

There were no repeated lines in blueman-applet/manager, and no errors in blueman-manager.

blueman-applet gave a "Connection Failed: DBusFailedError: No such file or directory" pop-up and the following errors (I believe 127 is missing file):

enumerate_connections (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/ShowConnected.py:50)
Found 0 existing connections 
ERROR:dbus.proxies:Introspect error on org.blueman.Mechanism:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 127
/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/Networking.py:44: Warning: The property GtkWidget:margin-left is deprecated and shouldn't be used anymore. It will be removed in a future version.
  d.expander.props.margin_left = 9
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
_________
...
---------
UpdatePowerState (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:145)
off True 
foff False 
on True 
current state False 
new state False 
ERROR:dbus.proxies:Introspect error on org.blueman.Mechanism:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 127
_________
...
_________
on_adapter_property_changed (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Powered 1 
ERROR:dbus.proxies:Introspect error on org.blueman.Mechanism:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 127
_________
...
_________
__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 56, in wrapper
    handler(name, value, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 34, in on_device_property_changed
    klass = Device(path).get_properties()["Class"] & 0x1fff
  File "/usr/local/lib/python2.7/site-packages/blueman/main/Device.py", line 47, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

_________
...
_________
__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 56, in wrapper
    handler(name, value, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 57, in on_device_property_changed
    d = Device(path)
  File "/usr/local/lib/python2.7/site-packages/blueman/main/Device.py", line 47, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

cschramm, 2015-04-14:
@juanitotc So do you actually still encounter the LimitsExceeded error?

In your last log mechanism cannot be started again. Do you have to replace the shebang again? (#206)

juanitotc, 2015-04-14:
Aaagh - sorry, I forgot to modify blueman-mechanism.

If I understand correctly, /usr/local is used to separate packages that are not part of the core distro  in /usr, so it shouldn't be that uncommon to find perl, python, etc in /usr/local...

After trying again, I only get the two errors below and no "limits exceeded" error.

__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 56, in wrapper
    handler(name, value, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 34, in on_device_property_changed
    klass = Device(path).get_properties()["Class"] & 0x1fff
  File "/usr/local/lib/python2.7/site-packages/blueman/main/Device.py", line 47, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

_________
...
_________
__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
ERROR:dbus.connection:Exception in handler for D-Bus signal:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
    self._handler(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 56, in wrapper
    handler(name, value, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 57, in on_device_property_changed
    d = Device(path)
  File "/usr/local/lib/python2.7/site-packages/blueman/main/Device.py", line 47, in __init__
    self.Properties = self.Device.get_properties()
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 148, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.UnknownObject: Method "GetAll" with signature "s" on interface "org.freedesktop.DBus.Properties" doesn't exist

_________

cschramm, 2015-04-14:
OK, so this issue is not reproducible on you side anymore. Then debugging it does not make sense of course. :wink: 

The two exceptions are just "cosmetic issues" related to the failed pairing.

@bjrosen is the only one left with this issue then... I'll wait for feedback. Hrm... Strange that it's that infrequent...

bjrosen, 2015-04-14:
I rebooted my system and now blueman seems to work.

Thanks,

Josh

cschramm, 2015-04-14:
Ah, it appears on (very) long running systems. That makes sense then. I hope to eliminate those "handler leaks" with #254.

closed 2015-04-14 by cschramm

bjrosen, 2015-04-14:
Chris, I've put a note on the Fedora Bugzilla entry for Blueman letting them know that Blueman is working now.

https://bugzilla.redhat.com/show_bug.cgi?id=1171517

cschramm, 2015-04-14:
Great! Added a reference to that bug to our wiki section on Fedora. :smiley:

monkee-ch, 2016-10-13:
Error still present as of today

infirit, 2016-10-13:
@monkee-ch, which blueman version are you using? If version 2.0 see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging and post the logs of blueman-applet.

monkee-ch, 2016-10-13:
blueman 2.0.4
I was able to connect my Bluetooth Headset via bluetoothctl
bluetoothctl
bluetoothctl pair   HEX:HEX:HEX
bluetoothctl trust  HEX:HEX:HEX
bluetoothctl connect   HEX:HEX:HEX

Now it also works with blueman...

Before i had the exact same Error Message as the Original Asker had:
" DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.xx" is not allowed to add more match rules (increase limits in configuration file if required)"

I'll keep an eye on it and will post logs if error reproduces.