blueman-project/blueman #190: Unable to set headphones as audio sink

juanitotc, 2015-02-19:
Using bluez-5.24

When pairing with headphones with an assumed pin of 0000, the pairing appears to proceed as normal, but the audio icons on the right hand side only remain for a few seconds before disappearing and there are no right click options to select "audio sink", etc.

If I right-click whilst the audio icons are still there immediately after pairing, there is an "audio profile" option, but clicking on it does nothing and the audio icons quickly disappear.

Note that with the same hardware, but bluez-4.101, things work as expected.

Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch UUIDs dbus.Array([dbus.String(u'0000110b-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) /org/bluez/hci0/dev_00_1B_66_00_31_88 () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event UUIDs dbus.Array([dbus.String(u'0000110b-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch Connected 0 /org/bluez/hci0/dev_00_1B_66_00_31_88 () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Connected 0 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating down 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected)

cschramm, 2015-02-19:
We need the corresponding output from blueman-applet to figure out what's going on.

juanitotc, 2015-02-20:
Sennheiser

$ blueman-applet
Loading configuration plugins
_________
<module> (/usr/local/lib/python2.7/site-packages/blueman/main/Config.py:1)
Skipping plugin Gconf
cannot import name GConf 
blueman-applet version 1.99.alpha1 starting
Using file config backend
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
['TransferService', 'StatusIcon', 'StandardItems', 'ShowConnected', 'SerialManager', 'RecentConns', 'PowerManager', 'PPPSupport', 'Networking', 'NetUsage', 'NMPANSupport', 'NMDUNSupport', 'Menu', 'KillSwitch', 'Headset', 'GameControllerWakelock', 'ExitItem', 'DiscvManager', 'DhcpClient', 'DBusService', 'AuthAgent', 'AppIndicator'] 
Using file config backend
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module NMPANSupport
cannot import name GConf 
_________
get_interface_version (/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using file config backend
_________
update_menuitems (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:114)
warning: Adapter is None 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
[... 474 lines cut ...]
Sennheiser PXC310BT 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Sennheiser PXC310BT 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected) 
_________
RemoveDevice (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:420)
<Device object at 0x7f6cb89e6190 (blueman+main+Device+Device at 0x23574e0)> 
_________
__del__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:60)
deleting device /org/bluez/hci0/dev_E4_E0_C5_C3_A5_24 
_________
Destroy (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/hci0/dev_E4_E0_C5_C3_A5_24 
_________
RemoveDevice (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:420)
<Device object at 0x7f6cb89e6870 (blueman+main+Device+Device at 0x2358220)> 
_________
__del__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:60)
deleting device /org/bluez/hci0/dev_48_5A_B6_3F_CE_FE 
_________
Destroy (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/hci0/dev_48_5A_B6_3F_CE_FE

cschramm, 2015-03-02:
I'm not sure what's happening there. There is an audio sink in the services of the device and I cannot see a reason for the corresponding menu entry disappearing - especially as there is no event from pulseaudio. I just merged in some big changes regarding services in BlueZ 5. Can you try the current master to see if there's any difference?

juanitotc, 2015-03-02:
I can now pair with the headphones (at the second attempt) and play music to them - thanks.

However, I am unable to pair with an iPhone at all :(

$ blueman-applet
Loading configuration plugins
_________
<module> (/usr/local/lib/python2.7/site-packages/blueman/main/Config.py:1)
Skipping plugin Gconf
cannot import name GConf 
blueman-applet version 1.99.alpha1 starting
Using file config backend
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
['GameControllerWakelock', 'TransferService', 'StatusIcon', 'StandardItems', 'SerialManager', 'RecentConns', 'NMDUNSupport', 'PPPSupport', 'PowerManager', 'Networking', 'Menu', 'KillSwitch', 'Headset', 'NetUsage', 'DiscvManager', 'DBusService', 'NMPANSupport', 'DhcpClient', 'AuthAgent', 'ShowConnected', 'ExitItem', 'AppIndicator'] 
_________
get_interface_version (/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module NMPANSupport
cannot import name GConf 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using file config backend
_________
update_menuitems (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:114)
warning: Adapter is None 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
[... 654 lines cut ...]
animating up 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
    *message.get_args_list()))
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/Device.py", line 28, in err
    raise exception
blueman.bluez.errors.DBusAuthenticationFailedError: Authentication Failed
_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch Connected 0 /org/bluez/hci0/dev_54_E4_3A_22_48_E9 () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Connected 0 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:166)
juanito_tc iPhone 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating down 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected) 
_________
save (/usr/local/lib/python2.7/site-packages/blueman/plugins/config/File.py:105)
Saving config

cschramm, 2015-03-02:
I see three separate(?) issues in your logs:

````
Traceback (most recent call last):
File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
ret = getattr(inst, function)(args, *kwargs)
File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
add_menu_item(manager_menu, service)
File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 32, in add_menu_item
items.append((item, service.priority + 100))
TypeError: unsupported operand type(s) for +: 'NoneType' and 'int'
````

This may be fixed by cc693e9cb7be8e17d83cbd6ec0fc9fcd891b187e, but according to the log the device does not provide a handsfree service, so it does not make much sense to me.

````
ERROR:dbus.proxies:Introspect error on org.blueman.Mechanism:/: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 127
````

This means the dbus service blueman-mechnism fails which is severe and can easily lead to other issues. There's probably more at /var/log/daemon.log.

````
Traceback (most recent call last):
File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
*message.get_args_list()))
File "/usr/local/lib/python2.7/site-packages/blueman/bluez/Device.py", line 28, in err
raise exception
blueman.bluez.errors.DBusAuthenticationFailedError: Authentication Failed
````

This is where the pairing fails. Unfortunately there is no indication why. Let's try to fix mechanism for you and see if it helps already.

juanitotc, 2015-03-02:
When blueman-manager starts I get this error:

org.freedesktop.DBus.Error.Spawn.ChildExited: Launch helper exited with unknown return code 127

You might not be able to connect to the Bluetooth network via this machine

..which might be the cause of the dbus error?

Before I pair with the headphones, it shows (incorrectly) handsfree, headset and audio sink as options. After I pair, it shows (correctly) only audio sink.

cschramm, 2015-03-02:
Have a look at your daemon log when ChildExited shows up.

juanitotc, 2015-03-02:
I don't have a dbus-daemon log - normally the daemon is started like this:

/usr/local/bin/dbus-daemon --system 2>/var/log/dbus

How do I get a log?

cschramm, 2015-03-02:
Logging is configured in the system bus settings (/etc/dbus-1/system.conf in my case). Normally syslog is used and in my case routes messages from the dbus daemon to /var/log/daemon.log.

juanitotc, 2015-03-02:
Got it - not much help though?

Mar  2 17:40:36 box daemon.info bluetoothd[8453]: Bluetooth daemon 5.24
Mar  2 17:40:36 box daemon.info bluetoothd[8453]: Starting SDP server
Mar  2 17:40:36 box daemon.info bluetoothd[8453]: Bluetooth management interface 1.6 initialized
Mar  2 17:40:36 box daemon.err bluetoothd[8453]: hci0 Load Connection Parameters failed: Unknown Command (0x01)

Mar  2 17:41:30 box daemon.info bluetoothd[8453]: Endpoint registered: sender=:1.12 path=/MediaEndpoint/A2DPSource
Mar  2 17:41:30 box daemon.info bluetoothd[8453]: Endpoint registered: sender=:1.12 path=/MediaEndpoint/A2DPSink
Mar  2 17:41:38 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:38 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:38 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:38 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:39 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:39 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:42 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:42 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
Mar  2 17:41:52 box daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
Mar  2 17:42:46 box daemon.info bluetoothd[8453]: /org/bluez/hci0/dev_00_1B_66_00_31_88/fd0: fd(21) ready
Mar  2 17:42:46 box daemon.err bluetoothd[8453]: Control: Refusing unexpected connect
Mar  2 17:42:47 box daemon.err bluetoothd[8453]: connect error: Connection refused (111)

cschramm, 2015-03-02:
Hmmm... Can you check the Exec path in the service file (/usr/share/dbus-1/system-services/org.blueman.Mechanism.service in my case) and try to execute it (as root)?

juanitotc, 2015-03-02:
$ sudo /usr/local/libexec/blueman-mechanism
Starting blueman-mechanism 
Loading configuration plugins
_________
<module> (/usr/local/lib/python2.7/site-packages/blueman/main/Config.py:1)
Skipping plugin Gconf
cannot import name GConf 
Traceback (most recent call last):
  File "/usr/local/libexec/blueman-mechanism", line 155, in <module>
    conf_service()
  File "/usr/local/libexec/blueman-mechanism", line 128, in __init__
    __import__("blueman.plugins.mechanism.%s" % plugin, None, None, [])
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/mechanism/RfKill.py", line 4, in <module>
    from blueman.plugins.applet.KillSwitch import RFKILL_TYPE_BLUETOOTH, RFKILL_OP_CHANGE_ALL
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py", line 9, in <module                                                       >
    from blueman.plugins.AppletPlugin import AppletPlugin
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/AppletPlugin.py", line 7, in <module>
    from blueman.plugins.ConfigurablePlugin import ConfigurablePlugin
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/ConfigurablePlugin.py", line 1, in <modul                                                       e>
    from blueman.main.Config import Config
  File "/usr/local/lib/python2.7/site-packages/blueman/main/Config.py", line 22, in <module>
    dprint("Skipping plugin %s\n%s" % (plugin, e))
  File "/usr/local/lib/python2.7/site-packages/blueman/Functions.py", line 66, in dprint
    sys.stdout.flush()
AttributeError: 'Tee' object has no attribute 'flush'
Exception AttributeError: "'NoneType' object has no attribute 'stdout'" in <bound method Tee.__del__ of                                                        <__main__.Tee object at 0x7fb38d3192d0>> ignored

cschramm, 2015-03-02:
Looks good so far. I've added a dummy method to avoid the flush crash in https://github.com/blueman-project/blueman/commit/e5ba83403a06ed162a6924921cb61cb2118e7fa9. You can try again with that to see if you get to the actual problem, but I think you're already past the point where the dbus service invocation fails as the first line "Starting blueman-mechanism" should already show up in the syslog, which it does not according to your information.

juanitotc, 2015-03-03:
I applied to two patches above, but it does not seem to change anything.

When I first start blueman-applet, there is a pop-up error message stating that networking might not be available.

When I first start blueman-manager, the is a pop-up asking if I want to enable the bluetooth (only with bluez-5, not with bluez-4).

As mentioned earlier, even with the error messages, I can successfully pair with the headphones and play music through them. With the two patches, the dbus-daemon error messages remain exactly the same.

Note also:

$ sudo /usr/local/libexec/blueman-mechanism
  File "/usr/local/libexec/blueman-mechanism", line 52
    def flush(self):
                   ^
IndentationError: unindent does not match any outer indentation level

cschramm, 2015-03-03:
Do you use the commit or did you add the flush method yourself? Python is picky with indentation. My commit works for me, so you should not get an IndentationError as well if you use it.

juanitotc, 2015-03-03:
I cloned blueman and recompiled - the blueman-mechanism part seems fixed now:

$ sudo /usr/local/lib/blueman/blueman-mechanism
Starting blueman-mechanism 
Loading configuration plugins
_________
<module> (/usr/local/lib/python2.7/site-packages/blueman/main/Config.py:1)
Skipping plugin Gconf
cannot import name GConf 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading RfKill 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading Rfcomm 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading Ppp 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading Network

cschramm, 2015-03-03:
So mechanism runs fine. You seem to have some generic issue with your dbus setup. Do you have policykit? Searching the web for "Launch helper exited with unknown return code 127" you find a couple of different causes and solutions.

juanitotc, 2015-03-03:
No, I'm not using polkit or consolekit - do I need to?

cschramm, 2015-03-03:
You should not need to, but then configure blueman with --disable-polkit.

juanitotc, 2015-03-03:
I had a couple of ideas:

1. Are the groups "audio" and "network" in, for example, Handsfree.py and NetworkAccessPoint.py, the standard linux groups or something else (neither exist on this system)?

2. There is a runtime dependency that is missing - either because blueman looks for it under /usr, but not /usr/local, or because I didn't know it was required. The "Launch helper exited with unknown return code 127" seems to be connected to the blueman network and power plugins?

cschramm, 2015-03-03:
> Are the groups "audio" and "network" in, for example, Handsfree.py and NetworkAccessPoint.py, the standard linux groups or something else (neither exist on this system)?

No, that's internal to blueman.

> There is a runtime dependency that is missing - either because blueman looks for it under /usr, but not /usr/local, or because I didn't know it was required. The "Launch helper exited with unknown return code 127" seems to be connected to the blueman network and power plugins?

I do not even think blueman-mechanism is ever invoked from what you provided. The dbus daemon uses some launch helper to start it and that fails. A couple of search results refer to policykit and while I would expect the invocation to work if you do not have it in your system at all, it may be an issue if you enabled polkit support in your blueman configuration.

juanitotc, 2015-03-03:
I recompiled with "--enable-policykit=no", but it did not change anything (and the policy is still installed) :(

As far as I know, I am not using any launch helper at all - I tried with nothing and with "dbus-launch --sh-syntax --exit-with-session", but the result was the same.

cschramm, 2015-03-03:
Not sure if that option works. At the end of the configuration step you should see "Policykit-1 Enabled:" to verify if it does.

juanitotc, 2015-03-03:
That was my point - I get "Policykit-1 Enabled: no" at the end of ./configure, but "make install" installs the policy anyway...

infirit, 2015-03-03:
>Not sure if that option works.

The option is fine but then goes on to do absolutely nothing for automake.

@juanitotc I [fixed](https://github.com/blueman-project/blueman/commit/1954e2d3ffbc79a0cf696c7fe12bca431d34d368) the build system to conditionally install the policy files.

juanitotc, 2015-03-04:
Since dbus is working on this system with gnome-session, it's hard to imagine that there's much wrong with it...

Is there a way to step through the blueman-applet/manager startup to where the first dbus attempt to start blueman-mechanism fails giving the ""Launch helper exited with unknown return code 127" errror?

cschramm, 2015-03-04:
https://bugs.launchpad.net/ubuntu/+source/dbus/+bug/576438 may be of interest to debug this, especially the hints in the first answer.

cschramm, 2015-03-04:
As your original issue is fixed in current master, let's move the discussion for the mechnism issue to #206. iPhone pairing is in #205.

closed 2015-03-04 by cschramm