blueman-project/blueman #191: blueman times out before remote device responds

juanitotc, 2015-02-20:
Using bluez-5.24

When trying to pair with an iPhone, blueman times out (in a few seconds) before the random pin message appears on the iPhone.

blueman-applet and blueman-manager output below:

iphone

$ blueman-applet
Loading configuration plugins
_________
<module> (/usr/local/lib/python2.7/site-packages/blueman/main/Config.py:1)
Skipping plugin Gconf
cannot import name GConf 
blueman-applet version 1.99.alpha1 starting
Using file config backend
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
['TransferService', 'StatusIcon', 'StandardItems', 'ShowConnected', 'SerialManager', 'RecentConns', 'PowerManager', 'PPPSupport', 'Networking', 'NetUsage', 'NMPANSupport', 'NMDUNSupport', 'Menu', 'KillSwitch', 'Headset', 'GameControllerWakelock', 'ExitItem', 'DiscvManager', 'DhcpClient', 'DBusService', 'AuthAgent', 'AppIndicator'] 
Using file config backend
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module NMPANSupport
cannot import name GConf 
_________
get_interface_version (/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
Load (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using file config backend
_________
update_menuitems (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:114)
warning: Adapter is None 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
[... 491 lines cut ...]
juanito_tc iPhone 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected) 
_________
RemoveDevice (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:420)
<Device object at 0x7f89ff095190 (blueman+main+Device+Device at 0x10c8da0)> 
_________
Destroy (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/hci0/dev_48_5A_B6_3F_CE_FE 
_________
RemoveDevice (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:420)
<Device object at 0x7f89ff0953c0 (blueman+main+Device+Device at 0x10ea7a0)> 
_________
__del__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:60)
deleting device /org/bluez/hci0/dev_E4_E0_C5_C3_A5_24 
_________
Destroy (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/hci0/dev_E4_E0_C5_C3_A5_24 
_________
init_services (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:73)
Loading services 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
juanito_tc iPhone

cschramm, 2015-02-20:
That's #189 again, isn't it?

juanitotc, 2015-02-20:
Sorry - I hadn't realised the subject had been changed and forgot I posted it.

Please delete.

closed 2015-02-20 by cschramm