blueman-project/blueman #193: No action while rejecting a file tranfer on the other end

phrackmod, 2015-02-21:
Currently using these versions --
blueman-git 1.9.9.20150208.278b556-1
bluez 5.28-1
Arch with Cinnamon

During the transfer of a file from PC to Mobile, I've rejected the request on my Mobile and the file transfer window still stays open on my PC without autoclosing.

rworkman, 2015-02-25:
I can reproduce this on bluez 4.101 as well, for what it's worth.

infirit, 2015-04-30:
With bluez5 when I use ``dbus-monitor`` on ``org.bluez.obex.ObjectPush1`` I never see anything like an error or message that the transfer was rejected. I do see the ``SendFile`` action with the path to the file but then all is silent.

Am looking on the wrong interface?

cschramm, 2015-04-30:
I think the Status property of the Transfer1 interface should change to error.

infirit, 2015-04-30:
Yes, I see it when I was looking on the sender org.bluez.obex. Then looking at the code we should then handle it in on_transfer_properties_changed. And if I understand the code correctly we don't handle it, only the transfer progress and complete.

infirit, 2015-04-30:
Yep, when I plug in something wrong in that function I see errors \o/. Hopefully back soon with a patch.

infirit, 2015-04-30:
Ok, initial patch only tested with Bluez5 but it should also take care of the error with Bluez4, https://github.com/infirit/blueman/commit/a11b268b55cd94b9ea879255a5922920baa29d5b.

@cschramm, ``on_session_error`` in blueman-sendto takes a name and a message. With Bluez5 there is nothing really useful in the message we get from dbus. However I do not have any example of how the dialog looked like before so just filled it with key (Status) and the value (error) which does not look all that nice.

cschramm, 2015-05-01:
Looks like I already fixed that in the obexd branch: https://github.com/blueman-project/blueman/blob/d67735507b2ec2e7be6d5cadca99261c0456bf96/blueman/bluez/obex/Transfer.py

In the meantime we could either take your fix or take the "Refactor blueman.bluez.obex" commit from that branch ("Fix too generic exception clause when obexd client cannot be initialized" is also fine for master).

I'm sorry for not taking a look at that earlier and letting you do the same work again, so please decide what you want to do, @infirit. :wink: 

Well, at a second thought... My commit lacks the python3 fixes, so we would first have to apply the matching parts of the "Fix python3 compatibility" commit on top.

infirit, 2015-05-01:
I saw a bug that was simple enough for me to fix and allow me to get more acquainted with the bluez/obex code. I understand more now so no worries :smile:.

Let's keep it in the obexd branch and merge the whole thing at once.

PR #272 (2015-05-01): Obexd client refactoring
Fixes #193 and builds a base for #263
closed 2015-05-01 by cschramm