blueman-project/blueman #195: Blueman gives the vague error "Connection Failed: Unable to get a session"

robbie0630, 2015-02-23:
I have been trying to connect my bluetooth speaker to my computer for at least a couple days. I finally decided to just ask for help on the project itself. I really want to set this up with my computer. Help would be appreciated.

cschramm, 2015-02-23:
Looks like an error from BlueZ 4. Is the device paired?

Please give more information on what you're using: https://github.com/blueman-project/blueman/wiki/Troubleshooting

robbie0630, 2015-02-23:
@cschramm This is the error I get when I tried to pair the device.

cschramm, 2015-03-02:
?

robbie0630, 2015-03-04:
@cschramm "Connection Failed: Unable to get a session" is the error I get when I tried to pair the device.

cschramm, 2015-03-04:
Sorry, we cannot help if you do not follow simple requests and answer simple questions.

closed 2015-03-04 by cschramm