blueman-project/blueman #197: Tray icon shows inactive after resume from suspend

rworkman, 2015-02-24:
In case it's not clear, it's the icon with the little red "x" in the corner. This shows up after waking from sleep, even though bluetooth is clearly active (because I'm still using the mouse) :-)

If I click the icon, it prompts with the message about "bluetooth needs to be turned on" and will successfully launch the applet if I click "Enable Bluetooth" while "Exit" does exactly that.

If I kill the applet and restart it, regardless of whether I tell it to Enable Bluetooth, the correct icon shows up and everything works fine. For the sake of clarity, everything seems to work regardless of the icon being displayed.

Here's what I see in the terminal output during resume:
```
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:62)
killswitch changed 0 
_________
io_event (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py:62)
State: False 
_________
UpdatePowerState (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:140)
off True 
foff False 
on True 
current state True 
new state False 
_________
UpdatePowerState (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:140)
Signalling False 
Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/gi/overrides/GLib.py", line 702, in <lambda>
    func_fdtransform = lambda _, cond, *data: callback(channel, cond, *data)
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py", line 52, in <lambda>
    lambda *args: ref() and ref().io_event(*args))
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/KillSwitch.py", line 97, in io_event
    self.Applet.Plugins.PowerManager.UpdatePowerState()
  File "/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py", line 178, in UpdatePowerState
    self.BluetoothStatusChanged(new_state)
  File "/usr/lib64/python2.7/site-packages/dbus/decorators.py", line 300, in emit_signal
    for location in self.locations:
AttributeError: 'PowerManager' object has no attribute 'locations'
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:16)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:60)
deleting device /org/bluez/862/hci0/dev_00_02_76_4E_76_D5 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/862/hci0/dev_00_02_76_4E_76_D5 
_________
__init__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:16)
caching initial properties 
_________
__del__ (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:60)
deleting device /org/bluez/862/hci0/dev_00_02_76_4E_76_D5 
_________
Destroy (/usr/lib64/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/862/hci0/dev_00_02_76_4E_76_D5 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:87)
prop Pairable 0 
_________
UpdatePowerState (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/PowerManager.py:140)
off True 
foff False 
on True 
current state False 
new state False 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:87)
prop Powered 0 
_________
on_adapter_property_changed (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:87)
Stop 
_________
on_adapter_removed (/usr/bin/blueman-applet:109)
Adapter removed  /org/bluez/862/hci0 
_________
enumerate_connections (/usr/lib64/python2.7/site-packages/blueman/plugins/applet/ShowConnected.py:45)
Found 0 existing connections 
```

cschramm, 2015-02-24:
I fixed the cause of the exception in the master branch, but that should not be the reason.

From the kill switch information it looks like your kill switch with ID 0 gets set and setting the state to disabled is what I would expect from that. What kill switch information do you get when you restart the applet?

rworkman, 2015-02-25:
That seems to have fixed it actually. Unless a longer sleep time is relevant (and it didn't seem to be yesterday), it seems fine after that change in 9e32e12a - if it reappears, I'll update this report. Otherwise, I'll close it tomorrow.

rworkman, 2015-02-25:
Several suspends later, it's still good. Closing this report - 9e32e12 fixed it.

closed 2015-02-25 by rworkman