blueman-project/blueman #200: Device not connecting (some error on stdout)

davidgfnet, 2015-03-01:
Hey

Using fedora 21. Two computers (laptop with sony bluetooth) and desktop with chinese bt adapter. I can pair but it won't connect the device. stdout says:

Destroy (/usr/lib/python2.7/site-packages/blueman/main/Device.py:98)
invalidating device /org/bluez/hci0/dev_00_07_04_CE_2F_33 
ERROR:dbus.connection:Unable to set arguments ('org.bluez.Device1', dbus.ObjectPath('/org/bluez/hci0/dev_00_07_04_CE_2F_33'), 'Connect', []) according to signature None: <type 'exceptions.ValueError'>: Unable to guess signature from an empty list
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 177, in on_connect
    reply_handler=success, error_handler=fail, timeout=200)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 137, in __call__
    **keywords)
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 584, in call_async
    message.append(signature=signature, *args)
ValueError: Unable to guess signature from an empty list

Right after pairing (seems to be successful but disconnects after a second and won't reconnect no matter how much I try).

Thanks!

davidgfnet, 2015-03-01:
When trying to force connection I also get this err:

ERROR:dbus.connection:Unable to set arguments ('org.bluez.Device1', dbus.ObjectPath('/org/bluez/hci0/dev_00_07_04_CE_2F_33'), 'Connect', []) according to signature None: <type 'exceptions.ValueError'>: Unable to guess signature from an empty list
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 177, in on_connect
    reply_handler=success, error_handler=fail, timeout=200)
  File "/usr/lib/python2.7/site-packages/dbus/proxies.py", line 137, in __call__
    **keywords)
  File "/usr/lib/python2.7/site-packages/dbus/connection.py", line 584, in call_async
    message.append(signature=signature, *args)
ValueError: Unable to guess signature from an empty list

cschramm, 2015-03-02:
Hi David, I just merged some big changes concerning services into master. Can you give it a try and see if the issue persists?

davidgfnet, 2015-03-02:
Hey!
Seems to work better than last time (more stable). I repaired the device and now I have some options to connect to it like "Headset", "Audio sink", etc. But whenever I click on any option to connect I get this error:

Falló la conexión: Traceback (most recent call last):
  File "/usr/lib64/python2.7/site-packages/dbus/service.py", line 826, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python2.7/site-packages/blueman/main/DbusService.py", line 31, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/DBusService.py", line 72, in connect_service
    self.Applet.Plugins.RecentConns.notify(service)
  File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/RecentConns.py", line 253, in notify
    for i in RecentConns.items:
TypeError: 'NoneType' object is not iterable

Any idea?

infirit, 2015-03-02:
I suspect disabling the RecentConns plugin will work around this. Try to disable it in the plugins dialog.

I think I see why it is happening, in ``notify`` ``initialize`` is called just after the code that is now failing. And ``recover_state`` called in ``initialize`` will set ``RecentConns.items`` to an empty list when ``None``.

If you are up for it try adding ``self.recover_state()`` just before ``for i in RecentConns.items:``.

cschramm, 2015-03-02:
@infirit Yes, that's right, `notify` could be called before `recover_state` and then it fails. But I'd expect `recover_state` to be called first, as it's triggered by `on_manager_state_changed` which in terms is called by blueman-applet's `on_dbus_name_owner_change` handler function with a positive state.

@davidgfnet So, anything about the name owner change in your output?

infirit, 2015-03-03:
@cschramm I see a couple of catch all exceptions try/except blocks into a ``pass`` in ``on_manager_state_changed``. Like this it _could_ be possible  ``recover_state`` is never called but we never saw the exception, right?

davidgfnet, 2015-03-03:
Disabling the plugin makes it work on the laptop.
It will, after some struggle, connect and work. I have to go to pulseaudio sink and set A2DP to hear proper music.
On my desktop it does not work, but it might be because it's a crappy chinese BT adapter which is failing. In fact sometime I have to unplug it otherwise my bios gets upset on booting.
In both cases it fails at disconnecting. It seems impossible to disconnect a device (no errors at all but it wont' do it). It will be "disconnecting" forever until it forgets my request and seems to be connected again. The headset is not aware as well (as it beeps on disconnect).

cschramm, 2015-03-03:
> it would be nice for it to remember the audio profile it was set in previous session. I do not know whether that falls on pulseaudio's land though.

All we do is setting the card's profile in pulseaudio (via pa_context_set_card_profile_by_index). According to [this](http://lists.freedesktop.org/archives/pulseaudio-discuss/2014-July/020961.html) pulseaudio should remember that profile and we do not reset it in any way. If the "Off" profile is selected in blueman, that means it got reported as "active_profile" by pulseaudio (via pa_context_get_card_info_list).

> Also it's confusing having 3 possible "connect" options. Could you explain them to me?

I do not know what you are seeing. They depend on the services provided by the device.

> @davidgfnet So, anything about the name owner change in your output?

...for debugging your issues with the RecentConns plugin.

davidgfnet, 2015-03-03:
Hey,
I'll have to give me some clues on that last point. I don't get what you mean.
What I see is three connect options: headset, headphone and audio sink (well in spanish, so I can't ell how it looks like in english). Usually PA sets the device to the Off profile, so I have to manually set A2DP.
Thanks!

cschramm, 2015-03-03:
When you start blueman-applet in a terminal, you usually see something like this:

on_dbus_name_owner_change (/usr/local/bin/blueman-applet:89)
org.bluez owner changed to :1.19 

That's where RecentConns.items should get initialized, so I wonder if you see it in your output.

closed 2015-04-04 by cschramm