blueman-project/blueman #205: iPhone pairing fails when networking not available

juanitotc, 2015-03-03:
This may be connected to #189

Using bluez-5.24

Blueman scanning finds an iPhone, but when clicking on pair the pairing message appears on the iPhone, but nothing happens.

Blueman was not compiled against dhcpd or networkmanager, but several network warning messages appear.

cschramm, 2015-03-03:
As stated in #190 the most concerning data in the log is the failing mechanism service. The pairing failure may just be a consequence of that.

The dbus interface org.bluez.Network1 should be provided by BlueZ and is expected to be present at the device object /org/bluez/hci0/dev_54_E4_3A_22_48_E9 as it probably provides a network service according to its UUIDs. I have not seen a case where that interface is missing, yet. Did you apply any special configuration to BlueZ?

juanitotc, 2015-03-03:
bluez was compiled with:

./configure --prefix=/usr/local --disable-static --localstatedir=/var --libexecdir=/usr/local/lib --disable-systemd --enable-library

rworkman, 2015-03-03:
Just a shot in the dark: perhaps bluez's dbus config is in /usr/local/etc/dbus-1/ instead of /etc/dbus-1/

juanitotc, 2015-03-03:
I don't think so, networking was available after pairing with the iPhone with blueman from 10 days ago and bluez-4.101 compiled to /usr/local.

In addition things work with gnome-bluetooth and the same bluez-5.24 compiled to /usr/local

cschramm, 2015-03-03:
Well, at least we could catch and ignore that problem, but I really wonder how it may happen that BlueZ does not provide the interface although there's a network service. Can you find out the UUIDs the device provides (e.g. with bluetoothctl)?

Anyway, the very first thing we need to fix for you is blueman-mechanism, see #190, as it may cause a lot of different issues if it's not working.

juanitotc, 2015-03-03:
info 54:E4:3A:22:48:E9
Device 54:E4:3A:22:48:E9
        Name: juanito_tc iPhone
        Alias: juanito_tc iPhone
        Class: 0x7a020c
        Icon: phone
        Paired: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacafe)

cschramm, 2015-03-03:
So in fact there is a NAP service which should make BlueZ provide the interface. I guess the best you can do is getting some BlueZ debug output. Should be possible by just providing the -d option to bluetoothd. The network support in BlueZ 5 is called a "plugin", whatever that means, so maybe it is disabled / excluded for you or otherwise it may be present, but fail, e.g. because "kernel lacks bnep-protocol support".

juanitotc, 2015-03-04:
I noticed that the bnep kernel module is not loaded until the bluetooth daemon is started, so I manually loaded the module prior to starting the daemon, but that did not change anything.

The bluez-5 network plugin appears to be present, but I get a blueman pop-up stating that network services might not be available.

I used blueman to successfully pair headphones and set them as an audio sink and attempted to pair with an iPhone without success. I see there is a notification daemon error, does this need to be running for things to work?

bluetoothd output below:

$ sudo /usr/local/lib/bluetooth/bluetoothd -nd
bluetoothd[10075]: Bluetooth daemon 5.24
bluetoothd[10075]: src/main.c:parse_config() parsing main.conf
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'DiscoverableTimeout'
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'PairableTimeout'
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'AutoConnectTimeout'
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'Name'
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'Class'
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'DeviceID'
bluetoothd[10075]: src/main.c:parse_config() Key file does not have key 'ReverseServiceDiscovery'
bluetoothd[10075]: src/gatt.c:gatt_init() Starting GATT server
bluetoothd[10075]: src/adapter.c:adapter_init() sending read version command
bluetoothd[10075]: Starting SDP server
bluetoothd[10075]: src/sdpd-service.c:register_device_id() Adding device id record for 0002:1d6b:0246:0518
bluetoothd[10075]: src/plugin.c:plugin_init() Loading builtin plugins
bluetoothd[10075]: src/plugin.c:add_plugin() Loading hostname plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading wiimote plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading autopair plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading policy plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading a2dp plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading avrcp plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading network plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading input plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading hog plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading gatt plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading scanparam plugin
bluetoothd[10075]: src/plugin.c:add_plugin() Loading deviceinfo plugin
bluetoothd[10075]: src/plugin.c:plugin_init() Loading plugins /usr/local/lib/bluetooth/plugins
bluetoothd[10075]: profiles/input/suspend-dummy.c:suspend_init() 
bluetoothd[10075]: profiles/input/suspend-dummy.c:suspend_init() Created suspend-dummy FIFO on /tmp/hogsuspend
bluetoothd[10075]: profiles/network/manager.c:read_config() Config options: Security=true
bluetoothd[10075]: plugins/hostname.c:read_dmi_fallback() chassis: laptop
bluetoothd[10075]: plugins/hostname.c:read_dmi_fallback() major: 0x01 minor: 0x03
bluetoothd[10075]: src/main.c:main() Entering main loop
bluetoothd[10075]: src/rfkill.c:rfkill_event() RFKILL event idx 0 type 1 op 0 soft 0 hard 0
bluetoothd[10075]: Bluetooth management interface 1.6 initialized
bluetoothd[10075]: src/adapter.c:read_version_complete() sending read supported commands command
bluetoothd[10075]: src/adapter.c:read_version_complete() sending read index list command
bluetoothd[10075]: src/rfkill.c:rfkill_event() RFKILL event idx 1 type 2 op 0 soft 0 hard 0
bluetoothd[10075]: src/adapter.c:read_commands_complete() Number of commands: 47
bluetoothd[10075]: src/adapter.c:read_commands_complete() Number of events: 23
bluetoothd[10075]: src/adapter.c:read_index_list_complete() Number of controllers: 1
bluetoothd[10075]: src/adapter.c:read_index_list_complete() Found index 0
bluetoothd[10075]: src/adapter.c:index_added() index 0
bluetoothd[10075]: src/adapter.c:btd_adapter_new() System name: BlueZ 5.24
bluetoothd[10075]: src/adapter.c:btd_adapter_new() Major class: 0
bluetoothd[10075]: src/adapter.c:btd_adapter_new() Minor class: 0
bluetoothd[10075]: src/adapter.c:btd_adapter_new() Modalias: usb:v1D6Bp0246d0518
bluetoothd[10075]: src/adapter.c:btd_adapter_new() Discoverable timeout: 180 seconds
bluetoothd[10075]: src/adapter.c:btd_adapter_new() Pairable timeout: 0 seconds
bluetoothd[10075]: src/adapter.c:index_added() sending read info command for index 0
bluetoothd[10075]: src/adapter.c:read_info_complete() index 0 status 0x00
bluetoothd[10075]: src/adapter.c:clear_uuids() sending clear uuids command for index 0
bluetoothd[10075]: src/attrib-server.c:btd_adapter_gatt_server_start() Start GATT server in hci0
bluetoothd[10075]: src/attrib-server.c:attrib_db_add_new() handle=0x0001
bluetoothd[10075]: src/attrib-server.c:attrib_db_add_new() handle=0x0004
bluetoothd[10075]: src/attrib-server.c:attrib_db_add_new() handle=0x0006
bluetoothd[10075]: src/attrib-server.c:attrib_db_add_new() handle=0x0007
bluetoothd[10075]: src/attrib-server.c:attrib_db_add_new() handle=0x0008
bluetoothd[10075]: src/adapter.c:adapter_service_add() /org/bluez/hci0
bluetoothd[10075]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x10001
bluetoothd[10075]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000007-0000-1000-8000-00805f9
[... 433 lines cut ...]
bluetoothd[10075]: profiles/audio/a2dp.c:a2dp_sink_server_remove() path /org/bluez/hci0
[... cut ...]
bluetoothd[10075]: src/adapter.c:adapter_service_remove() /org/bluez/hci0
bluetoothd[10075]: src/adapter.c:remove_uuid() sending remove uuid command for index 0
bluetoothd[10075]: src/sdpd-service.c:remove_record_from_server() Removing record with handle 0x10001
bluetoothd[10075]: src/adapter.c:btd_adapter_unref() Freeing adapter /org/bluez/hci0
bluetoothd[10075]: src/adapter.c:adapter_free() 0x1f63af0
bluetoothd[10075]: src/gatt.c:gatt_cleanup() Stopping GATT server
bluetoothd[10075]: Stopping SDP server
bluetoothd[10075]: Exit

cschramm, 2015-03-04:
> I see there is a notification daemon error, does this need to be running for things to work?

Sure, a notification daemon (implementing the Desktop Notifications Specification) is a requirement (see README.md).

Looks like that's where the pairing fails (see the agent error reply in the BlueZ output), but does not explain the network service exception and the mechanism failure.

juanitotc, 2015-03-04:
I had to update notification-daemon to get it working reliably, but that is done now.

Now the bluetoothd error message is gone and the random pin message pops up on both the iPhone and the laptop, but pairing fails - I guess that is back to

cschramm, 2015-03-04:
If you get the LimitsExceeded exception, it may, yes.

Did the notification-daemon requirement also fix the Network1 interface issue?

juanitotc, 2015-03-04:
No, unfortunately not, the Network1 interface error is still there, though "bluetoothd -nd" shows that the bluez-5 network plugin is there.

cschramm, 2015-03-04:
Network1 interface issue is now #208.

Concerning the iPhone pairing that's discussed here: Do you actually get the LimitsExceeded exception like in #189?

juanitotc, 2015-03-05:
With the notification-daemon running, the iPhone pairing still fails with the "No such interface 'org.bluez.Network1'" error, but for me pairing shouldn't fail because of this and should let me use the audio interfaces, no?

` _________
on_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:148)
adapter propery changed Discovering 0 
_________
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/local/lib/python2.7/site-packages/blueman/services/meta/NetworkService.py", line 12, in connected
    return self._service.get_properties()['Connected']
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Network1'
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:166)
juanito_tc iPhone 
_________
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/local/lib/python2.7/site-packages/blueman/services/meta/NetworkService.py", line 12, in connected
    return self._service.get_properties()['Connected']
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Network1'
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:166)
juanito_tc iPhone 
_________
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/local/lib/python2.7/site-packages/blueman/services/meta/NetworkService.py", line 12, in connected
    return self._service.get_properties()['Connected']
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Network1'
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:166)
juanito_tc iPhone 
_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch Connected 1 /org/bluez/hci0/dev_54_E4_3A_22_48_E9 () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Connected 1 
_________
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
[... 32 lines cut ...]
row update event Connected 0 
_________
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed 
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
    ret = getattr(inst, function)(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
    add_menu_item(manager_menu, service)
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
    if service.connected:
  File "/usr/local/lib/python2.7/site-packages/blueman/services/meta/NetworkService.py", line 12, in connected
    return self._service.get_properties()['Connected']
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Network1'
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:166)
juanito_tc iPhone 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating down 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected)  `

cschramm, 2015-03-05:
No, #208 should not be connected to the pairing failure.

Your current problem is:
````
Traceback (most recent call last):
File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 607, in msg_reply_handler
*message.get_args_list()))
File "/usr/local/lib/python2.7/site-packages/blueman/bluez/Device.py", line 28, in err
raise exception
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
````
Expecting that bluetoothd is running and responding to dbus requests (a lot of other things would go wrong otherwise), I'd suspect the security policy as it would also explain you mechanism issues. :wink: 

You could check the bluetoothd debugging output for anything related to the pairing request (the term is "bonding" there).

juanitotc, 2015-03-06:
Here's what I get from bluetoothd when the iPhone pairing times out:

bluetoothd[32080]: src/device.c:bonding_request_new() Requesting bonding for 54:E4:3A:22:48:E9
bluetoothd[32080]: src/agent.c:agent_ref() 0x13fe250: ref=3
bluetoothd[32080]: src/agent.c:agent_unref() 0x13fe250: ref=2
bluetoothd[32080]: src/adapter.c:suspend_discovery() 
bluetoothd[32080]: src/adapter.c:adapter_bonding_attempt() hci0 bdaddr 54:E4:3A:22:48:E9 type 0 io_cap 0x01
bluetoothd[32080]: src/adapter.c:connected_callback() hci0 device 54:E4:3A:22:48:E9 connected eir_len 24
bluetoothd[32080]: src/adapter.c:user_confirm_request_callback() hci0 54:E4:3A:22:48:E9 confirm_hint 0
bluetoothd[32080]: src/device.c:new_auth() Requesting agent authentication for 54:E4:3A:22:48:E9
bluetoothd[32080]: src/agent.c:agent_ref() 0x13fe250: ref=3
bluetoothd[32080]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.13, path=/org/blueman/agent/global, passkey=251957
bluetoothd[32080]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 54:E4:3A:22:48:E9 type 0 status 0x5
bluetoothd[32080]: src/device.c:device_bonding_complete() bonding 0x13f9fe0 status 0x05
bluetoothd[32080]: src/agent.c:send_cancel_request() Sending Cancel request to :1.13, /org/blueman/agent/global
bluetoothd[32080]: src/device.c:device_cancel_authentication() Canceling authentication request for 54:E4:3A:22:48:E9
bluetoothd[32080]: src/agent.c:agent_unref() 0x13fe250: ref=2
bluetoothd[32080]: src/device.c:device_bonding_failed() status 5
bluetoothd[32080]: src/agent.c:agent_unref() 0x13fe250: ref=1
bluetoothd[32080]: src/adapter.c:resume_discovery() 
bluetoothd[32080]: src/adapter.c:pair_device_complete() Authentication Failed (0x05)
bluetoothd[32080]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 54:E4:3A:22:48:E9 type 0 status 0x5
bluetoothd[32080]: src/device.c:device_bonding_complete() bonding (nil) status 0x05
bluetoothd[32080]: src/device.c:device_bonding_failed() status 5
bluetoothd[32080]: src/adapter.c:resume_discovery() 
bluetoothd[32080]: src/adapter.c:dev_disconnected() Device 54:E4:3A:22:48:E9 disconnected, reason 2
bluetoothd[32080]: src/adapter.c:adapter_remove_connection() 
bluetoothd[32080]: plugins/policy.c:disconnect_cb() reason 2
bluetoothd[32080]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 54:E4:3A:22:48:E9 type 0 status 0xe
bluetoothd[32080]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[32080]: src/device.c:device_bonding_failed() status 14
bluetoothd[32080]: src/adapter.c:resume_discovery() 
bluetoothd[32080]: src/adapter.c:remove_temp_devices() /org/bluez/hci0

cschramm, 2015-03-06:
#213

closed 2015-03-06 by cschramm