blueman-project/blueman #206: blueman-mechanism cannot be launched by dbus

cschramm, 2015-03-04:
@juanitotc has an issue with the the dbus launch helper for blueman-mechanism in #190. The syslog shows
````
daemon.notice dbus[8450]: [system] Activating service name='org.blueman.Mechanism' (using servicehelper)
daemon.notice dbus[8450]: [system] Activated service 'org.blueman.Mechanism' failed: Launch helper exited with unknown return code 127
````

juanitotc, 2015-03-04:
with ref to the troubleshooting suggestions:

$ grep daemon-launch /usr/local/etc/dbus-1/system.conf
  <servicehelper>/usr/local/lib/dbus-1.0/dbus-daemon-launch-helper</servicehelper>

$ ls -ld /usr/local/lib/dbus-1.0/dbus-daemon-launch-helper
lrwxrwxrwx    1 root     root            65 Mar  4 19:07 /usr/local/lib/dbus-1.0/dbus-daemon-launch-helper -> /tmp/tcloop/dbus/usr/local/lib/dbus-1.0/dbus-daemon-launch-helper
$ ls -ld /tmp/tcloop/dbus/usr/local/lib/dbus-1.0/dbus-daemon-launch-helper
-rwsr-xr-x    1 root     root        242920 Oct 20 11:24 /tmp/tcloop/dbus/usr/local/lib/dbus-1.0/dbus-daemon-launch-helper

Note that on this system applications are loop mounted squashfs packages symlinked to the root fs - is blueman OK with this (I ask because a while ago one of the networkmanager devs had to made a few mods to make things work with this setup)?

There is no messagebus group.

cschramm, 2015-03-04:
I think what dbus does is calling your dbus-daemon-launch-helper with the path to blueman-mechanism as an argument, but unfortunately I don't know how the environment has to try that manually.

> Note that on this system applications are loop mounted squashfs packages symlinked to the root fs - is blueman OK with this (I ask because a while ago one of the networkmanager devs had to made a few mods to make things work with this setup)

I think that should not be an issue. Can you point me to the network-manager changes?

juanitotc, 2015-03-05:
After some searching, I found it was folks instead of networkmanager - here's the patch:

Instead of ignoring symlinks, we should resolve their targets, and
deduplicate after that. This fixes cases where folks is run in an
environment with software installed in loop mounted file systems â€” each
backend .so file in the backend path is actually a symlink to the proper
.so file in the loop mounted file system. This is the case on
Tinycorelinux. folks was previously not loading any backends in such an
environment because it ignored all symlinks.

 folks/backend-store.vala | 43 +++++++++++++++++++++++++++++++++++++++----
 1 file changed, 39 insertions(+), 4 deletions(-)

diff --git a/folks/backend-store.vala b/folks/backend-store.vala
index e2ec287..8e20e89 100644
--- a/folks/backend-store.vala
+++ b/folks/backend-store.vala
@@ -632,6 +632,7 @@ public class Folks.BackendStore : Object {
           FileAttribute.STANDARD_NAME + "," +
           FileAttribute.STANDARD_TYPE + "," +
           FileAttribute.STANDARD_IS_SYMLINK + "," +
+          FileAttribute.STANDARD_SYMLINK_TARGET + "," +
           FileAttribute.STANDARD_CONTENT_TYPE;
 
       GLib.List<FileInfo> infos;
@@ -660,11 +661,45 @@ public class Folks.BackendStore : Object {
       foreach (var info in infos)
         {
           var file = dir.get_child (info.get_name ());
+
+          /* Handle symlinks by derefencing them. If we look at two symlinks
+           * with the same target, we donâ€™t end up loading that backend twice
+           * due to hashing the backendâ€™s absolute path in @modules_final.
+           *
+           * We canâ€™t just ignore symlinks due to the way Tinycorelinux installs
+           * software: /usr/local/lib/folks/41/backends/bluez/bluez.so is a
+           * symlink to
+           * /tmp/tcloop/folks/usr/local/lib/folks/41/backends/bluez/bluez.so,
+           * a loopback squashfs mount of the folks file system. Ignoring
+           * symlinks means we would never load backends in that environment. */
+          if (info.get_is_symlink ())
+            {
+              debug ("Handling symlink â€˜%sâ€™ to â€˜%sâ€™.",
+                  file.get_path (), info.get_symlink_target ());
+
+              var old_file = file;
+              file = dir.resolve_relative_path (info.get_symlink_target ());
+
+              try
+                {
+                  info =
+                      yield file.query_info_async (attributes,
+                          FileQueryInfoFlags.NONE);
+                }
+              catch (Error error)
+                {
+                  /* Translators: the first parameter is a folder path and the second
+                   * is an error message. */
+                  warning (_("Error querying info for target â€˜%sâ€™ of symlink â€˜%sâ€™: %s"),
+                      file.get_path (), old_file.get_path (), error.message);
+
+                  continue;
+                }
+            }
+
+          /* Handle proper files. */
           var file_type = info.get_file_type ();
           unowned string content_type = info.get_content_type ();
-          /* don't load the library multiple times for its various symlink
-           * aliases */
-          var is_symlink = info.get_is_symlink ();
 
           string? mime = ContentType.get_mime_type (content_type);
 
@@ -679,7 +714,7 @@ public class Folks.BackendStore : Object {
                     }
                 }
             }
-          else if (mime == "application/x-sharedlib" && !is_symlink)
+          else if (mime == "application/x-sharedlib")
             {
               var path = file.get_path ();
               if (path != null)
--

infirit, 2015-03-05:
@juanitotc small request to make things more readable. Please put logs/patches etc between backticks like I just did for the patch :smile: .

juanitotc, 2015-03-05:
I'll try, but in general this site is very difficult to work with in either of two browsers I use - fifth or opera - I just had to switch to windows to paste a log.

I take it that [code][/code] and [quote][/quote] don't work?

juanitotc, 2015-03-05:
`testing, testing`

juanitotc, 2015-03-05:
I loaded the entire filesystem to ram (i.e. no symlinks to squashfs packages) and the problem is still there, so there cannot be an issue with symlinks.

dbus is compiled with "-fno-exceptions", could this be a problem?

cschramm, 2015-03-05:
We do not have anything similar to that directory traversal in blueman, so symlinks should be fine.

> dbus is compiled with "-fno-exceptions", could this be a problem?

No, that's just a language feature.

Did you try invoking blueman-mechanism via dbus-daemon-launch-helper?

juanitotc, 2015-03-06:
Some progress :)

$ /usr/local/lib/dbus-1.0/dbus-daemon-launch-helper org.blueman.Mechanism
env: can't execute 'python': No such file or directory

..so it seems like when called in this way /usr/bin/env does not work. Changing the shebang to /usr/local/bin/python fixes this:

$ /usr/local/lib/dbus-1.0/dbus-daemon-launch-helper org.blueman.Mechanism
Starting blueman-mechanism 
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

(blueman-mechanism:32173): Gtk-CRITICAL **: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading RfKill 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading Rfcomm 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading Ppp 
_________
__init__ (/usr/local/lib/blueman/blueman-mechanism:109)
loading Network

cschramm, 2015-03-06:
So does it work? I'm don't know where the "Unable to init server: Could not connect: Connection refused" messages come from (maybe triggered in blueman-mechanism, line 110).

juanitotc, 2015-03-06:
How do I hack blueman-mechanism to temporarily force it accept the connection?

BTW blueman-mechanism, line 77 does not contain /usr/local, but I don't know if this affects anything?

cschramm, 2015-03-06:
As I said, I have no idea where that message is triggered and what it means. None of the three strings is part of blueman.

juanitotc, 2015-04-07:
In more recent compilations I have been changing the shebang to /usr/local/bin/python in /usr/local/lib/blueman/blueman-mechanism and the problem has gone away, so it seems this fixes it.

cschramm, 2015-04-07:
I really wonder why /usr/bin/env does not work for your dbus daemon. Is it that your root user has no python in his PATH?

closed 2015-04-07 by cschramm

juanitotc, 2015-04-08:
I double-checked and python is in the path of the root user.

The issue only seems to occur with the script in libexec, those in /bin seem to work fine with "/usr/bin/env python"

cschramm, 2015-04-08:
Interesting.

Not sure if there's a better shebang for libexec then...