blueman-project/blueman #208: No such interface 'org.bluez.Network1'

cschramm, 2015-03-04:
Looks like it is possible to have network a service UUID on a device, but no org.bluez.Network1 interface on its dbus object.

This appears in https://github.com/blueman-project/blueman/issues/89#issuecomment-64226892 and https://github.com/blueman-project/blueman/issues/205#issue-59595871.

````
Run (/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:203)
Function on_request_menu_items on Services Failed
Traceback (most recent call last):
File "/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py", line 207, in Run
ret = getattr(inst, function)(args, *kwargs)
File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 54, in on_request_menu_items
add_menu_item(manager_menu, service)
File "/usr/local/lib/python2.7/site-packages/blueman/plugins/manager/Services.py", line 29, in add_menu_item
if service.connected:
File "/usr/local/lib/python2.7/site-packages/blueman/services/meta/NetworkService.py", line 12, in connected
return self._service.get_properties()['Connected']
File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
raise parse_dbus_error(exception)
DBusException: org.freedesktop.DBus.Error.InvalidArgs: No such interface 'org.bluez.Network1'
````

The thing is, from the BlueZ source code for the network profile I'd say it will create the interface as soon at finds a NAP service when probing the profiles.

@juanitotc, you provided BlueZ debugging output in https://github.com/blueman-project/blueman/issues/205#issuecomment-77102897. Unfortunately you had issues with notification-daemon which is probably the reason we do not see BlueZ probing the profiles for device 54:E4:3A:22:48:E9. Can you retry this now? I'm expecting debugging output like "src/device.c:device_probe_profiles() Probing profiles for device 54:E4:3A:22:48:E9".

juanitotc, 2015-03-05:
"profiles/network/server.c:server_register() Registered interface org.bluez.NetworkServer1 on path /org/bluez/hci0" looks promising, but the attempts to pair with the iPhone still time out :(

Full "bluetoothd -nd" out put below:

cschramm, 2015-03-05:
Still no probing in there, probably due to the failed pairing. So I guess it's possible to discover a network service on an unpaired device while BlueZ will only pick it up when paired. I'll implement a simple workaround to avoid the crash.

cschramm, 2015-03-05:
OK, exception should not appear anymore in master.

commit 00513d0e (2015-03-05): Do not raise an exception if a network service lacks its dbus interface

Closes #208
closed 2015-03-05 by cschramm