blueman-project/blueman #213: iPhone pairing fails

juanitotc, 2015-03-06:
Since the previous network issue and the blueman-mechanism problem has been found, I'll start a new thread.

The key issues seems to be "dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken."

Where is the owner/security policy for the message bus stated (I am not using consolekit or polkit)?

Note that using backticks does not work for me for posting code, is there another was to insert code tags?

bluetoothd, blueman-applet and blueman-manager logs posted below:

cschramm, 2015-03-06:
The interesting part of the BlueZ output is:

````
bluetoothd[9138]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.13, path=/org/blueman/agent/global, passkey=728002
bluetoothd[9138]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 54:E4:3A:22:48:E9 type 0 status 0x5
bluetoothd[9138]: src/device.c:device_bonding_complete() bonding 0x80d6a0 status 0x05
bluetoothd[9138]: src/agent.c:send_cancel_request() Sending Cancel request to :1.13, /org/blueman/agent/global
bluetoothd[9138]: src/device.c:device_cancel_authentication() Canceling authentication request for 54:E4:3A:22:48:E9
````
The corresponding part in the blueman-applet output:
````
RequestConfirmation (/usr/local/lib/python2.7/site-packages/blueman/main/applet/BluezAgent.py:196)
Agent.RequestConfirmation

Cancel (/usr/local/lib/python2.7/site-packages/blueman/main/applet/BluezAgent.py:143)
Agent.Cancel
````

One thing that would be interesting is the time difference between those events, especially if there is a larger gap between agent_request_confirmation, bonding_attempt_complete, and send_cancel_request.

But actually I suspect another notification issue here. You should get a notification when RequestConfirmation is triggered. Do you?

juanitotc, 2015-03-06:
I'm not totally sure what is meant by "notification", but I get a pop-up on the laptop asking for the OK to use a random pin to pair and a corresponding pop-up on the iPhone asking for the OK to pair with a device sending the random pin.

Maybe it's an issue of the order - which pop-up should I click on OK first, the laptop or the iPhone?

cschramm, 2015-03-06:
Well, try both way. From my perspective it does not matter, but one of the sides somehow triggers the `src/adapter.c:auth_failed_callback`. As BlueZ then cancels the confirmation request it sent to blueman, I guess it's triggered by the iPhone. It may also happen before you click anything. Just have a look at the BlueZ and / or blueman output to see at which point in time the cancel appears.

juanitotc, 2015-03-07:
It's somewhat difficult to watch two terminal windows at the same time, but it appears that, no matter which popup I click OK on first, the timeout appears in the blueman window before the bonding cancellation in the bluetoothd output.

Note also (to check that this isn't some iPhone weirdness) I tried to pair with a laptop running win7 and the same thing happens.

juanitotc, 2015-03-07:
After trying five or six times, I finally suceeded in pairing with the iPhone - to double-check things were working I then paired with some headphones and managed to play music from the iPhone via the laptop and out to the headphones to verify things were working.

I'm guessing this means that there is some kind of timing issue/race condition when trying to pair with something other than headphones/headset?

BTW, the pop-up to confirm or deny the request to pair comes up by default in the adwaita dark theme and it is quite difficult to see the "confirm" or "deny" buttons (black/brown on a brown background).

juanitotc, 2015-03-07:
Using bluez-5.28

So far pairing works first time with the updated bluez :)

cschramm, 2015-03-27:
Maybe same as https://github.com/blueman-project/blueman/issues/216#issuecomment-77686095. I'll close this.

closed 2015-03-27 by cschramm