blueman-project/blueman #216: blueman will not accept incoming pairing request

juanitotc, 2015-03-07:
Using bluez-5.24

When generating an incoming pair request from a laptop running win7, blueman pops-up a notification asking to confirm or deny the pairing request. After clicking on "confirm", blueman does not pair.

From blueman-manager:

on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch Connected 1 /org/bluez/hci0/dev_C4_46_19_E9_C1_EC () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Connected 1 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
laptop 
_________
monitor_power_levels (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:178)
starting monitor 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating up

Then bluetoothd:

bluetoothd[10108]: src/device.c:new_auth() Requesting agent authentication for C4:46:19:E9:C1:EC
bluetoothd[10108]: src/agent.c:agent_ref() 0xf1b2a0: ref=2
bluetoothd[10108]: src/agent.c:agent_request_confirmation() Calling Agent.RequestConfirmation: name=:1.14, path=/org/blueman/agent/global, passkey=054810
bluetoothd[10108]: src/agent.c:agent_ref() 0xf1b2a0: ref=3
bluetoothd[10108]: src/adapter.c:btd_adapter_confirm_reply() hci0 addr C4:46:19:E9:C1:EC success 1
bluetoothd[10108]: src/agent.c:agent_unref() 0xf1b2a0: ref=2
bluetoothd[10108]: src/agent.c:agent_unref() 0xf1b2a0: ref=1
bluetoothd[10108]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr C4:46:19:E9:C1:EC type 0 status 0x5
bluetoothd[10108]: src/device.c:device_bonding_complete() bonding (nil) status 0x05
bluetoothd[10108]: src/device.c:device_cancel_authentication() Canceling authentication request for C4:46:19:E9:C1:EC
bluetoothd[10108]: src/device.c:device_bonding_failed() status 5
bluetoothd[10108]: src/adapter.c:resume_discovery() 
bluetoothd[10108]: src/adapter.c:dev_disconnected() Device C4:46:19:E9:C1:EC disconnected, reason 3
bluetoothd[10108]: src/adapter.c:adapter_remove_connection() 
bluetoothd[10108]: plugins/policy.c:disconnect_cb() reason 3
bluetoothd[10108]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr C4:46:19:E9:C1:EC type 0 status 0xe
bluetoothd[10108]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
bluetoothd[10108]: src/device.c:device_bonding_failed() status 14
bluetoothd[10108]: src/adapter.c:resume_discovery() 

Then blueman-manager:

_________
on_device_property_changed (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:156)
list: device_prop_ch Connected 0 /org/bluez/hci0/dev_C4_46_19_E9_C1_EC () {} 
_________
row_update_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:261)
row update event Connected 0 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
laptop 
_________
level_setup_event (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating down 
_________
update (/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected)

juanitotc, 2015-03-07:
After many items (and without changing anything), I suceeded in pairing - maybe there is the same timing issue as pairing in the other direction?

juanitotc, 2015-03-07:
Using bluex-5.28

So far pairing works first time with the updated bluez :)

cschramm, 2015-03-27:
Looks like a BlueZ issue then. I'll close this.

closed 2015-03-27 by cschramm