blueman-project/blueman #217: Cannot set iPhone as network access point

juanitotc, 2015-03-07:
Using bluez-5.24

..and after finally pairing with the iPhone :)

With networkmanager running, I get a pop-up that bnep0 is unable to get an ip address.

From blueman-applet:

err (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DhcpClient.py:49)
org.freedesktop.DBus.Python.Exception: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/blueman/plugins/mechanism/Network.py", line 31, in DhcpClient
    dh.Connect()
  File "/usr/local/lib/python2.7/site-packages/blueman/main/DhcpClient.py", line 39, in Connect
    raise Exception("no DHCP client found, please install dhcpcd or dhclient")
Exception: no DHCP client found, please install dhcpcd or dhclient

From bluetoothd:

bluetoothd[10108]: profiles/network/connection.c:connection_connect() id 4374
bluetoothd[10108]: profiles/network/connection.c:bnep_conn_cb() 
bluetoothd[10108]:  connected
bluetoothd[10108]: src/service.c:change_state() 0xf3ebe0: device 54:E4:3A:22:48:E9 profile network-nap state changed: disconnected -> connected (0)
bluetoothd[10108]: src/adapter.c:remove_temp_devices() /org/bluez/hci0
bluetoothd[10108]: profiles/network/connection.c:connection_connect() id 4374
bluetoothd[10108]: profiles/network/connection.c:bnep_conn_cb() 
bluetoothd[10108]: bnep0 connected
bluetoothd[10108]: src/agent.c:agent_disconnect() Agent :1.14 disconnected
bluetoothd[10108]: src/agent.c:remove_default_agent() Default agent cleared
bluetoothd[10108]: src/agent.c:agent_destroy() agent :1.14
bluetoothd[10108]: src/agent.c:agent_unref() 0xf1b2a0: ref=0
bluetoothd[10108]: src/agent.c:agent_ref() 0xf44600: ref=1
bluetoothd[10108]: src/agent.c:register_agent() agent :1.36
bluetoothd[10108]: src/agent.c:add_default_agent() Default agent set to :1.36 /org/blueman/agent/global

from blueman-manager:

set_op (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:77)
op: regenerating instance <ManagerDeviceMenu object at 0x7ffba68029b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x1420cc0)> 
_________
success (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:105)
success (dbus.String(u'bnep0'),) 
_________
unset_op (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:91)
op: regenerating instance <ManagerDeviceMenu object at 0x7ffba67f6050 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x1420a40)> 
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
juanito_tc iPhone 
_________
unset_op (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:91)
op: regenerating instance <ManagerDeviceMenu object at 0x7ffba68029b0 (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x1420cc0)> 
/usr/local/lib/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 2105 was not found when attempting to remove it
  return fn(*args, **kwargs)
_________
Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:165)
juanito_tc iPhone

cschramm, 2015-03-07:
If you want nm to manage that, you need to select it in the local services settings.

cschramm, 2015-03-07:
Wrong. You need to enable the plugin / extension and disable the dchp plugin.

juanitotc, 2015-03-07:
OK...

How do I do that?

cschramm, 2015-03-07:
Right click on the applet and choose Extensions. Then you have a list of extensions you can enable / disable. You may encounter a specific bug when you try to enable the nm plugin (should be called NMPANSupport) without disabling the conflicting dhcp plugin first. If so, please open a ticket for that. :wink:

juanitotc, 2015-03-07:
I'm using a very basic wm - flwm - that does not have a "dock" for applets so there's nothing to right-click on...

juanitotc, 2015-03-07:
As per dconf-editor org.blueman.general plugin-list:

Stores which plugins have been disabled/enabled by the user. Note plugins have a default whether it should be loaded or not and this only stores the plugins enabled/disabled by the user

..but what format do I use to disable DhcpClient

cschramm, 2015-03-07:
You're a really challenging user. :laughing:

You're at the right spot to do it by hand. Doesn't it have a value for you yet? I think it should be an array of plugin names, but I cannot check it as I'm currently in transit to off-line-land. Maybe @infirit can give guidance with this.

infirit, 2015-03-07:
It is a list/array of string with the plugin name. If the string begins with an exclamation mark (!) it is disabled. So the list format is something like: ``['!DhcpClient', 'OtherEnabledPlugin']``.

juanitotc, 2015-03-07:
gsettings set org.blueman.general plugin-list "['DhcpClient']" will change the key, but to how to make the difference between "enabled" and "disabled"? It seems to be treated as "enabled"...

juanitotc, 2015-03-07:
Thanks:

gsettings set org.blueman.general plugin-list "['!DhcpClient', 'NMPANSupport']"

Makes the appropriate change, but blueman-applet says:

on_adapter_property_changed (/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:91)
prop UUIDs dbus.Array([dbus.String(u'00001200-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001800-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001801-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001116-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
('\x1b[33;01mwarning:\x1b[39;49;00m', 'Plugin NMPANSuppport not found')

but:

$ ls /usr/local/lib/python2.7/site-packages/blueman/plugins/applet/NMP*
/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/NMPANSupport.py

cschramm, 2015-03-07:
Typppo. :wink:

juanitotc, 2015-03-07:
Indeed, but even without the typo it doesn't work :(

infirit, 2015-03-07:
>raise Exception("no DHCP client found, please install dhcpcd or dhclient")
^^

juanitotc, 2015-03-07:
Yes, but networkmanager is greyed out.

The networkmanager daemon is running, but blueman won't load NMPANSupport

cschramm, 2015-03-26:
@infirit Looks like NMPANSupport did not get ported to gsettings. :cry:

cschramm, 2015-03-26:
#227

infirit, 2015-03-26:
There is nothing to be ported wrt blueman functionality. The gconf bits in there are for NM keys only, https://github.com/blueman-project/blueman/issues/181.

Unless this is moved over to the NM dbus api GConf gir and typelib will be needed.

cschramm, 2015-03-26:
Sure... Hm. I'll look into porting that to the dbus API, but I don't know if it provides the functionality.

@juanitotc You need gconf for the plugin to load.

juanitotc, 2015-03-27:
..but gconf is old and depreciated and thus not present on my system...

cschramm, 2015-04-16:
Can we close this?

juanitotc, 2015-04-16:
Sure :)

closed 2015-04-16 by cschramm