blueman-project/blueman #218: Cannot transfer a file from blueman laptop

juanitotc, 2015-03-07:
Using bluez-5.24 and obexd-0.48

When trying to send a file from a laptop using blueman, I get a pop-up box with the error message:

Method "CreateSession" with signature "va{sv}" on interface "org.bluez.obex.Client1" doesn't exist

cschramm, 2015-03-07:
As far as I know BlueZ 5 is not compatible with obexd 0.48. It brings it's own obexd. The 0.x versions are for BlueZ 4.

juanitotc, 2015-03-07:
with bluez-5.28

OK, so I compiled obexd with the latest bluez and obexd is located in the bluez libexecdir and not the obexd libexecdir so blueman cannot find it - ref pop-up "obexd is probably not installed"

cschramm, 2015-03-07:
I think we have a too broad exception clause in blueman-sendto, line 98, and you're actually encountering an exception that is not reflected by that message. The installation paths do not matter as we only use the dbus API.

juanitotc, 2015-03-07:
I see bluez does install share/dbus-1/services/obexd.service?

cschramm, 2015-03-07:
It may be provided by the main bluez service.

Best way to debug this is to remove the except clause to see the actual exception, I guess.

juanitotc, 2015-03-07:
I can try - what edits do I need to make?

juanitotc, 2015-03-07:
Supposing I hacked correctly:

__init__ (/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:21)
caching initial properties 
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Traceback (most recent call last):
  File "/usr/local/bin/blueman-sendto", line 372, in <module>
    SendTo()
  File "/usr/local/bin/blueman-sendto", line 323, in __init__
    self.do_send()
  File "/usr/local/bin/blueman-sendto", line 332, in do_send
    sender = Sender(self.device, self.adapter, self.files)
  File "/usr/local/bin/blueman-sendto", line 97, in __init__
    self.client = obex.Client()
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/obex/Client.py", line 11, in __init__
    if self.__class__.get_interface_version()[0] < 5:
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/obex/Base.py", line 19, in get_interface_version
    lookup_object('org.bluez.obex', '/org/bluez/obex')
  File "/usr/local/lib/python2.7/site-packages/blueman/bluez/errors.py", line 143, in warp
    raise parse_dbus_error(exception)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.Spawn.ChildExited: Process org.bluez.obex exited with status 1
_________
child_closed (/usr/local/lib/python2.7/site-packages/blueman/Functions.py:154)
['/usr/local/bin/blueman-sendto', u'--source=A0:A8:CD:2C:3E:17', u'--device=C4:46:19:E9:C1:EC'] closed

cschramm, 2015-03-27:
> Process org.bluez.obex exited with status 1

So your obexd service is available, but fails. Can you start obexd manually (/usr/lib/bluetooth/obexd on my system)?

juanitotc, 2015-03-31:
If I start obexd manually (/usr/local/lib/bluetooth/obexd) then I can successfully transfer files to another laptop - it's not obvious why blueman cannot do this itself though (unless it's the /usr/local/bin/python thing).

I cannot receive files from another laptop, but I think this is already the subject of another bug report.

cschramm, 2015-03-31:
So it's just dbus that cannot start your obexd. Please fire up a python shell and type:
````
import dbus
dbus.SessionBus().get_object('org.bluez.obex', '/org/bluez/obex')
````
You'll probably get the same error as in blueman. The question then is why. Maybe your dbus-daemon's log (syslog) has more on it.

> I cannot receive files from another laptop, but I think this is already the subject of another bug report.

Yes, receiving is not currently possible.

juanitotc, 2015-03-31:
It's looking for org.bluez.obex.service, but this file is not provided by bluex-5.28...

import dbus
dbus.SessionBus().get_object('org.bluez.obex', '/org/bluez/obex')

$ python
Python 2.7.8 (default, Oct 12 2014, 12:22:14) 
[GCC 4.9.1] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import dbus
>>> dbus.SessionBus().get_object('org.bluez.obex', '/org/bluez/obex')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/local/lib/python2.7/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/local/lib/python2.7/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/local/lib/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.bluez.obex was not provided by any .service files
>>>

cschramm, 2015-03-31:
Then I'd expect the same error in blueman as this is exactly what it does at

````
File "/usr/local/lib/python2.7/site-packages/blueman/bluez/obex/Base.py", line 19, in get_interface_version
lookup_object('org.bluez.obex', '/org/bluez/obex')
````

Did you change anything in the meantime? Does blueman still show the ChildExited error?

juanitotc, 2015-04-01:
I tried again to double-check - after going through the send file dialogue, selecting a file and clicking on send, blueman pops-up a warning that obexd does not appear to exist on the system.

Here's the blueman-applet/manager output:

tc@boxdell:~$ blueman-applet
blueman-applet version 1.99.alpha2 starting
/usr/local/lib/python2.7/site-packages/gi/overrides/__init__.py:175: Warning: The property GSettings:schema is deprecated and shouldn't be used anymore. It will be removed in a future version.
  return super_init_func(self, **new_kwargs)
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
['TransferService', 'StatusIcon', 'StandardItems', 'ShowConnected', 'SerialManager', 'RecentConns', 'PowerManager', 'PPPSupport', 'Networking', 'NetUsage', 'NMPANSupport', 'NMDUNSupport', 'Menu', 'KillSwitch', 'Headset', 'GameControllerWakelock', 'ExitItem', 'DiscvManager', 'DhcpClient', 'DBusService', 'AuthAgent', 'AppIndicator'] 
_________
(/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module AppIndicator
cannot import name AppIndicator3 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/plugins/applet/DiscvManager.py:118)
warning: Adapter is None 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
Not loading DhcpClient because it's conflict has higher priority 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.ExitItem.ExitItem'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
[... 372 lines cut ...]
JACK-M14X 
_________
(/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:322)
animating down 
_________
(/usr/local/lib/python2.7/site-packages/blueman/gui/DeviceList.py:179)
stopping monitor (not connected) 
_________
(/usr/local/lib/python2.7/site-packages/blueman/bluez/BlueZInterface.py:8)
Detected BlueZ 5 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:21)
caching initial properties 
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:65)
deleting device /org/bluez/hci0/dev_20_68_9D_60_CF_09 
_________
(/usr/local/lib/python2.7/site-packages/blueman/main/Device.py:87)
invalidating device /org/bluez/hci0/dev_20_68_9D_60_CF_09 
_________
(/usr/local/lib/python2.7/site-packages/blueman/Functions.py:154)
['/usr/local/bin/blueman-sendto', u'--source=A0:A8:CD:2C:3E:17', u'--device=20:68:9D:60:CF:09'] closed 
^C

cschramm, 2015-04-03:
After removing the except clause you reported the traceback in https://github.com/blueman-project/blueman/issues/218#issuecomment-77690408. Base.py, line 19 does basically the same as you did in https://github.com/blueman-project/blueman/issues/218#issuecomment-88084562, but the traceback in blueman means the service is available but fails and the traceback of your manual test means the service is not even available which is contradicting.

Can you confirm you can still produce the same traceback in blueman while getting the ServiceUnknown exception when trying manually?

juanitotc, 2015-04-03:
Just to be sure I understood, you'd like me to manually start the obexd and then try to send a file?

cschramm, 2015-04-03:
No, with "manual test" I mean the python code you types in https://github.com/blueman-project/blueman/issues/218#issuecomment-88084562. It is the same as what blueman does, but gives a contradicting result to the exception in https://github.com/blueman-project/blueman/issues/218#issuecomment-77690408 which I just can't make a head or tail of.

cschramm, 2015-04-03:
> If you paste the contents of your obexd dbus service file I can create one to suit and test.

My service file is `/usr/share/dbus-1/services/org.bluez.obex.service` and contains:
````
[D-BUS Service]
Name=org.bluez.obex
Exec=/usr/lib/bluetooth/obexd
SystemdService=dbus-org.bluez.obex.service
````

Anyway, you must have some service configuration for org.bluez.obex already as https://github.com/blueman-project/blueman/issues/218#issuecomment-77690408 shows that dbus tries to start something and does not give up immediately (opposed to https://github.com/blueman-project/blueman/issues/218#issuecomment-88084562).

juanitotc, 2015-04-04:
As far as I can tell, repeating the obex test results in the same output from the python shell and the same output from blueman-manager.

Perhaps bluez-5.x has another way to launch the obexd (since it does not install org.bluez.obex.service)?

If I create org.bluez.obex.service then everything works fine.

cschramm, 2015-04-04:
So we can close this? :smiley: 

All I can say for the service file is that (the current version of) BlueZ's Makefile.obexd contains:
````
dbussessionbusdir = @DBUS_SESSIONBUSDIR@
dbussessionbus_DATA = obexd/src/org.bluez.obex.service
````

juanitotc, 2015-04-05:
Please do, if there's an issue, it's with bluez

closed 2015-04-05 by cschramm