blueman-project/blueman #220: missing sendto plugin for caja and nemo

NiceandGently, 2015-03-09:
sendto plugins for caja and nemo are missing, or does the nautilus-plugin work with caja/nemo?

NiceandGently, 2015-03-11:
Ok, first i wanna say that bueman is now working in fedora with 1.99.alpha2, so the chances are huge that it will get into fedora 22.
But nemo hasn't a sento plugin , so i changed the title here.
In a test build we renamed the nautilus sendto plugin to caja, but bluemen is build with GTK3 and caja use GTK2.
So can this work with differrent gtk+ versions.
Personaly i don't have a device for testing only a tongle and i don't see anything about bluetooth in caja sendto menu. Is this because of the missing bluetooth device?

cschramm, 2015-03-26:
Fixed by https://github.com/blueman-project/blueman/commit/768b0d1f138f9ba22d7900044b7f24bcac15925d?

cschramm, 2015-03-26:
Oh sorry, that one got reverted. @infirit can you comment on this please?

infirit, 2015-03-26:
It was a hack and did not work, it unfortunately got into my master branch by accident.

I have even tried to see if it worked in Nautilus but it does not show up. Does it work at all with recent Nautilus?

infirit, 2015-03-26:
The current nautilus sendto extension does not allow for plugins which explains it not showing up.

cschramm, 2015-03-27:
Hm. Is there another way then? Maybe similar to thunar?

NiceandGently, 2015-03-27:
Why not renaming the plugin to caja?

cschramm, 2015-03-28:
Looks like support for nautilus-sendto is not possible anymore. empathy also dropped their plugin:

> The nautilus-sendto plugin has been removed as latest nautilus-sendo
   versions don't support plugins any more.

It is possible to add entries to the context menu by creating scripts in `~/.local/share/nautilus/scripts`, but this a) is user-specific and not possible globally and b) creates the entry in an odd sub menu called `Scripts`.

Renaming and using it for caja-sendto makes sense to me, but we have to make it work. :laughing:

cschramm, 2015-03-28:
@stefano-k @infirit Should we add the plugin to blueman or caja-sento itself?

cschramm, 2015-03-28:
Basically, caja-sendto needs to be built against GTK 3, but I think there are more issues. You can see the output of what's going on by running caja-sendto directly in a terminal.

monsta, 2015-05-26:
Hi @cschramm,

The reporter of https://github.com/mate-desktop/caja-extensions/issues/12 said that a simple symlinking of nautilus-sendto plugin to caja-sendto plugin dir worked for him. Is the GTK+3 build of caja-sendto actually required?

cschramm, 2015-05-26:
He's probably working with the old 1.23 from the Ubuntu repositories which uses GTK+ 2.

Current versions of blueman use GTK+ 3 and the nautilus stuff has completely fallen by the wayside. A GTK+ 3 build would have been necessary, but unfortunately we cannot get our Python codebase running when linked against caja-sendto due to the use of GObject introspection anyway.

I'm planning to contribute a bluetooth / blueman plugin which does not use any python at all to caja-sendto, but did not find the time yet.

monsta, 2015-05-26:
Ah, right... Ubuntu 14.04 repos have blueman 1.23. That explains it, thanks.

NiceandGently, 2015-05-26:
@cschramm 
As you can see we started packaging blueman for fedora.
https://bugzilla.redhat.com/show_bug.cgi?id=489564#c62
Is there a minimum required GTK3 version?
We can pack blueman for rhel7 family too via epel7 repo, but rhel7 use GT+-3.8.x

cschramm, 2015-05-27:
There should not be any specific minimum version.

However, we had a bug with versions up to 3.10.6 which could currently result in a crash in the message area code. It has to do with some unknown conditions, probably the active theme being an important one of them. See https://github.com/blueman-project/blueman/blob/master/Dependencies.md for the note about this and #42 for a discussion of the issue.

If you see the crash in RHEL, I think it could be worked around by changing some stuff in our MessageArea code, but we did not investigate that further.

leigh123linux, 2015-05-27:
FYI RHEL7 is currently being rebased to gtk-3.14 and bluez5

https://bugzilla.redhat.com/show_bug.cgi?id=1174373

cschramm, 2015-05-27:
Then I'd suggest trying with the current GTK+ 3.8 and if you encounter the bug wait for the rebase to be finished and depend on 3.14.

infirit, 2016-09-28:
Simple python-caja extension that launches ``blueman-sendto``.

```python
from gi.repository import Caja, GObject, Gio
import os

class BluemanSendtoExtension(GObject.GObject, Caja.MenuProvider):
    def __init__(self):
        pass

    def on_menu_activate(self, menu, files):
        send_files = []

        for f in files:
            if f.is_directory():
                print("Skipping directory")
                continue
            elif f.is_gone():
                print("Skipping none existing file")
                continue

            gfile = f.get_location()
            send_files.append("\"%s\"" % gfile.get_path())

        if len(send_files) == 0:
            return
        else:
            cmd = "blueman-sendto %s" % " ".join(send_files)
            print("Running: %s" % cmd)
            os.popen(cmd)

    def get_file_items(self, window, files):
        if len(files) == 0:
            return

        # We do not support sending whole directories
        for f in files:
            if f.is_directory():
                return

        item = Caja.MenuItem(
            name='BluemanSendto::blueman_send_files',
            label='Send files over bluetooth',
            tip='Sends files over bluetooth with blueman',
            icon='blueman'
        )

        item.connect('activate', self.on_menu_activate, files)

        return [item]
```

infirit, 2016-10-07:
Merged da73795caa1d3946ce9a566553b674af64c82ed4, closing.

closed 2016-10-07 by infirit