blueman-project/blueman #221: D-Bus authentication failed

yars068, 2015-03-14:
I have a git shapshot of blueman since mar-13-2015, running on Slackware64-14.1 with Xfce 4-10, and I use bluez-5.25, when I'd attempt to pair with my LG P-713 phone, I get fail. Here is a part of my ~/.xsession-errors: http://pastebin.com/4VyUJVVy
On the phone, I see notification with generated code to pair with, but this notification closed immediately.
IMO this is not a problem in Blueman, because I tried pairing using bluetoothctl as root, with same authentication problem. Here is output of dbus-monitor: http://pastebin.com/xZ3r9rsp
UPD: Installing Networkmanager 1.0 instead of stock 0.9.8.8 and Bluez-5.29 is a solution for me, for this problem with pairing.

cschramm, 2015-03-27:
> Networkmanager 1.0 instead of stock 0.9.8.8 and Bluez-5.29

Can you say which of those updates solves the issue? They should be independent of each other.

cschramm, 2015-03-27:
Maybe same as https://github.com/blueman-project/blueman/issues/216#issuecomment-77686095

LordBlick, 2015-03-29:
My console output from blueman-applet contains warning:
** (blueman-applet:1561): WARNING **: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
All I use seems to work, but there was need to install gnome-settings-daemon, so xfce users needs install some additional software, that ussualy is no relatedt with that window enviroment.

cschramm, 2015-03-29:
@LordBlick

I don't see the connection to this issue. Are you reporting a separate one?

org.a11y.Bus is provided by AT-SPI and part of the [Gnome A11y](https://wiki.gnome.org/Accessibility) stack. At least in Debian the tasks for Xfce, MATE, and LXDE all include the screen reader gnome-orca and thus AT-SPI. I cannot see a connection to gnome-settings-daemon, but AT-SPI is probably somehow a dependency of that in your system.

Anyway, if you get the warning I think you do not have AT-SPI installed while you have some (GTK+) component that is trying to use it, which is probably a misconfiguration. blueman does not directly work with it, so the warning should be meaningless and not cause any issues. What do you mean with "there was need to install gnome-settings-daemon"?

closed 2015-04-12 by cschramm