blueman-project/blueman #223: searching devices does not find everything

kemnade-uni, 2015-03-18:
When I type 
hcitool lescan
I see some devices

in blueman I cannot find them when I click on search
I am using a debian jessie system.

cschramm, 2015-03-27:
> I am using a debian jessie system.

So you have blueman 1.99.alpha1 and BlueZ 5.23 I suppose.

Do you have multiple bluetooth adapters? What's the result of scanning with bluetoothctl?

kemnade-uni, 2015-03-27:
Am Fri, 27 Mar 2015 04:44:22 -0700
schrieb Christopher Schramm <notifications@github.com>:

> > I am using a debian jessie system.
> 
> So you have blueman 1.99.alpha1 and BlueZ 5.23 I suppose.
> 
yes.
> Do you have multiple bluetooth adapters? What's the result of
> scanning with bluetoothctl?
> 
I can not reproduce it anymore now. Maybe I have blocked some reception
when I was using the mouse in contrast to using the keyboard.

closed 2015-03-27 by cschramm