blueman-project/blueman #225: can't do anything

Viqtak, 2015-03-21:
the only menu I can interact with is the view and help tab.
http://i.imgur.com/AKYnZ9S.png
I am sorry if this issue is already  addressed. would like some guidance since I am very new to linux.
running ubuntu mate 1.8.2

infirit, 2015-03-21:
https://github.com/blueman-project/blueman/wiki/Troubleshooting

Specifically the output of ```bluez-test-adapater list```

Viqtak, 2015-03-21:
seems like I do not have the adaptor. thank you

closed 2015-03-21 by infirit

fainemesquita, 2015-12-27:
I have the same problem, how/where can I get one adaptor?

infirit, 2015-12-27:
Here is a list, https://wiki.ubuntu.com/HardwareSupportComponentsBluetoothUsbAdapters.

As long as it is supported by bluez it should work in blueman.

fainemesquita, 2016-01-04:
No... I can't even find the bluetooth here:

faine@faine-notebook ~ $ lsusb

Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 04f2:b35f Chicony Electronics Co., Ltd
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Dont know what else to do.