blueman-project/blueman #231: add desktop file for blueman-applet

nerijus, 2015-03-28:
Could you please include the blueman-applet.desktop, I suppose then it'll be possible to add applet to Cinnamon, MATE, etc?

cschramm, 2015-03-28:
We sure can include a file with that name, but what should it contain? Please try to describe your actual problem.

nerijus, 2015-03-28:
The problem is that I cannot add applet to, for example, Cinnamon panel by choosing "add to panel". Nor I cannot run blueman-applet from the menu. I have to launch it from terminal and only then it is visible in the traybar.
The file should  probably be smth like this:

[Desktop Entry]
Name=Bluetooth applet
Comment=Bluetooth blueman applet
Exec=blueman-applet
Icon=blueman-device
Terminal=false
Type=Application
Categories=X-XFCE;Settings;HardwareSettings;X-XFCE-SettingsDialog;GTK;
StartupNotify=true
OnlyShowIn=XFCE;MATE;Cinnamon;
NoDisplay=false
X-XfceSettingsName=Bluetooth
X-XfcePluggable=false

cschramm, 2015-03-28:
blueman-applet is not meant to be a panel item, but only shows an icon in the notification area when running. So your problem is that blueman-applet does not start automatically in Cinnamon.

The way it is meant to work (and does at least in Xfce, MATE, and Gnome) is by generating and installing `$(sysconfdir)/xdg/autostart/blueman.desktop` from [`data/blueman.desktop.in`](https://github.com/blueman-project/blueman/blob/master/data/blueman.desktop.in).

Can you verify that file is present on your system? Any blueman related information in `~/.xsession-errors`?

Any other Cinnamon users who can confirm it is (not) starting?

nerijus, 2015-03-28:
Yes, the file is present (sorry I didn't notice it before), and after restarting the Cinnamon session the icon is visible. But it is not visible in MATE, unless I apply the following diff:

--- blueman.desktop.orig	2015-03-28 18:04:17.000000000 +0200
+++ blueman.desktop	2015-03-28 22:00:21.339733677 +0200
@@ -84,4 +84,4 @@
 Type=Application
 Categories=GTK;GNOME;
 X-Desktop-File-Install-Version=0.22
-OnlyShowIn=GNOME;XFCE;LXDE;
+OnlyShowIn=GNOME;XFCE;LXDE;MATE;

cschramm, 2015-03-28:
Huh? OnlyShowIn is not part of [that file](https://github.com/blueman-project/blueman/blob/744890df8740730175d35810d0eb22f1b34d5e6b/data/blueman.desktop.in).

nerijus, 2015-03-28:
But I see https://github.com/blueman-project/blueman/blob/master/data/blueman.desktop.in does not have "OnlyShowIn=" line, it is autogenerated by my rpm spec file (I adapted it from blueman-1.23-7.fc20.src.rpm):

%install
....
desktop-file-install --vendor="" \
 --add-category="GTK;GNOME;" \
 --add-only-show-in="GNOME;XFCE;LXDE" \
 --delete-original \
 --dir=$RPM_BUILD_ROOT%{_sysconfdir}/xdg/autostart/  \
 $RPM_BUILD_ROOT%{_sysconfdir}/xdg/autostart//blueman.desktop

cschramm, 2015-03-28:
That's the choice of the Fedora maintainers then. I guess Cinnamon is not a part of Fedora, so it may make sense for them.

closed 2015-03-28 by cschramm