blueman-project/blueman #232: no "Connect To: Headset" option

nerijus, 2015-03-29:
My bluetooth headset works when paired with a notebook. But it does not when paired with a PC (with another bluetooth adapter). There is no Connect To: Headset option (please see the screeenshots attached). Could it be a bluetooth adapter issue (for example, it does not support audio)? The bluetooth headset is the same in both cases.
[image]
[image]

cschramm, 2015-03-29:
I don't think the hardware plays a roll here.

First, I'd check for version differences, especially blueman, BlueZ, the kernel, and pulseaudio.

Second, are there any errors reported by blueman? Error output from the applet should go to ~/.xsession-errors. For output from blueman-manager you need to start it in a terminal.

nerijus, 2015-03-29:
I want to start blueman-manager, but it does not anymore:
$ blueman-manager 
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
['PulseAudioProfile', 'Services'] 
_________
Load (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:55)
Unable to load plugin module PulseAudioProfile
cannot import name Audio 
_________
__load_plugin (/usr/lib/python2.7/site-packages/blueman/main/PluginManager.py:128)
loading <class 'blueman.plugins.manager.Services.Services'> 
blueman-manager version 1.99.alpha2 starting
There is an instance already running

$ ps axw|grep bluem
does not show any

nerijus, 2015-03-29:
Notebook is Fedora 21 32 bit and runs Cinnamon, PC is F21 64 bit and runs Window Maker. Output from the notebook, where it works:
http://paste.fedoraproject.org/204488/76533101
From PC, where it does not:
http://paste.fedoraproject.org/204489/27653364

cschramm, 2015-03-29:
Your output contained `cannot import name Audio` which is an explanation for the audio services not being available. Do you have `/usr/lib/python2.7/site-packages/blueman/services/Audio.py` on that system?

cschramm, 2015-03-29:
No, sorry, that's unrelated as it appears on both systems. I'll take a closer look at it.

cschramm, 2015-03-29:
There's nothing in the output indicating what's going wrong. If there are also no errors from blueman-applet in ~/.xsession-errors please get the full output by killing and starting it manually in a terminal.

nerijus, 2015-03-30:
Output of blueman-applet on the non working PC:
http://paste.fedoraproject.org/204549/14276743
Please say if you need output from the working setup.

cschramm, 2015-03-30:
OK, no errors there. Then either the UUIDs reported for the device are wrong or blueman's service classes are not loaded.

Please fire up `bluetoothctl` and show me the UUIDs from `info 00:08:E0:51:A8:17`.

nerijus, 2015-03-30:
$ bluetoothctl 
[NEW] Controller 00:13:46:00:12:CB  [default]
[NEW] Device 00:08:E0:51:A8:17 mini-a
[NEW] Device 00:07:E0:33:2D:B4 nerijus
[bluetooth]# info 00:08:E0:51:A8:17
Device 00:08:E0:51:A8:17
	Name: mini-a
	Alias: mini-a
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
[bluetooth]# connect 00:08:E0:51:A8:17
Attempting to connect to 00:08:E0:51:A8:17
[CHG] Device 00:08:E0:51:A8:17 Connected: yes
[mini-a]# info 00:08:E0:51:A8:17
Device 00:08:E0:51:A8:17
	Name: mini-a
	Alias: mini-a
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no

nerijus, 2015-03-30:
And from the working notebook:
$ bluetoothctl 
[NEW] Controller 00:0D:F0:28:6C:54 localhost.localdomain-0 [default]
[NEW] Device 00:0B:0D:94:D6:64 G-Rays2
[NEW] Device 00:08:E0:51:A8:17 mini-a
[NEW] Device 68:EB:AE:34:6A:E9 Emuku
[bluetooth]# info 00:08:E0:51:A8:17
Device 00:08:E0:51:A8:17
	Name: mini-a
	Alias: mini-a
	Class: 0x240404
	Icon: audio-card
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)

nerijus, 2015-03-30:
bluetooth systemd logs from non working PC:
bluetoothd[30849]: Bluetooth daemon 5.29
bluetoothd[30849]: Starting SDP server
bluetoothd[30849]: Bluetooth management interface 1.8 initialized
bluetoothd[30849]: Failed to obtain handles for "Service Changed" characteristic
bluetoothd[30849]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSource
bluetoothd[30849]: Endpoint registered: sender=:1.43 path=/MediaEndpoint/A2DPSink

From working notebook:
bluetoothd[1394]: Bluetooth daemon 5.29
bluetoothd[1394]: Starting SDP server
bluetoothd[1394]: Bluetooth management interface 1.8 initialized
bluetoothd[1394]: Failed to obtain handles for "Service Changed" characteristic
bluetoothd[1394]: Endpoint registered: sender=:1.29 path=/MediaEndpoint/A2DPSource
bluetoothd[1394]: Endpoint registered: sender=:1.29 path=/MediaEndpoint/A2DPSink

nerijus, 2015-03-30:
The USB stick in PC is D-Link Corp. DBT-122 Bluetooth adapter, which supports Bluetooth 2.0 Class 2 only (also see Specification here - http://www.dlink.com/uk/en/support/product/dbt-122-wireless-usb-bluetooth-adapter ). Is it possible that my bluetooth headset requires a later standard?

nerijus, 2015-03-30:
I stopped bluetooth service and copied /var/lib/bluetooth/00:13:46:00:12:CB/00:08:E0:51:A8:17/info and /var/lib/bluetooth/00:13:46:00:12:CB/cache/00:08:E0:51:A8:17 files to PC from the notebook (as the ones in PC were less complete). After starting bluetooth service bluetoothctl 'info' command shows UUIDs, blueman-manager tries to connect, it shows some upload and download data going, but after a while:
Connection Failed: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

nerijus, 2015-03-31:
I connected DBT-122 Bluetooth adapter to the notebook, but it does not show UUIDs for headset here too. So I'm thinking that adapter is too old for headset.

cschramm, 2015-03-31:
Yes, for some reason it looks like the devices cannot talk to each other at all. Beside the UUIDs the adapter also does not get the device class attribute. I've never seen such a thing, but it's clearly not a blueman issue. If there is something going wrong that can be fixed, which I doubt, then it's on the BlueZ level. If you want to get more information on that level you can start bluetoothd manually and provide the -d parameter for debug output.

One thing we could possibly do in blueman is detect such a condition and indicate to the user that something is wrong. I think the missing device class and not a single service UUID are both pretty good indicators. Closing this issue for now, but I'm open to discussing such a handling for blueman.

closed 2015-03-31 by cschramm

Plaque-fcc, 2015-04-12:
Eh, no option to connect to Headset/Audiosink.

The device alias is «Serious»; here's the output:
http://fpaste.org/210112/

cschramm, 2015-04-13:
@Plaque-fcc Please try [the handsfree branch](https://github.com/blueman-project/blueman/tree/handsfree).

infirit, 2015-04-22:
@Plaque-fcc did you get a chance to test the patch @cschramm posted :smile:?