blueman-project/blueman #234: JBL Clip Bluetooth Connection

JR51, 2015-04-01:
I am unable to connect my speaker using Linux Mint 17.1.  When requesting A2DP Sink I always get "Device Added but Failed to Connect"   Help!

cschramm, 2015-04-01:
I think you're experiencing a known issue in blueman 1.23-git201403102151 which is fixed in 1.23-git201406261335-deb. You can [get the later version from utopic](http://packages.ubuntu.com/utopic/blueman). Make sure to restart your system to see the effect.

JR51, 2015-04-02:
I followed the instructions in cschramm's link to install the 20140626 version, but no joy.
Software manager shows this as the installed version.  Version: 1.23-git201406261335-deb-1ubuntu1

JR51, 2015-04-04:
> Unfortunate. Then please follow the guidelines in our Wiki to gather the
> output from blueman-applet while trying to connect to the audio service, so
> we can have a look at it.

Here's the troubleshooting data - hope I got it right.
/John
pineywoods@pineywoods ~ $ killall blueman-applet
blueman-applet: no process found
pineywoods@pineywoods ~ $ blueman-applet
Loading configuration plugins
blueman-applet version 1.23 starting
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['AuthAgent', 'KillSwitch', 'SerialManager', 'ShowConnected', 'NetUsage',
'PulseAudio', 'DiscvManager', 'StatusIcon', 'NMPANSupport',
'TransferService', 'Menu', 'PowerManager', 'DhcpClient', 'PPPSupport',
'ExitItem', 'DBusService', 'StandardItems', 'Headset', 'AppIndicator',
'RecentConns', 'NMDUNSupport', 'Networking']
Using gconf config backend
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'>
Using gconf config backend
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'>
Using gconf config backend
_________
load_nap_settings
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
Using gconf config backend
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
Not loading PPPSupport because it's conflict has higher priority
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PulseAudio.PulseAudio'>
Using gconf config backend
_________
pa_context_event
(/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:306)
1
_________
on_load
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PulseAudio.py:160)
PulseAudio version: [4, 0, 0]
_________
__load_plugin
[... 125 lines cut ...]
(/usr/lib/python2.7/dist-packages/blueman/main/PulseAudioUtils.py:318)
1
_________
on_adapter_property_changed
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:87)
prop UUIDs
dbus.Array([dbus.String(u'00001000-0000-1000-8000-00805f9b34fb'),
dbus.String(u'00001001-0000-1000-8000-00805f9b34fb'),
dbus.String(u'0000112d-0000-1000-8000-00805f9b34fb'),
dbus.String(u'00001112-0000-1000-8000-00805f9b34fb'),
dbus.String(u'0000111f-0000-1000-8000-00805f9b34fb'),
dbus.String(u'0000111e-0000-1000-8000-00805f9b34fb'),
dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'),
dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'),
dbus.String(u'00001105-0000-1000-8000-00805f9b34fb'),
dbus.String(u'00001106-0000-1000-8000-00805f9b34fb')],
signature=dbus.Signature('s'), variant_level=1)
^CTraceback (most recent call last):
  File "/usr/bin/blueman-applet", line 106, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 58, in __init__
    gtk.main()
KeyboardInterrupt
pineywoods@pineywoods ~ $
<http://mirrors.kernel.org/ubuntu/pool/universe/b/blueman/blueman_1.23-git201406261335-deb-1ubuntu1_i386.deb>

cschramm, 2015-04-04:
This looks like the startup only. Did you actually open manager and try to connect the audio service?

JR51, 2015-04-04:
So I started blueman, then tried to connect to A2DP, got the failed to
connect message, then ran blueman-applet and as expected got
pineywoods@pineywoods ~ $ blueman-applet
Loading configuration plugins
blueman-applet version 1.23 starting
There is an instance already running
So I probably didn't understand your last message.
/John

cschramm, 2015-04-04:
You need to kill the running applet and start it again in a terminal. When it's running, use it to open the device manager and try to connect to the audio service. This will produce more output and that is portion that's important to see what's happening.

JR51, 2015-04-04:
Maybe this has it.... I ran the applet from the terminal, then opened
Blueman from the settings menu and tried to connect to A2DP

pineywoods@pineywoods ~ $ killall blueman-applet
pineywoods@pineywoods ~ $ blueman-applet
Loading configuration plugins
blueman-applet version 1.23 starting
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['AuthAgent', 'KillSwitch', 'SerialManager', 'ShowConnected', 'NetUsage',
'PulseAudio', 'DiscvManager', 'StatusIcon', 'NMPANSupport',
'TransferService', 'Menu', 'PowerManager', 'DhcpClient', 'PPPSupport',
'ExitItem', 'DBusService', 'StandardItems', 'Headset', 'AppIndicator',
'RecentConns', 'NMDUNSupport', 'Networking']
Using gconf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'>
Using gconf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'>
Using gconf config backend
_________
load_nap_settings (/usr/lib/python2.7/dist-packages/blueman/plugins/
applet/Networking.py:30)
Loading NAP settings
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
Using gconf config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'>
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'>
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
Not loading PPPSupport because it's conflict has higher priority
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
PluginManager.py:127)
loading <class 'blueman.plugins.applet.PulseAudio.PulseAudio'>
Using gconf config backend
_________
pa_context_event (/usr/lib/python2.7/dist-packages/blueman/main/
PulseAudioUtils.py:306)
1
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/
applet/PulseAudio.py:160)
PulseAudio version: [4, 0, 0]
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/
[... 151 lines cut ...]
caching initial properties
_________
notify (/usr/lib/python2.7/dist-packages/blueman/plugins/
applet/RecentConns.py:237)
<Device object at 0xa8d220cc (blueman+main+Device+Device at 0x9f8ac00)>
org.bluez.AudioSink dbus.Array([], signature=dbus.Signature('s'))
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/756/hci0/dev_00_12_6F_62_9A_6F
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/756/hci0/dev_00_12_6F_62_9A_6F
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/
applet/RecentConns.py:136)
rebuilding menu
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:60)
deleting device /org/bluez/756/hci0/dev_00_12_6F_62_9A_6F
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/756/hci0/dev_00_12_6F_62_9A_6F

cschramm, 2015-04-05:
The device just disconnects when you try to use the service. Please start blueman-manager from a terminal to see if it gives any more information when trying to disconnect.

You could also try 1.99.alpha1 to see if it works in a more recent version: http://packages.ubuntu.com/vivid/blueman

JR51, 2015-04-05:
Christopher,
Good news is my JBL Click is now working.   Blueman Manager still gives me
the fail to connect message when I select A2DP. (In windows the JBL didn't
work when I selected the headphone option.)  I don't have an explanation
for the change but I was poking around in the pulse-audio menus and may
have turned something on that had been off previously.   Also, I'd never
started blueman-manager from the terminal before, don't know why that would
have made a difference.   Anyway it now connects automatically when I turn
it on, so all is good.  Thanks /John

cschramm, 2015-04-05:
:smiley: 

> I'd never started blueman-manager from the terminal before, don't know why that would have made a difference.

Shouldn't. That's just for getting its output.

closed 2015-04-05 by cschramm