blueman-project/blueman #235: Auto-Conecting  Debian Wheezy/Jessie

rsm-imap-cc, 2015-04-03:
I'm not sure if this is a bug or not but I wanted to ask the question...

I use this program to connect my speakers. On my system blueman is started by default so it is always running..
When I had Debian Wheezy if I turned on the speakers the connection was made automatically. On Debian Jessie I have to do:

1) Turn on the speaker and set it in paring mode
2) Select the speaker over blueman and set a connection

The speakers are marked as trusted device.

Is there any way of making an auto-conection as in Debian Wheezy?

Thanks!!

cschramm, 2015-04-03:
You should at least not have to re-pair the device. If you start your system and open blueman, the speaker is not included in the device list and you have to discover it first?

rsm-imap-cc, 2015-04-03:
The device is in the device list. But i think that the problem is that it doesn't detects that the device is turned on.
When I try to connect it says "Connection Failed: Host is down", Then if I set my speaker in paring mode and I connect, everything works good

cschramm, 2015-04-03:
Unfortunately that sounds like the change is actually in BlueZ and not in blueman. You could try downgrading to wheezy's bluez package and see what happens.

rsm-imap-cc, 2015-04-03:
I just found the half of the fix, the problem is that I do not have a "Public" directory.

line 57 of /usr/lib/python2.7/dist-packages/blueman/main/applet/Transfer.py

I sujest to create the directory in case if it doesn't exists. Do you want me to submit the patch?

This only fixes the fact that I can conect without using the paring button. The auto-conect stils not working!

cschramm, 2015-04-03:
OK, so 1) :white_check_mark:.

Concerning 2) I'd actually expect that behavior. An auto-connection must have been initialized either by the device, which should still work if that's the case, or by some component on your system - especially BlueZ itself or pulseaudio.

I think the best way to find out what changed would be to downgrade blueman, bluez, and pulseaudio to the wheezy versions, check that auto-connecting works, and then upgrade one by one to see which change breaks it. Note that bluez should be upgraded last.

rsm-imap-cc, 2015-04-03:
Good idea!, I'll check that in a live version, I really miss the auto-connect feature

rsm-imap-cc, 2015-04-03:
The auto-connection problem is being caused by bluez, if I downgrade everthing works as before :)

closed 2015-04-03 by rsm-imap-cc

cschramm, 2015-04-04:
Good to know, although unsatisfying for the users... :disappointed: 

As auto-connecting comes up once in a while, this may be a future feature of blueman, but most likely not in 2.0.

jod35fan, 2015-04-06:
Bluez5 can be configured in /etc/bluetooth/main.conf. I Can send pictures from phone without reparing with latest blueman from sources.

[General]

# Default adaper name
# %h - substituted for hostname
# %d - substituted for adapter id
# Defaults to 'BlueZ'
#Name = %h-%d

# Default device class. Only the major and minor device class bits are
# considered. Defaults to '0x000000'.
#Class = 0x000100

# How long to stay in discoverable mode before going back to non-discoverable
# The value is in seconds. Default is 180, i.e. 3 minutes.
# 0 = disable timer, i.e. stay discoverable forever
DiscoverableTimeout = 0

# How long to stay in pairable mode before going back to non-discoverable
# The value is in seconds. Default is 0.
# 0 = disable timer, i.e. stay pairable forever
PairableTimeout = 0

# Automatic connection for bonded devices driven by platform/user events.
# If a platform plugin uses this mechanism, automatic connections will be
# enabled during the interval defined below. Initially, this feature
# intends to be used to establish connections to ATT channels. Default is 60.
#AutoConnectTimeout = 60

# Use vendor id source (assigner), vendor, product and version information for
# DID profile support. The values are separated by ":" and assigner, VID, PID
# and version.
# Possible vendor id source values: bluetooth, usb (defaults to usb)
#DeviceID = bluetooth:1234:5678:abcd

# Do reverse service discovery for previously unknown devices that connect to
# us. This option is really only needed for qualification since the BITE tester
# doesn't like us doing reverse SDP for some test cases (though there could in
# theory be other useful purposes for this too). Defaults to 'true'.
#ReverseServiceDiscovery = true

# Enable name resolving after inquiry. Set it to 'false' if you don't need
# remote devices name and want shorter discovery cycle. Defaults to 'true'.
#NameResolving = true

# Enable runtime persistency of debug link keys. Default is false which
# makes debug link keys valid only for the duration of the connection
# that they were created for.
#DebugKeys = false

# Restricts all controllers to the specified transport. Default value
# is "dual", i.e. both BR/EDR and LE enabled (when supported by the HW).
# Possible values: "dual", "bredr", "le"
#ControllerMode = dual

# Enables Multi Profile Specification support. This allows to specify if
# system supports only Multiple Profiles Single Device (MPSD) configuration
# or both Multiple Profiles Single Device (MPSD) and Multiple Profiles Multiple
# Devices (MPMD) configurations.
# Possible values: "off", "single", "multiple"
#MultiProfile = off

# Permanently enables the Fast Connectable setting for adapters that
# support it. When enabled other devices can connect faster to us,
# however the tradeoff is increased power consumptions. This feature
# will fully work only on kernel version 4.1 and newer. Defaults to
# 'false'.
#FastConnectable = false

#[Policy]
#
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
#ReconnectUUIDs=