blueman-project/blueman #237: Could blueman be made to work with udhcpc

juanitotc, 2015-04-04:
Just a suggestion, but could blueman be made to work with udhcpc (busybox) for network connections?

juanitotc, 2015-04-14:
I just noticed that there had been some updates to the udhcpc branch so I tried again.

When attempting to set an iPhone as a network access point, I get the error "Connection Failed: The name org.freedesktop.NetworkManager was not provided by any .service files:

..but if I'm trying to use udhcpc, I shouldn't need networkmanager?

cschramm, 2015-04-14:
I just rebased the branch on master. Still have to test an iPhone myself.

Are you sure you have enabled the DhcpClient plugin (and thus disabled the NM integration)?

juanitotc, 2015-04-14:
Ah, forgot that.

So, cutting and pasting the following with dconf-editor into org.blueman.general plugin-list:

`['TransferService', 'StatusIcon', 'StandardItems', 'ShowConnected', 'SerialManager', 'RecentConns', 'PowerManager', 'PPPSupport', 'Networking', 'NetUsage', '!NMPANSupport', '!NMDUNSupport', 'Menu', 'KillSwitch', 'Headset', 'GameControllerWakelock', 'ExitItem', 'DiscvManager', 'DhcpClient', 'DBusService', 'AuthAgent', 'AppIndicator']`

it works :)

Not to appear ungrateful, but ref:

`$ ps aux | grep udhcpc
 3163 root     /sbin/udhcpc -b -i eth0 -x hostname boxdell -p /var/run/udhcpc.eth0.pid
 7737 root     /sbin/udhcpc -t 20 -n -i bnep0`

..would it be possible to pick up the hostname and create a pid file?

BTW, where is the blueman "plugin manager" meant to appear in the gnome desktop environment (I don't see it anywhere)?

juanitotc, 2015-04-14:
http://forum.tinycorelinux.net/index.php/topic,18121.msg111250.html#msg111250

cschramm, 2015-04-14:
Great. No idea why the connection works for you now (and I hope it will last...), but we can definitely merge udhcpc then.

> would it be possible to pick up the hostname and create a pid file?

Hostname seems to make sense. I'll add that to dhclient and dnsmasq as well if possible.

I don't think a pid file makes sense as blueman manages the process (if it does not terminate itself it should be killed after a timeout).

> where is the blueman "plugin manager" meant to appear in the gnome desktop environment (I don't see it anywhere)?

It's in the menu of our applet. I guess it's "Extensions" in English.

> http://forum.tinycorelinux.net/index.php/topic,18121.msg111250.html#msg111250

Thanks for the "nice man" title. Now I finally know what strange system you are working with there. :wink:

cschramm, 2015-04-16:
#258 implements the hostname argument

commit 62e5c4ad (2015-04-07): Implement udhcpc support

Refactor blueman.main.DhcpClient to avoid duplication. Adding udhcpc support is actually done by just adding it to the COMMANDS array.

Closes #237
commit 32dea24e (2015-04-28): Send hostname when using udhcpc

Closes #237
Closes #258
commit 2a66ed4d (2015-04-28): Send hostname when using udhcpc

Closes #237
Closes #258
closed 2015-04-28 by cschramm