blueman-project/blueman #239: Auto-connect services

cschramm, 2015-04-04:
Once in a while users want to have an auto-connect feature where a known service is connected automatically blueman is started or the device becomes available. @rsm-imap-cc figured out in #235 there are cases where BlueZ 4 does such an auto-connect, but BlueZ 5 does not, so this becomes more important with the migration.

jod35fan, 2015-04-06:
...and this Bluez5 migration takes years, if  Debian responsbles do not code and package anything. Meanwhile debian 8 users must use kdevil, where in xfce you must click 3 times button in 3 panels before it works. I think it is a waste of time to average user to have current blueman and bluez5 packages  in testing repositories. Tune blueman a bit and package the latest bluez5 with minor modifications that i have documented. Debian Jessie is otherwise very good, but this  bluetooth problem is not.

cschramm, 2018-03-21:
A draft for the "backend" pieces is in #853.

I do not have a concept for the GUI pieces yet. Some ideas:

* An option to enable auto-connect for the selected service in the device setup wizard. A list with removable auto-connect entries in the plugin settings.
* Entries to toggle auto-connect for specific services in the device menu, e.g. submenus instead of the immediate connect / disconnect items that contain the actual connect / disconnect item and an additional enable / disable auto-connect item.
* Auto-connect checkboxes next to specific services listed in an extended version of the "info" view.

bth73, 2019-01-25:
Hi, I'm hoping there's a fix for the auto connect feature. It's the last thing I need fixed  to have a perfect Asus T100 linux mint 19.1 install. My install is a few weeks old. I tried altering main bluetooth config file, 
# AutoEnable defines option to enable all controllers when they are found.
# This includes adapters present on start as well as adapters that are plugged
# in later on. Defaults to 'false'.
AutoEnable=true

The last line was already uncommented. Then I tried this:
# The ReconnectUUIDs defines the set of remote services that should try
# to be reconnected to in case of a link loss (link supervision
# timeout). The policy plugin should contain a sane set of values by
# default, but this list can be overridden here. By setting the list to
# empty the reconnection feature gets disabled.
ReconnectUUIDs=00001112-0000-1000-8000-00805f9b34fb, 0000111f-0000-1000-8000-00805f9b34fb, 0000110a-0000-1000-8000-00805f9b34fb
Removed the comment from last line.
No change with either.
I'm sort of a middle weight, so need complete instructions if there's a workaround or fix you can lay on me. Good at reading, following directions,copy and pasting.

cschramm, 2019-03-30:
@bth73: You could write a script that calls `bluetoothctl connect` and e.g. trigger it with a udev rule.

closed 2021-04-14 by cschramm