blueman-project/blueman #240: Can't connect or disconnect on Fedora 21

bjrosen, 2015-04-05:
Here are the devices on my system according to bluetoothctl
bluetoothctl 
[NEW] Controller 5C:F3:70:65:BA:9F nimitz [default]
[NEW] Device CC:C5:0A:66:B9:F3 HY-SWT
[NEW] Device 00:0C:8A:73:D4:20 Bose AE2w 01.02.00
[NEW] Device 40:EF:4C:87:C0:FA Grace Digital
I can connect and disconnect with bluetoothctrl but not blueman. Any attempt to connect or disconnect gets
Connection Failed: DBusFailedError: Input/output error...

cschramm, 2015-04-05:
What service do you try to connect?

bjrosen, 2015-04-05:
Audio SInk

cschramm, 2015-04-06:
So issuing a `connect` in bluetoothctl work while clicking on the Audio Sink service in blueman does not?

What blueman version is it?

bjrosen, 2015-04-06:
Yes bluetoothctl works and blueman does not. The version is 1.99.alpha2

cschramm, 2015-04-08:
Connecting an audio service just uses the BlueZ API to issue the same as you do with `connect` in `bluetoothctl`, so this gets a little tricky to debug. Please check your syslog for messages from bluetoothd (probably in /var/log/daemon.log) when trying to connect with blueman.

bjrosen, 2015-04-08:
There was an bluez update yesterday, now I'm able to connect to devices but not disconnect. I can disconnect with bluetoothctl.

Here are the messages from /var/log/messages

Apr  8 07:13:07 nimitz org.freedesktop.Notifications: (xfce4-notifyd:642): Gtk-WARNING **: Failed to set text from markup due to error parsing markup: Error on line 1: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity - escape ampersand as &amp;
Apr  8 07:13:16 nimitz org.blueman.Applet: ** (blueman-manager:645): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:2:19: Theming engine 'adwaita' not found
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:3:14: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:83:14: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:83:16: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:103:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:104:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:122:14: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:128:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:141:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:142:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:181:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:182:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:228:16: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:228:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:246:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:247:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:289:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:291:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:333:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:334:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:354:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:355:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:370:14: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:378:19: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:379:20: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:389:14: Not using units is deprecated. Assuming 'px'.
Apr  8 07:13:16 nimitz org.blueman.Applet: (blueman-manager:645): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:389:16: Not using units is deprecated. Assuming 'px'.
[... cut ...]
Apr  8 07:24:23 nimitz org.blueman.Applet: return getattr(self.Device, name)
Apr  8 07:24:23 nimitz org.blueman.Applet: AttributeError: 'Device' object has no attribute 'Disconnect'
Apr  8 07:24:53 nimitz dbus[833]: [system] Rejected send message, 11 matched rules; type="method_return", sender=":1.55" (uid=500 pid=1524 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.546" (uid=0 pid=29792 comm="/usr/libexec/bluetooth/bluetoothd ")

bjrosen, 2015-04-08:
BTW I how have version 1.99.alpha2 of Blueman running.

cschramm, 2015-04-08:
@cschramm Need to get started fixing all those ugly GTK warnings...

@bjrosen That's an error in the generic disconnect method. I just fixed it in #248. Can you give it a try?

bjrosen, 2015-04-08:
It;s not in the repo yet. I'm using the heikoda blueman git repo. Is there a different repo that you want me to use?

cschramm, 2015-04-08:
> I'm using the heikoda blueman git repo.

What's that?

> Is there a different repo that you want me to use?

Well, https://github.com/blueman-project/blueman ? The fix is in branch disconnect-fix there.

bjrosen, 2015-04-08:
There doesn't seem to be a Fedora package on that page. I've been using this repo
https://copr-be.cloud.fedoraproject.org/results/heikoada/blueman-git/fedora-21-x86_64/

cschramm, 2015-04-08:
I see. That repository seems to produce daily builds of our master branch. The fix will be there as soon as I merge and close #248 which I'll do within the next two days.

closed 2015-04-09 by cschramm

cschramm, 2015-04-09:
It's in master since two hours now, but it looks like that repository has a bug so that it will not build it until the next change.

cschramm, 2015-04-09:
Ah, now it's built. :smiley:

bjrosen, 2015-04-09:
This version just showed up on Fedora, disconnect still doesn't work
blueman-1.99.alpha2-201504090731gitc75112c.fc21.x86_64.rpm

Apr  9 09:00:18 nimitz org.blueman.Applet: ** (blueman-manager:21477): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:2:19: Theming engine 'adwaita' not found
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:3:14: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:83:14: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:83:16: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:103:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:104:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:122:14: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:128:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:141:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:142:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:181:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:182:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:228:16: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:228:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:246:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:247:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:289:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:291:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:333:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:334:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:354:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:355:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:370:14: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:378:19: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:379:20: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:389:14: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:389:16: Not using units is deprecated. Assuming 'px'.
Apr  9 09:00:18 nimitz org.blueman.Applet: (blueman-manager:21477): Gtk-WARNING **: Theme parsing error: gtk-widgets.css:411:20: Not using units is deprecated. Assuming 'px'.
[... cut ...]
Apr  9 09:01:01 nimitz systemd: Stopped target Paths.
Apr  9 09:01:01 nimitz systemd: Stopping Timers.
Apr  9 09:01:01 nimitz systemd: Stopped target Timers.
Apr  9 09:01:01 nimitz systemd: Stopping Sockets.
Apr  9 09:01:01 nimitz systemd: Stopped target Sockets.
Apr  9 09:01:01 nimitz systemd: Starting Shutdown.
Apr  9 09:01:01 nimitz systemd: Reached target Shutdown.
Apr  9 09:01:01 nimitz systemd: Starting Exit the Session...
Apr  9 09:01:01 nimitz systemd: Received SIGRTMIN+24 from PID 21527 (kill).

cschramm, 2015-04-09:
201504091235gite5da7b4 is the one you need

bjrosen, 2015-04-09:
Hopefully that one will show up tomorrow.

bjrosen, 2015-04-12:
Latest version can't connect or disconnect,
1.99.alpha2-201504121418gitf214a05.fc21

Connection Failed: Connection ":1.169" is not allowed to add more match 

Apr 12 12:02:14 nimitz dbus[833]: [system] Rejected send message, 11 matched rules; type="method_return", sender=":1.664" (uid=500 pid=1607 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.1098" (uid=0 pid=31643 comm="/usr/libexec/bluetooth/bluetoothd ")
Apr 12 12:02:17 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 12 12:02:17 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 12 12:02:17 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 12 12:02:17 nimitz org.blueman.Applet: handler(name, value, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 29, in on_device_property_changed
Apr 12 12:02:17 nimitz org.blueman.Applet: klass = Device(path).get_properties()["Class"] & 0x1fff
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 12 12:02:17 nimitz org.blueman.Applet: "PropertyChanged")
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 12 12:02:17 nimitz org.blueman.Applet: obj.handle_signal(*args, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
Apr 12 12:02:17 nimitz org.blueman.Applet: self._handle_signal(wrapper, 'PropertiesChanged', interface, self.get_object_path(), **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 33, in _handle_signal
Apr 12 12:02:17 nimitz org.blueman.Applet: self.get_bus().add_signal_receiver(handler, signal, interface, 'org.bluez', obj_path, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 157, in add_signal_receiver
Apr 12 12:02:17 nimitz org.blueman.Applet: watch = self.watch_name_owner(bus_name, callback)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 373, in watch_name_owner
Apr 12 12:02:17 nimitz org.blueman.Applet: return NameOwnerWatch(self, bus_name, callback)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 80, in __init__
Apr 12 12:02:17 nimitz org.blueman.Applet: arg0=bus_name)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 160, in add_signal_receiver
Apr 12 12:02:17 nimitz org.blueman.Applet: self.add_match_string(str(match))
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 398, in add_match_string
Apr 12 12:02:17 nimitz org.blueman.Applet: BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 12 12:02:17 nimitz org.blueman.Applet: message, timeout)
Apr 12 12:02:17 nimitz org.blueman.Applet: DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.169" is not allowed to add more match rules (increase limits in configuration file if required)
Apr 12 12:02:17 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 12 12:02:17 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 12 12:02:17 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 12 12:02:17 nimitz org.blueman.Applet: handler(name, value, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 52, in on_device_property_changed
Apr 12 12:02:17 nimitz org.blueman.Applet: d = Device(path)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 12 12:02:17 nimitz org.blueman.Applet: "PropertyChanged")
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 12 12:02:17 nimitz org.blueman.Applet: obj.handle_signal(*args, **kwargs)
Apr 12 12:02:17 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
Apr 12 12:02:17 nimitz org.blueman.Applet: self._handle_signal(wrapper, 'PropertiesChanged', interface, self.get_object_path(), **kwargs)
[... cut ...]
Apr 12 12:02:24 nimitz org.blueman.Applet: (blueman-manager:32018): Gtk-WARNING **: Theme parsing error: gnome-applications.css:110:19: Not using units is deprecated. Assuming 'px'.
Apr 12 12:02:24 nimitz org.blueman.Applet: (blueman-manager:32018): Gtk-WARNING **: Theme parsing error: gnome-applications.css:110:21: Not using units is deprecated. Assuming 'px'.
Apr 12 12:02:32 nimitz org.blueman.Applet: /usr/lib64/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 965 was not found when attempting to remove it
Apr 12 12:02:32 nimitz org.blueman.Applet: return fn(*args, **kwargs)

bjrosen, 2015-04-12:
Here is the log for bluetoothctl connect which works
Apr 12 12:04:11 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 12 12:04:11 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 12 12:04:11 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 12 12:04:11 nimitz org.blueman.Applet: handler(name, value, **kwargs)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 29, in on_device_property_changed
Apr 12 12:04:11 nimitz org.blueman.Applet: klass = Device(path).get_properties()["Class"] & 0x1fff
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 12 12:04:11 nimitz org.blueman.Applet: "PropertyChanged")
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 12 12:04:11 nimitz org.blueman.Applet: obj.handle_signal(*args, **kwargs)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
Apr 12 12:04:11 nimitz org.blueman.Applet: self._handle_signal(wrapper, 'PropertiesChanged', interface, self.get_object_path(), **kwargs)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 33, in _handle_signal
Apr 12 12:04:11 nimitz org.blueman.Applet: self.get_bus().add_signal_receiver(handler, signal, interface, 'org.bluez', obj_path, **kwargs)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 157, in add_signal_receiver
Apr 12 12:04:11 nimitz org.blueman.Applet: watch = self.watch_name_owner(bus_name, callback)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 373, in watch_name_owner
Apr 12 12:04:11 nimitz org.blueman.Applet: return NameOwnerWatch(self, bus_name, callback)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 80, in __init__
Apr 12 12:04:11 nimitz org.blueman.Applet: arg0=bus_name)
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 160, in add_signal_receiver
Apr 12 12:04:11 nimitz org.blueman.Applet: self.add_match_string(str(match))
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 398, in add_match_string
Apr 12 12:04:11 nimitz org.blueman.Applet: BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
Apr 12 12:04:11 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 12 12:04:11 nimitz org.blueman.Applet: message, timeout)
Apr 12 12:04:11 nimitz org.blueman.Applet: DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.169" is not allowed to add more match rules (increase limits in configuration file if required)
Apr 12 12:04:11 nimitz pulseaudio[1607]: [pulseaudio] module-bluez5-device.c: Profile has no transport
Apr 12 12:04:11 nimitz bluetoothd[31643]: /org/bluez/hci0/dev_CC_C5_0A_66_B9_F3/fd2: fd(25) ready
Apr 12 12:04:11 nimitz rtkit-daemon[786]: Successfully made thread 32063 of process 1607 (/usr/bin/pulseaudio (deleted)) owned by '500' RT at priority 5.
Apr 12 12:04:11 nimitz kernel: [1002349.163299] input: CC:C5:0A:66:B9:F3 as /devices/virtual/input/input43
Apr 12 12:04:11 nimitz kernel: input: CC:C5:0A:66:B9:F3 as /devices/virtual/input/input43

bjrosen, 2015-04-12:
Apr 12 12:07:10 nimitz dbus[833]: [system] Rejected send message, 11 matched rules; type="method_return", sender=":1.664" (uid=500 pid=1607 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.1098" (uid=0 pid=31643 comm="/usr/libexec/bluetooth/bluetoothd ")
Apr 12 12:07:12 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 12 12:07:12 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 12 12:07:12 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 12 12:07:12 nimitz org.blueman.Applet: handler(name, value, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 29, in on_device_property_changed
Apr 12 12:07:12 nimitz org.blueman.Applet: klass = Device(path).get_properties()["Class"] & 0x1fff
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 12 12:07:12 nimitz org.blueman.Applet: "PropertyChanged")
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 12 12:07:12 nimitz org.blueman.Applet: obj.handle_signal(*args, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
Apr 12 12:07:12 nimitz org.blueman.Applet: self._handle_signal(wrapper, 'PropertiesChanged', interface, self.get_object_path(), **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 33, in _handle_signal
Apr 12 12:07:12 nimitz org.blueman.Applet: self.get_bus().add_signal_receiver(handler, signal, interface, 'org.bluez', obj_path, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 157, in add_signal_receiver
Apr 12 12:07:12 nimitz org.blueman.Applet: watch = self.watch_name_owner(bus_name, callback)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 373, in watch_name_owner
Apr 12 12:07:12 nimitz org.blueman.Applet: return NameOwnerWatch(self, bus_name, callback)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 80, in __init__
Apr 12 12:07:12 nimitz org.blueman.Applet: arg0=bus_name)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 160, in add_signal_receiver
Apr 12 12:07:12 nimitz org.blueman.Applet: self.add_match_string(str(match))
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 398, in add_match_string
Apr 12 12:07:12 nimitz org.blueman.Applet: BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 12 12:07:12 nimitz org.blueman.Applet: message, timeout)
Apr 12 12:07:12 nimitz org.blueman.Applet: DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.169" is not allowed to add more match rules (increase limits in configuration file if required)
Apr 12 12:07:12 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 12 12:07:12 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 12 12:07:12 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 12 12:07:12 nimitz org.blueman.Applet: handler(name, value, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 52, in on_device_property_changed
Apr 12 12:07:12 nimitz org.blueman.Applet: d = Device(path)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 12 12:07:12 nimitz org.blueman.Applet: "PropertyChanged")
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 12 12:07:12 nimitz org.blueman.Applet: obj.handle_signal(*args, **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
Apr 12 12:07:12 nimitz org.blueman.Applet: self._handle_signal(wrapper, 'PropertiesChanged', interface, self.get_object_path(), **kwargs)
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 33, in _handle_signal
Apr 12 12:07:12 nimitz org.blueman.Applet: self.get_bus().add_signal_receiver(handler, signal, interface, 'org.bluez', obj_path, **kwargs)
[... cut ...]
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 398, in add_match_string
Apr 12 12:07:12 nimitz org.blueman.Applet: BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
Apr 12 12:07:12 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 12 12:07:12 nimitz org.blueman.Applet: message, timeout)
Apr 12 12:07:12 nimitz org.blueman.Applet: DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.169" is not allowed to add more match rules (increase limits in configuration file if required)

cschramm, 2015-04-13:
That's #189. Unfortunately that's a little hard to debug as I cannot identify which match rules are installed. There's probably some signal handler that gets installed over and over again. Can you give me your max_match_rules_per_connection from `/etc/dbus-1/session.conf`? Maybe I can reproduce this by decreasing mine. @juanitotc Yours is as much interesting of course.

bjrosen, 2015-04-13:
Here is the dbus-1/session.conf

<!DOCTYPE busconfig PUBLIC "-//freedesktop//DTD D-Bus Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>
  
  <type>session</type>

  
  <keep_umask/>

  <listen>unix:tmpdir=/tmp</listen>

  <standard_session_servicedirs />

  <policy context="default">
    
    <allow send_destination="*" eavesdrop="true"/>
    
    <allow eavesdrop="true"/>
    
    <allow own="*"/>
  </policy>

  
  <includedir>session.d</includedir>

  
  <include ignore_missing="yes">session-local.conf</include>

  <include if_selinux_enabled="yes" selinux_root_relative="yes">contexts/dbus_contexts</include>

  

  
  <limit name="max_incoming_bytes">1000000000</limit>
  <limit name="max_outgoing_bytes">1000000000</limit>
  <limit name="max_message_size">1000000000</limit>
  <limit name="service_start_timeout">120000</limit>  
  <limit name="auth_timeout">240000</limit>
  <limit name="max_completed_connections">100000</limit>  
  <limit name="max_incomplete_connections">10000</limit>
  <limit name="max_connections_per_user">100000</limit>
  <limit name="max_pending_service_starts">10000</limit>
  <limit name="max_names_per_connection">50000</limit>
  <limit name="max_match_rules_per_connection">50000</limit>
  <limit name="max_replies_per_connection">50000</limit>

</busconfig>

bjrosen, 2015-04-13:
Installed the latest, can't connect or disconnect
blueman-1.99.alpha2-201504122014git2096961.fc21.x86_64
Connection Failed: Connection ":1.169" is not allowed to add more match rules (increase limits in configuration file if required)

Apr 13 10:00:04 nimitz yum[13645]: Installed: blueman-1.99.alpha2-201504122014git2096961.fc21.x86_64
Apr 13 10:01:01 nimitz systemd: Failed at step CGROUP spawning /usr/lib/systemd/systemd: No such file or directory
Apr 13 10:02:01 nimitz python: detected unhandled Python exception in '/usr/share/yumex/yum_childtask.py'
Apr 13 10:02:01 nimitz dbus[833]: [system] Activating service name='org.freedesktop.problems' (using servicehelper)
Apr 13 10:02:01 nimitz dbus[833]: [system] Successfully activated service 'org.freedesktop.problems'
Apr 13 10:02:38 nimitz dbus[833]: [system] Rejected send message, 11 matched rules; type="method_return", sender=":1.664" (uid=500 pid=1607 comm="/usr/bin/pulseaudio --start --log-target=syslog ") interface="(unset)" member="(unset)" error name="(unset)" requested_reply="0" destination=":1.1098" (uid=0 pid=31643 comm="/usr/libexec/bluetooth/bluetoothd ")
Apr 13 10:02:40 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 13 10:02:40 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 13 10:02:40 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 13 10:02:40 nimitz org.blueman.Applet: else:
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/GameControllerWakelock.py", line 29, in on_device_property_changed
Apr 13 10:02:40 nimitz org.blueman.Applet: self.suspend_screensaver("resume")
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 13 10:02:40 nimitz org.blueman.Applet: objtype = args[0]
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
Apr 13 10:02:40 nimitz org.blueman.Applet: self._handler_wrappers[handler] = wrapper
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 33, in _handle_signal
Apr 13 10:02:40 nimitz org.blueman.Applet: else:
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 157, in add_signal_receiver
Apr 13 10:02:40 nimitz org.blueman.Applet: watch = self.watch_name_owner(bus_name, callback)
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 373, in watch_name_owner
Apr 13 10:02:40 nimitz org.blueman.Applet: return NameOwnerWatch(self, bus_name, callback)
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 80, in __init__
Apr 13 10:02:40 nimitz org.blueman.Applet: arg0=bus_name)
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 160, in add_signal_receiver
Apr 13 10:02:40 nimitz org.blueman.Applet: self.add_match_string(str(match))
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/bus.py", line 398, in add_match_string
Apr 13 10:02:40 nimitz org.blueman.Applet: BUS_DAEMON_IFACE, 'AddMatch', 's', (rule,))
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 651, in call_blocking
Apr 13 10:02:40 nimitz org.blueman.Applet: message, timeout)
Apr 13 10:02:40 nimitz org.blueman.Applet: DBusException: org.freedesktop.DBus.Error.LimitsExceeded: Connection ":1.169" is not allowed to add more match rules (increase limits in configuration file if required)
Apr 13 10:02:40 nimitz org.blueman.Applet: ERROR:dbus.connection:Exception in handler for D-Bus signal:
Apr 13 10:02:40 nimitz org.blueman.Applet: Traceback (most recent call last):
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib64/python2.7/site-packages/dbus/connection.py", line 230, in maybe_handle_message
Apr 13 10:02:40 nimitz org.blueman.Applet: self._handler(*args, **kwargs)
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 51, in wrapper
Apr 13 10:02:40 nimitz org.blueman.Applet: else:
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/plugins/applet/SerialManager.py", line 52, in on_device_property_changed
Apr 13 10:02:40 nimitz org.blueman.Applet: for k in self.scripts.keys():
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/Device.py", line 51, in __init__
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/main/SignalTracker.py", line 36, in Handle
Apr 13 10:02:40 nimitz org.blueman.Applet: objtype = args[0]
Apr 13 10:02:40 nimitz org.blueman.Applet: File "/usr/lib/python2.7/site-packages/blueman/bluez/PropertiesBlueZInterface.py", line 57, in handle_signal
[... cut ...]
Apr 13 10:03:06 nimitz org.blueman.Applet: (blueman-manager:14022): Gtk-WARNING **: Theme parsing error: gnome-applications.css:110:21: Not using units is deprecated. Assuming 'px'.
Apr 13 10:03:11 nimitz org.blueman.Applet: /usr/lib64/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 686 was not found when attempting to remove it
Apr 13 10:03:11 nimitz org.blueman.Applet: return fn(*args, **kwargs)
Apr 13 10:03:48 nimitz org.mate.panel.applet.MateWeatherAppletFactory: (mateweather-applet-2:1806): GLib-CRITICAL **: Source ID 39630 was not found when attempting to remove it

bjrosen, 2015-04-14:
Installed
blueman-1.99.alpha2-201504122014git2096961.fc21.x86_64

Still doesn't connect or disconnect
Apr 13 22:38:45 nimitz org.blueman.Applet: /usr/lib64/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 5896 was not found when attempting to remove it
Apr 13 22:38:45 nimitz org.blueman.Applet: return fn(*args, **kwargs)

juanitotc, 2015-04-14:
Here's the relevant part of /usr/local/etc/dbus-1/session.conf:

  
  <limit name="max_incoming_bytes">1000000000</limit>
  <limit name="max_incoming_unix_fds">250000000</limit>
  <limit name="max_outgoing_bytes">1000000000</limit>
  <limit name="max_outgoing_unix_fds">250000000</limit>
  <limit name="max_message_size">1000000000</limit>
  
  <limit name="service_start_timeout">120000</limit>  
  <limit name="auth_timeout">240000</limit>
  <limit name="pending_fd_timeout">150000</limit>
  <limit name="max_completed_connections">100000</limit>  
  <limit name="max_incomplete_connections">10000</limit>
  <limit name="max_connections_per_user">100000</limit>
  <limit name="max_pending_service_starts">10000</limit>
  <limit name="max_names_per_connection">50000</limit>
  <limit name="max_match_rules_per_connection">50000</limit>
  <limit name="max_replies_per_connection">50000</limit>

</busconfig>

commit e5da7b41 (2015-04-09): Fix generic disconnect method

Closes #240
Closes #248