blueman-project/blueman #241: TV Pairs Incorrectly

juanitotc, 2015-04-06:
Blueman shows a TV (Sony KDL-46W900A) as:

`DTV Bluetooth
Video display and loudspeaker`

When I attempt to pair with the TV, blueman assumes it is an "intelligent" device and sends a pairing pin rather than assuming it is a "dumb" device and sending a pin of "0000". Nothing appears on the TV, but blueman displays the paired icon against the TV icon.

The only bluetooth service shown by blueman is "audio source", but shouldn't it show "audio sink" (loudspeaker) and/or "video sink" (display)?

When I try to set the TV as an audio source I get:

`fail (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:116)
fail (DBusException(dbus.String(u'Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.'),),) 
/usr/local/lib/python2.7/site-packages/gi/overrides/__init__.py:92: Warning: Source ID 3333 was not found when attempting to remove it
  return fn(*args, **kwargs)`

cschramm, 2015-04-06:
Please check the device info with `bluetoothctl` to see if blueman is getting the Paired and UUID attributes wrong.

juanitotc, 2015-04-06:
Hmm, I guess only audio source is available?

[bluetooth]# info E4:E0:C5:C3:A5:24
Device E4:E0:C5:C3:A5:24
        Name: DTVBluetooth
        Alias: DTVBluetooth
        Class: 0x08043c
        Icon: audio-card
        Paired: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)

cschramm, 2015-04-06:
That's what it looks like, but it also says that it is NOT paired. Are you sure blueman displays it as paired at the same time? Then blueman is getting something really wrong.

juanitotc, 2015-04-07:
In this case I hadn't paired with the device.

It's strange that the device announces itself as "video display and loudspeaker", yet only offers "audio source"...

BTW, as mentioned somewhere else, the pin request pop-up buttons are almost invisible unless I roll the mouse pointer over then - see attached.
[image]

cschramm, 2015-04-07:
So you see the same services when paired? Then blueman's behavior is correct. Maybe you need to enable some services on the TV explicitly for making them available?

> using flwm the pairing pin request pop-up buttons (confirm, cancel) are almost invisible unless I roll the mouse pointer over them

That is an unfortunate theming issue, but for convenience we could offer a build configuration option to force blueman to use its own GTK+ window for notifications with actions even if the notification daemon is capable of showing actions. If you're interested, please open a new issue for that. The implementation shouldn't be much more than two lines.

closed 2015-04-07 by cschramm