blueman-project/blueman #242: No obex functionality with iPhone

juanitotc, 2015-04-06:
After pairing with an iPhone, the audio sink, audio source and network access point services are shown, but the obex file transfer icon remains greyed out.

Shouldn't blueman be able to transfer files with obex to/from an iPhone?

cschramm, 2015-04-06:
Do you talk about the "Send a File" item in the device menu? There's no way from the blueman code that this is disabled.

juanitotc, 2015-04-06:
I'm speaking of the obex paper airplane "Send a File" icon in the main blueman-manager window.

cschramm, 2015-04-06:
Strange, can I get a screenshot of that?

juanitotc, 2015-04-06:
Please see attached.
[image]

cschramm, 2015-04-06:
Aah, _that_ one. :wink: 

I just checked and it looks like if that's the case the device does not offer an OPP or FTP service. You can check the UUIDs in bluetoothctl to see if that's correct.

juanitotc, 2015-04-07:
Indeed, I had just assumed that I'd be able to send files to an iPhone...

[bluetooth]# info 54:E4:3A:22:48:E9
Device 54:E4:3A:22:48:E9
...
        Class: 0x7a020c
        Icon: phone
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacafe)
        UUID: Service Discovery Serve.. (00001000-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
        UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: bluetooth:v004Cp6D03d0820

cschramm, 2015-04-07:
Probably not as iOS does not expose its file system to the user. Maybe there is some iOS app, though, you can start to provide a service for receiving files.

closed 2015-04-07 by cschramm