blueman-project/blueman #243: Blueman does not start correctly with flwm

juanitotc, 2015-04-06:
When starting blueman normally (double-clicking on an icon) in flwm, the bluez5 enable bluetooth dialogue box pops-up, but after clicking on "yes" the blueman window does not display.

In order to get the blueman window, I need to kill the blueman-manager pid and start blueman-manager from a terminal window. Killing the pid and double-clicking on the icon does not work.

juanitotc, 2015-04-06:
..forgot to mention that with bluez4 (no enable pop-up) things work fine.

cschramm, 2015-04-06:
Is it correct / on purpose that bluetooth is disabled?

If you start it in a terminal, do you get that question as well before getting to the manager window?

juanitotc, 2015-04-06:
If I start blueman-manager in a terminal I get the enable bluetooth pop-up and, after answering yes, I get the blueman-manager window.

If I start blueman-manager by double-clicking on the icon I get the enable bluetooth pop-up and, after answering yes, I do not get the blueman-manager window.

cschramm, 2015-04-06:
I see. That probably means blueman-applet fails to enable bluetooth. Please start blueman-applet in a terminal window, then start blueman-manager by double-clicking the icon and see what output blueman-applet produces when clicking "Yes".

jod35fan, 2015-04-08:
Blueman relies on underlayng software so if you want to have bluetooth on after reboot, do this:
 Create new file /etc/udev/rules.d/10-local.rules and set bluetooth power up and discoverable
ACTION=="add", KERNEL=="hci0", RUN+="/usr/bin/hciconfig hci0 up"
ACTION=="add", KERNEL=="hci0", RUN+="/usr/bin/hciconfig hci0 piscan"
Next time when booting you have bluetooth automatically on and in discovery mode. Blueman gets this status from bluez5 and will not popup anything.

cschramm, 2015-04-08:
...what does not solve the issue @juanitotc reported.

jod35fan, 2015-04-08:
...more helpful information to everyone, i just installed wheezy version of blueman and bluez4.99 to jessie 32-bit. Bluetooth service failled to start because d-bus service was not started. I needed to set Requires=network.target to bluetooth.service file that location can be found with systemctl status bluetooth.service.

This was a hard drive dualcore 3.0Ghz pc. In my 3.8Ghz quadcore ssd pc i also need to add
[Service]
Restart=on-failure
RestartSec=5

How about report to bluez team to 
add --compat flag to bluetooth.service file in bluez5.25-5.30 sources and  add --compat flag to bluetooth.service file in bluez5 sources and 
run configure -with-systemdsystemunitdir=/etc/systemd/system  --with-systemduserunitdir=/etc/systemd/system

edit sdpd-server.c to set more access rights:
chmod(SDP_UNIX_PATH, S_IRUSR | S_IWUSR | S_IRGRP | S_IWGRP | S_IROTH | S_IWOTH);

To make obexd work, add the following path to obexd/src/org.bluez.obex.service :
Exec=/usr/libexec/bluetooth/obexd

and cp org.bluez.obex.service /usr/share/dbus-1/services/

Maybe some year we will have working bluez5 blueman package.

cschramm, 2015-04-08:
@jod35fan Please stop spamming our issue tracker. I don't want to use delete.

infirit, 2015-04-11:
>Is it correct / on purpose that bluetooth is disabled

If I understand you correctly here then, yes in bluez5 this was an intentional change. The bluez folks believe this is up to userspace to power the adapters.

juanitotc, 2015-04-14:
After many attempts, I've only been able to reproduce this once and I forgot to start blueman-applet from a terminal :(

I'd be inclined to say this is some kind of flwm glitch and leave things for now - I'll get back if I manage to isolate the error.

closed 2015-04-14 by cschramm