blueman-project/blueman #245: Cannot see pairing pin notification buttoms

juanitotc, 2015-04-07:
Ref;

`...using flwm the pairing pin request pop-up buttons (confirm, deny) are almost invisible unless I roll the mouse pointer over them.`

and:

`That is an unfortunate theming issue, but for convenience we could offer a build configuration option to force blueman to use its own GTK+ window for notifications with actions even if the notification daemon is capable of showing actions. If you're interested, please open a new issue for that. The implementation shouldn't be much more than two lines.`

If you could do that, it would be much appreciated - as well as being able to see the buttons, it would avoid having notification-daemon as a dependency and having to start it with the X-session.

closed 2015-04-12 by cschramm

juanitotc, 2015-04-19:
Using bluez-4 and git master from today.

When pairing a gtk pop-up asks for confirmation of the pairing pin, but when then setting trust, a notification-daemon pop-up asks for confirmation of the trust status. Finally, when attempting to set a device as a network access point (udhcpc), there is neither gtk pop-up nor notification-daemon pop-up and blueman-manager segfaults.

Generate (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:170)
juanito_tc iPhone 
_________
unset_op (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:96)
op: regenerating instance <ManagerDeviceMenu object at 0xb5f8dbbc (blueman+gui+manager+ManagerDeviceMenu+ManagerDeviceMenu at 0x882f768)> 
_________
fail (/usr/local/lib/python2.7/site-packages/blueman/gui/manager/ManagerDeviceMenu.py:121)
fail (DBusException(dbus.String(u'Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.'),),) 

(blueman-manager:7638): Gtk-CRITICAL **: gtk_style_context_get_property: assertion 'priv->widget != NULL || priv->widget_path != NULL' failed
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:158: Warning: gtype.c:4221: type id '0' is invalid
  self.show()
/usr/local/lib/python2.7/site-packages/blueman/gui/MessageArea.py:158: Warning: can't peek value table for type '<invalid>' which is not currently referenced
  self.show()
Segmentation fault

infirit, 2015-04-19:
@juanitotc that is https://github.com/blueman-project/blueman/issues/42

There may be a workaround for this but let's continue the discussion there.

cschramm, 2015-04-19:
Not sure where that notification-daemon comes from. What's its exact content?

juanitotc, 2015-04-20:
After using blueman to pair with the iPhone, the iPhone still showed "not connected".

Tapping on "not connected" on the iPhone produced a blueman notification-daemon pop-up with three buttons - "always trust", "trust once" and "deny" (from memory, so that might not be exact).

After clicking on "always trust", the iPhone showed "connected" and blueman showed the trust icon superimposed on the iPhone icon.

At this point I tried to use blueman to set the iPhone as a "network access point" and blueman-manager segfaulted.

commit ecd80c4a (2015-04-12): Add configuration option to force notification fallback

Using gsettings the user can now choose not to use a notification-daemon at all. Previously the fallback GTK+ notifications were only used if we had to show actions, but the daemon did not report to support them.
Additional changes:

* Fix error message when libnotify cannot be initialized (it is no longer pynotify)
* Fix issue from GTK+ 3 transition in NotificationDialog (window vs. get_window())

Closes #245
Closes #246
PR #246 (2015-04-12): Add configuration option to force notification fallback
~~Using --disable-notification-daemon~~ the user can now choose not to use a notification-daemon at all. Previously the fallback GTK+ notifications were only used if we had to show actions, but the daemon did not report to support them.
Additional changes:

* Fix error message when libnotify cannot be initialized (it is no longer pynotify)
* ~~Fix blueman.Constants.POLKIT (was always false since 1954e2d3ffbc79a0cf696c7fe12bca431d34d368)~~
* Fix issue from GTK+ 3 transition in NotificationDialog (window vs. get_window())

Closes #245
Close [... cut]