blueman-project/blueman #247: Bluez5: Show known devices after reboot,  fix exception

jod35fan, 2015-04-08:
This appliest to blueman-master.zip donwloaded and compiled week ago.

With Bluez5.30 there no dbus method to find detailed info about paired devices after pc reboot. 
However Bluez5 stores detailed info to /var/lib/bluetooth/usbadapter/device/info and works without 
pairing again if configured so in /etc/bluetooth/main.conf (get it from Bluez5 sources).

So Blueman could save device address, name and class to for example user_home/.config/blueman.txt:

[
{"MAC": "12:34:56:78:22:25", "DevName": "i9377" , "Class": 524},
{"MAC": "12:34:56:78:99:22", "DevName": "M1", "Class": 524}
]

In DeviceList.py the following code will display this data in blueman-manager:
def DisplayKnownDevices(self, autoselect=False):
        self.clear()
        devices = self.Adapter.list_devices()
        for device in devices:
            self.device_add_event(Device(device))
        if devices == []:
             try:
                path = os.path.join(os.path.expanduser('~'), '.config/blueman.txt')
                with open(path) as data_file:    
                    devdata = json.load(data_file)
                data_file.close()
                for dev in devdata:
                    info = {}
                    info["Address"] = dev["MAC"]
                    info["Name"] = dev["DevName"]
                    info["Class"]= dev["Class"]
                    info["Alias"] = info["Name"]
                    info["Fake"] = True
                    info["RSSI"] = 50
                    d = FakeDevice(info)
                    self.device_add_event(d)
             except IOError:
               	pass
               
        if autoselect:
            self.selection.select_path(0)

FakeDevice.py needs also some modification to make this work:

class FakeDevice:
    def __init__(self, info):
        self.info = info
        self.Address = info["Address"]
        self.Fake = True
        self.UUIDs = []
        self.Class = info["Class"]
        self.SupportedTechnologies="BR/EDR";
        self.Trusted=True
        self.Paired=True
        self.Blocked=False
        self.Alias=info["Alias"]
        self.Temp = False
        self.Valid = False
        self.Icon = "blueman"

    def get_properties(self):
        return self.info

When pairing add new device to blueman.txt.

In SerialManager.py the following throws dbus.connection:Exception because org.bluez.Device1
interface does not exit anymore when disconnecting after file transfer. Exception can be disabled by replasing the code with dprint("disconnected").

def on_device_property_changed(self, key, value, path):
        if key == "Connected" and not value:
            d = Device(path)
            self.terminate_all_scripts(d.Address)

cschramm, 2015-04-08:
I see a propose for a solution here, but not the problem you are trying to fix.

> With Bluez5.30 there no dbus method to find detailed info about paired devices after pc reboot.

Please describe what you're referring to with this statement.

In general: Please use GitHub Markdown to make the post more readable, especially when it contains code.

jod35fan, 2015-04-08:
"but not the problem you are trying to fix."
Go to bluez5 sources/test and run list-devices.py without pairing and pairing after reboot. There is no api to access /var/lib/bluetooth/usbadapter/device if the remote device is not  paired after after  pc reboot. So the information could be stored locally because everything works withou pairing and empty blueman-manager is confusing and makes user pair again for no sensible reason. 

Now the blueman code is based to bluez4 and bluez5 causes changes to FakeDevice/BluezDevce logic.

#!/usr/bin/python

from __future__ import absolute_import, print_function, unicode_literals

import dbus

bus = dbus.SystemBus()

manager = dbus.Interface(bus.get_object("org.bluez", "/"),
					"org.freedesktop.DBus.ObjectManager")

def extract_objects(object_list):
	list = ""
	for object in object_list:
		val = str(object)
		list = list + val[val.rfind("/") + 1:] + " "
	return list

def extract_uuids(uuid_list):
	list = ""
	for uuid in uuid_list:
		if (uuid.endswith("-0000-1000-8000-00805f9b34fb")):
			if (uuid.startswith("0000")):
				val = "0x" + uuid[4:8]
			else:
				val = "0x" + uuid[0:8]
		else:
			val = str(uuid)
		list = list + val + " "
	return list

objects = manager.GetManagedObjects()

all_devices = (str(path) for path, interfaces in objects.iteritems() if
					"org.bluez.Device1" in interfaces.keys())

for path, interfaces in objects.iteritems():
	if "org.bluez.Adapter1" not in interfaces.keys():
		continue

	print("[ " + path + " ]")

	properties = interfaces["org.bluez.Adapter1"]
	for key in properties.keys():
		value = properties[key]
		if (key == "UUIDs"):
			list = extract_uuids(value)
			print("    %s = %s" % (key, list))
		else:
			print("    %s = %s" % (key, value))

	device_list = [d for d in all_devices if d.startswith(path + "/")]

	for dev_path in device_list:
		print("    [ " + dev_path + " ]")

		dev = objects[dev_path]
		properties = dev["org.bluez.Device1"]

		for key in properties.keys():
			value = properties[key]
			if (key == "UUIDs"):
				list = extract_uuids(value)
				print("        %s = %s" % (key, list))
			elif (key == "Class"):
				print("        %s = 0x%06x" % (key, value))
			elif (key == "Vendor"):
				print("        %s = 0x%04x" % (key, value))
			elif (key == "Product"):
				print("        %s = 0x%04x" % (key, value))
			elif (key == "Version"):
				print("        %s = 0x%04x" % (key, value))
			else:
				print("        %s = %s" % (key, value))

	print("")

cschramm, 2015-04-08:
I see paired but offline devices in both blueman-manager and bluetoothctl. You don't?

jod35fan, 2015-04-08:
"I see paired but offline devices in both blueman-manager and bluetoothctl. You don't?"

Do you see after reboot?

cschramm, 2015-04-08:
Yes.

jod35fan, 2015-04-08:
"Yes."

Because i installed from sources, I have no blueman directory in .gconf or .configure. Is this the reason?

If so, blueman should create Config data directory and not rely on setup.

cschramm, 2015-04-08:
If your paired devices are not visible in bluetoothctl after a reboot anymore, you're having some issues with BlueZ.

jod35fan, 2015-04-08:
[bluetooth]# list
Controller 00:15:83:3E:F0:66 kaveri [default]
[bluetooth]# devices
[bluetooth]# 

What bluez version you are using?

cschramm, 2015-04-08:
5.23. We're having users with all intermediary versions. I haven't seen a user of BlueZ 5.30 yet.

You can try downgrading or have a look at BlueZ' debug output to get a clue what's wrong with it.

jod35fan, 2015-04-08:
root@kaveri:/home/jessie/Downloads/bluez-5.25#  bluetoothctl
[NEW] Controller 00:15:83:3E:F0:66 kaveri [default]
[NEW] Device 12:34:56:78:9C:45 M1

Yep, the problem was in 5.30. I  compiled 5.25 and at least it sees all paired devices. Maybe now blueman works without my modifications. Anyway it was interesting to learn some python in 5 days;-)

5.30 is here:
http://www.bluez.org/

closed 2015-04-08 by jod35fan