blueman-project/blueman #249: Bluetooth disabled on system boot

andresclari, 2015-04-10:
After boot, Blueman requires "Turning Bluetooth On", there should be an option to do this automatically.

On Antergos (Arch) with MATE desktop, after installing Blueman, and having it work just fine, paired my keyboard a Logitech K810.

Not sure if I'm missing some deep config option, or this is something missing on Blueman itself.

PS: I do have the Bluetooth service enabled at boot, it's the actual dongle that doesn't seem to be turning on automatically.

cschramm, 2015-04-10:
In my opinion it's good that blueman does not enable bluetooth without asking. gnome-bluetooth, on the other side, does power the default adapter automatically.

I would like to get some more opinions on this.

-------

Debian's bluez package comes with the following udev rule for enabling new adapters automatically in `/lib/udev/rules.d/50-bluetooth-hci-auto-poweron.rules`:
````
ACTION=="add", SUBSYSTEM=="bluetooth", KERNEL=="hci[0-9]*", RUN+="/bin/hciconfig %k up"
````

Maybe the Arch maintainers should add that too?

In any case, you as a user, can create such a rule yourself.

andresclari, 2015-04-11:
The Udev rule solved the issue for me, but I still think it'd be helpful to have an option to do this automatically, although might be a bit iffy of a solution with how the different distros handle Udev rule locations, on Arch it was on "/etc/udev/rules.d". Also I had to install the package "bluez-utils" for the hciconfig command to work... So all of this would have to be automated in case you guys want to implement some automatic option for it.  
  
I added your solution to the wiki, so anyone else looking can fix that quickly while a solution is found for this.  

Probably should leave this open for more info or as a feature request for now.
Thanks.

infirit, 2015-04-11:
Not that iffy. The power manager plugin can be modified to handle this with a config option that powers up things on applet startup.

It will need to have a gsettings schema added, the options + gsettings details added to the plugin (see recentconns for reference) then finally update the on_load function to query gsettings whether to power up by default. One will need to check if self.CurrentState is False and call RequestPowerState(True).

I don't have the ability to do this right now but it's pretty simple I think.

commit 8049a9ed (2015-04-16): Implement auto power-on for adapters

In case adapters are not powered, we power them on automatically now. This can be disabled by a new PowerManager setting if not wanted.

Closes #249
Closes #250
closed 2015-04-16 by cschramm