blueman-project/blueman #251: Code quality according to pylint

infirit, 2015-04-13:
I ran pylint (no options) on the blueman and it came with a __very__ large list of things it is not happy about.

Some we can ignore like the gettext underscore but it does come up with some real problems.

Now, my question is, can we come up with some rules on code quality? For example, enforce the line lenght of 100 or enforce the requirement of docstrings?

cschramm, 2015-04-13:
I don't have much Python experience, so I don't know the tools, but love such things (see [my rails-audit ruby gem](https://github.com/cschramm/rails-audit)). I'm also working on some CI for blueman and will be happy to integrate them there. :smiley: 

> can we come up with some rules on code quality?

Of course!

> enforce the line lenght of 100

I'm generally using 120, but there are situations with long (interpolated) strings where I actually violate that due to no good solution. :unamused: 

> enforce the requirement of docstrings?

Not sure. I'm an advocate of the theory that extensive documentation leads to bad naming and overly complex code and that it's better to try to write "self-documenting code". Excessive documentation can also contribute to unreadable code and unmaintained documentation can even lead to confusion (left over comments, but the code changed significantly). Anyway, if an enforcement of docstrings is the consensus, I'm on.

Is there something like the [Ruby style guide](https://github.com/bbatsov/ruby-style-guide) for Python? I know there are PEPs on code style, but do they suffice as a "rule book"? What rules is pylint using (without options)? Are they from PEPs or some other document?

infirit, 2015-04-13:
>Is there something like the Ruby style guide for Python?

[PEP 8](https://www.python.org/dev/peps/pep-0008/) is the go to place for most of these things.

[Pylint](http://www.pylint.org/) is the standard code analysis tool and integrates well in several ide's. It checks many of the points included in PEP 8 but can be modified to check for 120 characters. It checks style and looks for bad practices in the code. Run it and you'll get a sense of it.

As for docstrings, it is not meant to be a replacement for proper documentation. It is a short description of what the function or class __does__. It is then used by tools like ide's to show in the popup when using a function. It also shows up when using the built-in help function. See [this](http://www.pythonforbeginners.com/basics/python-docstrings) for a short introduction and examples. Another use of docstrings is for test ([doctest](https://docs.python.org/3.3/library/doctest.html)). It is very useful and this is what ppl mean when they say Python is self documenting :smile_cat: 

In general, we should not be dogmatic about these things though. But there are things that are worrying, the cathall try/except blocks for example that will hide genuine bugs, pylint will warn about this.

I created a config file [1] for pylint with the line length set 120 and disabled check for docstring (C0111). The docstring as it causes lots of noise right now. When we come to a conclusion on them we can enable it again. Try it out and let me know what you think of it, ```pylint -f html blueman/```.

I think what is left is worth looking into.

[1] http://pastebin.com/vMWPRiQr

cschramm, 2015-04-13:
Quite a list... (as expected). Feel free to add a pylintrc to the repository.

infirit, 2015-04-13:
It looks intimidating but most of them are warnings. There are only a few errors which we will need to look at first. I pushed the file to master.

cschramm, 2015-05-15:
Unfortunately I'm failing to make the Gtk, Gdk, Glib, ... available via GIR on travis: https://travis-ci.org/blueman-project/blueman/jobs/62664843 :cry:

infirit, 2017-08-19:
Lots of these errors and warnings are fixed in #735

infirit, 2017-12-22:
See #810, we should close this once that is merged.

commit 32b6a7b2 (2015-04-12): Remove nautilus-sendto plugin

Current nautilus-sendto does not support plugins anymore. There does not seem a useful way for an integration anymore.

While caja-sendto still has the same plugin support, the current plugin does not work anymore after the GTK+ 3 migration. #220 #230

Closes #230
Closes #251
closed 2017-12-22 by cschramm