blueman-project/blueman #255: Device added successfuly, but failed to connet

garthmiller, 2015-04-13:
I'm using:
lubuntu 14
Blueman applet 1.23

When I try to connect a bluetooth wireless speaker the Bluetooth Assistant give me the following message: "Device added successfuly, but failed to connet".

The device is discovered with no problem.  I've got no clue what to try.

I ran the blueman-applet from a terminal window and this is the output:
Dell-Latitude-D600:~$ blueman-applet
Loading configuration plugins
_________
<module> (/usr/lib/python2.7/dist-packages/blueman/main/Config.py:20)
Skipping plugin Gconf
No module named gconf 
blueman-applet version 1.23 starting
Stale PID, overwriting
Using file config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['PPPSupport', 'DBusService', 'NMPANSupport', 'ShowConnected', 'RecentConns', 'StatusIcon', 'Menu', 'SerialManager', 'AppIndicator', 'DiscvManager', 'NMDUNSupport', 'Headset', 'NetUsage', 'AuthAgent', 'PowerManager', 'Networking', 'StandardItems', 'ExitItem', 'PulseAudio', 'TransferService', 'DhcpClient', 'KillSwitch'] 
Using file config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module NMPANSupport
No module named gconf 
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
No module named appindicator 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using file config backend
_________
load_nap_settings (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.RecentConns.RecentConns'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
Not loading PPPSupport because it's conflict has higher priority 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
Using file config backend
_________
update_menuitems (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:114)
warning: Adapter is None 
_________
update_state (/usr/lib/python2.7/dist-packages/blueman/main/KillSwitchNG.py:162)
State: True 
_________
on_switch_added (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:45)
killswitch registered 0 
_________
on_obex_owner_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/TransferService.py:70)
obex owner changed:  
_________
on_dbus_name_owner_change (/usr/bin/blueman-applet:81)
org.bluez owner changed to :1.1 
_________
set_nap (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:63)
set nap False 
_________
register_agent (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/AuthAgent.py:51)
Registering agent 
_________
__init__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:16)
caching initial properties 
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:136)
rebuilding menu 
_________
on_obex_owner_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/TransferService.py:70)
obex owner changed: :1.46 
Using file config backend
_________
enumerate_connections (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/ShowConnected.py:48)
Found 0 existing connections 
_________
on_started (/usr/lib/python2.7/dist-packages/blueman/main/applet/Transfer.py:72)
opp Started 
_________
on_started (/usr/lib/python2.7/dist-packages/blueman/main/applet/Transfer.py:72)
ftp Started 
_________
on_adapter_property_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:87)
prop UUIDs dbus.Array([dbus.String(u'00001000-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001001-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000112d-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001112-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000111f-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000111e-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001105-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1) 
_________
on_adapter_property_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:87)
prop UUIDs dbus.Array([dbus.String(u'00001000-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001001-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000112d-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001112-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000111f-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000111e-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110c-0000-1000-8000-00805f9b34fb'), dbus.String(u'0000110e-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001105-0000-1000-8000-00805f9b34fb'), dbus.String(u'00001106-0000-1000-8000-00805f9b34fb')], signature=dbus.Signature('s'), variant_level=1)

cschramm, 2015-04-14:
When running in the terminal window, please right click on the device and try to connect to the service so that we can see what fails in the output.

Shortcut: If you use 1.23-git201403102151-1ubuntu1 please upgrade to at least [1.23-git201406261335-deb-1ubuntu1](http://packages.ubuntu.com/utopic/blueman) as it has an important fix for compatibility with pulseaudio.

garthmiller, 2015-04-14:
Cschramm:

Thanks for looking at this.  I am using:

Here is the terminal output from blueman-applet when I tried to connect to the device.
After I started the applet I
1. Highlighted the device.
2. Clicked on: Device -> Setup.
3. The "A2DP Sink (Send Audio)" radio button already selected.
4. I clicked on "Forward"
5. Got the message "Device added successfuly, but failed to connect"
6. I clicked "Close"
7. I exited the application.

 blueman-applet
Loading configuration plugins
_________
<module> (/usr/lib/python2.7/dist-packages/blueman/main/Config.py:20)
Skipping plugin Gconf
No module named gconf 
blueman-applet version 1.23 starting
Using file config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['PPPSupport', 'DBusService', 'NMPANSupport', 'ShowConnected', 'RecentConns', 'StatusIcon', 'Menu', 'SerialManager', 'AppIndicator', 'DiscvManager', 'NMDUNSupport', 'Headset', 'NetUsage', 'AuthAgent', 'PowerManager', 'Networking', 'StandardItems', 'ExitItem', 'PulseAudio', 'TransferService', 'DhcpClient', 'KillSwitch'] 
Using file config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module NMPANSupport
No module named gconf 
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
No module named appindicator 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using file config backend
_________
load_nap_settings (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
[... 212 lines cut ...]
Generate (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py:215)
Memorex MW303 
_________
level_setup_event (/usr/lib/python2.7/dist-packages/blueman/gui/manager/ManagerDeviceList.py:347)
animating down 
_________
update (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:191)
stopping monitor (not connected) 
_________
update (/usr/lib/python2.7/dist-packages/blueman/gui/DeviceList.py:191)
stopping monitor (not connected) 
_________
save (/usr/lib/python2.7/dist-packages/blueman/plugins/config/File.py:117)
Saving config 
Exception AttributeError: "'NoneType' object has no attribute 'Bus'" in <bound method File.__del__ of <File object at 0xb5614d9c (blueman+plugins+ConfigPlugin+ConfigPlugin at 0xa287f40)>> ignored
_________
child_closed (/usr/lib/python2.7/dist-packages/blueman/Functions.py:153)
['/usr/bin/blueman-assistant', u'--device=FC:58:FA:1F:23:DC'] closed 
_________
save (/usr/lib/python2.7/dist-packages/blueman/plugins/config/File.py:117)
Saving config 
Exception AttributeError: "'NoneType' object has no attribute 'Bus'" in <bound method File.__del__ of <File object at 0xb57a4b44 (blueman+plugins+ConfigPlugin+ConfigPlugin at 0x9bd3b40)>> ignored
_________
child_closed (/usr/lib/python2.7/dist-packages/blueman/Functions.py:153)
/usr/bin/blueman-manager closed

cschramm, 2015-04-14:
Hrm. Strange. I cannot see the PulseAudio applet plugin in the output. Can you check if it's enabled (right click applet, then "Extensions")?

garthmiller, 2015-04-15:
The PulseAudio extension did not have a check next to it.  I checked it.  I rebooted and started the applet.  The PulseAudio extension is checked.  I ran the applet from the terminal again.  Here is the output.  Still the same error message.

Dell-Latitude-D600:~$ blueman-applet
Loading configuration plugins
_________
<module> (/usr/lib/python2.7/dist-packages/blueman/main/Config.py:20)
Skipping plugin Gconf
No module named gconf 
blueman-applet version 1.23 starting
Using file config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['PPPSupport', 'DBusService', 'NMPANSupport', 'ShowConnected', 'RecentConns', 'StatusIcon', 'Menu', 'SerialManager', 'AppIndicator', 'DiscvManager', 'NMDUNSupport', 'Headset', 'NetUsage', 'AuthAgent', 'PowerManager', 'Networking', 'StandardItems', 'ExitItem', 'PulseAudio', 'TransferService', 'DhcpClient', 'KillSwitch'] 
Using file config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module NMPANSupport
No module named gconf 
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
No module named appindicator 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
Using file config backend
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
Using file config backend
_________
load_nap_settings (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'> 
Using file config backend
_________
on_load (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DhcpClient.DhcpClient'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
__load_plugin (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
[... 170 lines cut ...]
dbus.ObjectPath('/org/bluez/778/hci0/dev_FC_58_FA_1F_23_DC') 
_________
save (/usr/lib/python2.7/dist-packages/blueman/plugins/config/File.py:117)
Saving config 
Exception AttributeError: "'NoneType' object has no attribute 'Bus'" in <bound method File.__del__ of <File object at 0xb5627d9c (blueman+plugins+ConfigPlugin+ConfigPlugin at 0x970df40)>> ignored
_________
child_closed (/usr/lib/python2.7/dist-packages/blueman/Functions.py:153)
['/usr/bin/blueman-assistant', u'--device=FC:58:FA:1F:23:DC'] closed 
_________
save (/usr/lib/python2.7/dist-packages/blueman/plugins/config/File.py:117)
Saving config 
Exception AttributeError: "'NoneType' object has no attribute 'Bus'" in <bound method File.__del__ of <File object at 0xb5768b44 (blueman+plugins+ConfigPlugin+ConfigPlugin at 0x88f5b40)>> ignored
_________
child_closed (/usr/lib/python2.7/dist-packages/blueman/Functions.py:153)
/usr/bin/blueman-manager closed 
^CTraceback (most recent call last):
  File "/usr/bin/blueman-applet", line 106, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 58, in __init__
    gtk.main()
KeyboardInterrupt
_________
save (/usr/lib/python2.7/dist-packages/blueman/plugins/config/File.py:117)
Saving config 
Exception AttributeError: "'NoneType' object has no attribute 'Bus'" in <bound method File.__del__ of <File object at 0xb575fc34 (blueman+plugins+ConfigPlugin+ConfigPlugin at 0x9ab7020)>> ignored

cschramm, 2015-04-15:
Can't see an issue there. Please try connecting the audio service directly in the device's menu, not via setup, but I don't expect a better result...

There could be some information from bluetoothd in your syslog (/var/log/daemon.log).

You can also try a more recent version of blueman. 1.99.alpha1 should be compatible with Ubuntu. You can get it from Debian jessie: https://packages.debian.org/jessie/blueman

garthmiller, 2015-04-15:
I installed version 1.99.alpha1 of blueman.  It still gives me the same error message.  I notice that there is no PulseAudio plugin listed when I check the plugins menu.  Would it help to see the terminal output again?

cschramm, 2015-04-15:
There is no such plugin in that version anymore as it is not needed. I think you're having some issue with your pulseaudio setup. Do you have the pulseaudio-module-bluetooth package?

garthmiller, 2015-04-15:
Dear Cschramm:

I didn't have the pulseaudio-module-bluetooth or even the pulseaudio module.  I installed them both and went back to version: 1.23-get201406261335-deb-1ubntu1.  Now it works.

(Maybe the packages were missing because I have lubuntu 14.  It is supposed to be a "light" version of ubuntu).

Thank you so much for your help!

cschramm, 2015-04-15:
:smiley:

closed 2015-04-15 by cschramm

garthmiller, 2015-05-12:
Dear Christopher Schramm:

You helped me with blueman-applet about a month ago (I had a problem with
pulse audio).  It was working fine, but now it stopped.  I think it was
after the software updater told me to install some security updates.

Below is the output from running the blueman-applet in a terminal window.
Any idea what i should do? (sorry to bother you again).

$ blueman-applet
Loading configuration plugins
blueman-applet version 1.23 starting
Stale PID, overwriting
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
['PPPSupport', 'DBusService', 'NMPANSupport', 'ShowConnected',
'RecentConns', 'StatusIcon', 'Menu', 'SerialManager', 'AppIndicator',
'DiscvManager', 'NMDUNSupport', 'Headset', 'NetUsage', 'AuthAgent',
'PowerManager', 'Networking', 'StandardItems', 'ExitItem', 'PulseAudio',
'TransferService', 'DhcpClient', 'KillSwitch']
Using gconf config backend
_________
Load (/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:54)
Unable to load plugin module AppIndicator
No module named appindicator
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
Using gconf config backend
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.DBusService.DBusService'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Headset.Headset'>
Using gconf config backend
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Networking.Networking'>
Using gconf config backend
_________
load_nap_settings
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/Networking.py:30)
Loading NAP settings
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.Menu.Menu'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
_________
__load_plugin
(/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py:127)
loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
Using gconf config backend
_________
on_load
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/KillSwitch.py:25)
Using the new killswitch system
[... 125 lines cut ...]
UpdatePowerState
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:156)
off True
foff False
on True
current state True
new state False
_________
UpdatePowerState
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:156)
Signalling False
_________
check
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:116)
callbacks done
_________
set_adapter_state
(/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:87)
False
^CTraceback (most recent call last):
  File "/usr/bin/blueman-applet", line 106, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 58, in __init__
    gtk.main()
KeyboardInterrupt

cschramm, 2015-05-12:
Again, I cannot see an issue in the log.

Can you check what changed on your system in your /var/log/apt/history.log?