blueman-project/blueman #262: GDBus

cschramm, 2015-04-15:
We should look into porting the dbus parts to glib's GDBus via introspection. This would drop the dbus-python requirement.

infirit, 2016-01-12:
I have been poking at this to better understand the Bluez dbus layer. I have an [example script](https://gist.github.com/infirit/d7750963235267cd3d41) that works and prints the changed properties.

A big confusion, for me at least, was that ``interface_name`` is not ``org.bluez.Adapter1`` but ``org.freedesktop.DBus.Properties``. Then when we connect to ``g-signal`` on the proxy we actually get the proper interface as part of the signal.

Also I understood that GDBusProxy and friends will handle watches so we do not have to add them directly through the connection like we do now. So a lot of the complexity of tracking signals for those watches can be removed.

I also saw it caches properties so we can drop that from ``blueman.main.Device`` :tada:.

I have been trying to subclass GDBusProxy but have yet to make this work. If I can make this work we probably can remove even more code.

infirit, 2016-01-12:
Update, was able to finally subclass, add a new signal, connect to the new signal with [this](https://gist.github.com/infirit/81031b2afb53fc183b24). Only problem now is that none of the signals are fired :crying_cat_face:. But other than that everything seems to work properly. I do get the signal to fire by constructing it the traditional way.. le sigh

infirit, 2016-01-15:
I just noticed the gdbus branch and re-based it against master in my fork (https://github.com/infirit/blueman/tree/gdbus)

No luck in the signal firing department yet.

infirit, 2016-01-17:
After some more experimenting with scripts, reading blueman's bluez dbus code and reading the GDbus docs I finally got fair understanding.

The main thing I like to add is that, from my understanding of gdbus docs, we should not be interacting with the bus directly. We should connect to the GDBusProxy [``g-properties-changed``](https://lazka.github.io/pgi-docs/#Gio-2.0/classes/DBusProxy.html#Gio.DBusProxy.signals.g_properties_changed) signal in our PropertiesBase class, loop over the changed and invalidated properties and emit our own ``property-changed`` signal.

I'll come up with a couple of example script that hopefully confirm this is indeed what we need to do.

infirit, 2016-01-18:
Here it is, an [example for an adapter](https://gist.github.com/infirit/56832c816bfab16c53c8). It replicates the current functionality without interacting directly with the bus. AFAICS this is all we need in the PropertiesBase class but let me know if I missed something.

infirit, 2016-01-21:
Ok, making progress but I do not understand something. What is the reason we have [_any_adapter](https://github.com/blueman-project/blueman/blob/master/apps/blueman-applet#L64) connected up in blueman-applet?

cschramm, 2016-01-22:
The plugin system has this method of triggering predefined methods the single plugins can define. What blueman-applet does is it triggers the on_adapter_property_changed method of applet plugins if any property of any adapter changed. This is used by the DiscvManager plugin to check if the DiscoverableTimeout, Discoverable, or Powered properties of the default adapter changed. It in fact looks a little awkward as DiscvManager could just connect to the adapter's property-changed signal itself. It was probably meant as some kind of abstraction which is now provided by our blueman.bluez classes.

infirit, 2016-01-22:
OK, thanks. There are similar cases for bluez.Device. The reason I ask is because I tried to get rid of subscribing directly to the dbus. And as these don't provide a object path things broke down.

I'll see if I can move this into the plugins.
-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

infirit, 2016-01-23:
Ok, here is my proposal to split the direct interaction with the dbus into new classes that handle the use case where we just want to intercept all Adapter1 or Device1 PropertiesChanged signals. It also simplifies the exiting classes as we do not have to worry about this anymore and just handle valid devices and adapters. The commit is https://github.com/infirit/blueman/commit/91306cd9433a671d8a7e2d296a11d8e1c5e7e756 and a quick review with or without rejection of the solution would be nice :smile:

cschramm, 2016-01-23:
Looks good, I'm just not sure if we really need the AnyAdapter. The only use case is now in PowerManager and I'm not 100 % sure if it makes sense there. We definitely need the AnyDevice though (to avoid subscribing to every single device that shows up).

infirit, 2016-01-24:
fyi and for future reference, ``set_cached_property`` __only__ updates the cache and __NOT__ the property on the bus object. So we still need separate properties interface to handle property updates.

infirit, 2016-02-02:
Looking at the NM plugins and we still support old apis in there, any objection if this gets dropped with the GDBus migration?

cschramm, 2016-02-03:
I'm fine with that. Just make sure it gets into the changelog.

infirit, 2016-05-20:
Browsing the internets how to register an object (org.blueman.Applet) I found that we are supposed to do something like nelow.

 * Create and introspection xml
 * Load it into a GDbusNodiInfo object
 * connect to the required bus
 * create functions to handle the methods, getting and setting properties
 * Register the object with ``connecting.register_object``

Totally untested and no idea yet what we need to handle in the function but it will be something like:
```python
from gi.repository import Gio
introspection_xml = '''
<node name='/org/blueman/Applet'>
  <interface name='org.blueman.Applet'>
    <method name='DisconnectDevice'>
      <args type='o' name='object_path' direction='in'/>
    </method>
    <method name='QueryPlugins'/>
  </interface>
</node>
'''

def dbus_method_cb(*args, **kwargs):
    # handle method call

def dbus_set_property_cb(*args, **kwargs):
    # handle setting property

def dbus_get_peoperty_cb(*args, **kwargs):
    # handle getting property

node_info = Gio.DBusNodeInfo.new_for_xml(introspection_xml)
connection = Gio.bus_get_sync(Gio.BusType.SESSION)

connection.register_object('/org/blueman/Applet',
                           node_info.interfaces[0],
                           dbus_method_cb,
                           dbus_get_property_cb,
                           dbus_set_property_cb)
````

infirit, 2016-07-09:
For reference, https://bugzilla.gnome.org/show_bug.cgi?id=656330#c16

It looks like it is worth waiting for this as it makes thinks much easier. It probably does mean we need to require a minimum version of pygobject so we may want to push the applet service port to 2.2

infirit, 2016-07-16:
What would you feel about using the module like pithos is doing? See [1].

The risk is bugs and some effort to move to whatever is merged.

[1] https://github.com/pithos/pithos/commit/517f0eba0d7eda997fbc4a10161554e06c5b6f5b

cschramm, 2016-07-22:
I guess that's fine. I don't see another way to move on with this. Do you?

infirit, 2016-07-25:
>Do you?

Not without creating our own decorators which would probably look a lot like what the pithos folks use now.

Very rough, barely working effort #560

infirit, 2016-08-06:
So here is the status

Client code: There are two applet plugins I can not port as I lack hardware for it. The plugins are Netusage (#565) and NMDUNSupport (#568) with the former having some trouble because of a dropped interface. I have one PR open for NMPANSupport ( #567) which needs a bit of testing by others and then merging.

Service code: It is not ready and will need some work. I added sender support and python2 compatibility to the DBusServiceObject but there is no async callbacks support at all yet on which we rely heavily in blueman. I am not familiar enough with this stuff to get this done for 2.1. What I have so far is in #560.

cschramm, 2019-05-05:
This is basically done with #1008. It would be nice to port some things to `org.freedesktop.DBus.Properties` but we at least do not want to block 2.1(.beta1) with that.

closed 2019-05-05 by cschramm