blueman-project/blueman #264: Add ability to start blueman-services from blueman-manager

juanitotc, 2015-04-20:
For window managers that do not have a panel, would it be possible to add a menu item to blueman-manager to be able to run blueman-services?

The main reason being to swap between networkmanager and udhcpc functionality to set a network access point without having to paste a long string into dconf-editor or similar.

cschramm, 2015-04-20:
blueman-services is a separate app. I guess you mean the plugins configuration of the applet?

juanitotc, 2015-04-20:
affirmative :)

cschramm, 2015-04-20:
Shouldn't be a problem. I'll look into it.

juanitotc, 2015-05-12:
The plugins branch works fine - thanks.

On a separate, but related, point - is it possible to start blueman-services from blueman-manager?

cschramm, 2015-05-12:
Added.

juanitotc, 2015-05-12:
thanks - that works too.

Only PAN with bluez-4 and udhcpc to go now :)

commit 9c1e27a8 (2015-05-12): Add plugins and local services buttons to View menu

Allows to open those dialogs without using the applet.

Closes #264
Closes #282
PR #282 (2015-05-12): Add plugins button to View menu
Allows to open plugins dialog without using the applet.

Closes #264
Closes #282
closed 2015-05-12 by cschramm