blueman-project/blueman #266: Python3: All connected devices are missing the signal bars

infirit, 2015-04-21:


cschramm, 2015-05-10:
Not sure what's meant here. Work for me. Python 3.4.

infirit, 2015-05-10:
I am talking about the orange, blue and green signal bars on the right side. They not there under 3.3 for me.

cschramm, 2015-05-10:
Yeap, they work for me with Python 3.4. Could you try with that version without much hassle? (Trying 3.3 is a little bit ugly on my side currently)

infirit, 2015-05-10:
Strange, I need this https://github.com/infirit/blueman/commit/798202d1c92bd7c1af1bedde213fddea5c7f8aef for it to work. I don't understand what changed in 3.4 to make this work for you, I'll try 3.4 later.

infirit, 2015-05-10:
Nope, still nothing with 3.4, I see the following in blueman-manager's output. And when I actually print the exception I see ``expected bytes, dbus.String found``.

```
_________
monitor_power_levels (/usr/lib64/python3.4/site-packages/blueman/gui/DeviceList.py:183)
Failed to get power levels 
```

cschramm, 2015-05-11:
Forget it. I was running blueman-applet with python 3, but blueman-manager started with my default python 2.7. I also see "Failed to get power levels" in 3.4. Can we add your fix?

infirit, 2015-05-11:
>Can we add your fix

Pushed it as b444452bddcdf3ea118cd2a00da5550e0b242130. Hopefully these are the last.

closed 2015-05-11 by infirit