blueman-project/blueman #267: blueman on xubuntu 14.04 (bluez4)

infirit, 2015-04-21:
Ok, got a xubuntu vm setup and blueman master has some issues on it. Just documenting here from within the vm. See the applet logs here: https://gist.github.com/infirit/b90c697143c6d0d66ee1

 * Out of the box NAP fails with NM and Blueman
 * Sending a file from the phone works.
 * Before pairing selecting the phone will error about services.
 * On startup and on several other occasions trying to invalidate NoneType object (fakedevice?)

infirit, 2015-04-21:
NAP is also broken with blueman 1.23 from the repos. But sending a file to the phone works.

cschramm, 2015-04-22:
From the log:

````python
Traceback (most recent call last):
File "/usr/lib/python2.7/dist-packages/blueman/main/PluginManager.py", line 212, in Run
ret = getattr(inst, function)(*args, **kwargs)
File "/usr/lib/python2.7/dist-packages/blueman/plugins/manager/Services.py", line 51, in on_request_menu_items
for service in device.get_services():
File "/usr/lib/python2.7/dist-packages/blueman/main/Device.py", line 68, in get_services
return [service for service in services if service]
File "/usr/lib/python2.7/dist-packages/blueman/main/Device.py", line 67, in <genexpr>
services = (self.get_service(uuid) for uuid in self.UUIDs)
File "/usr/lib/python2.7/dist-packages/blueman/main/Device.py", line 64, in get_service
return cls(self, uuid)
File "/usr/lib/python2.7/dist-packages/blueman/Service.py", line 25, in __init__
proxy = bus.get_object("org.bluez", self.__device.get_object_path())
File "/usr/lib/python2.7/dist-packages/dbus/bus.py", line 241, in get_object
follow_name_owner_changes=follow_name_owner_changes)
File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 244, in __init__
_dbus_bindings.validate_object_path(object_path)
TypeError: must be string, not None
````

I guess that's number 4. Obviously `self.__device.get_object_path()` is `None` in `Service.__init__`. Looking at `Device.get_object_path` that probably means it's a FakeDevice. I'll look into the history how this was handled before. Guess we just should not list any service for a FakeDevice.

````python
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:92)
invalidating device None
Exception AttributeError: "'NoneType' object has no attribute 'Signals'" in <bound method Device.__del__ of <Device object at 0x7f55100e8320 (blueman+main+Device+Device at 0x27eab00)>> ignored
````

That's your number 5. I've seen this before, but cannot think of a reason why this is happening. In `__del__` we call `self.Destroy()` and in `Destroy` we call `self.Valid = False; self.Signals.DisconnectAll()`. The error suggests that `self` is `None` at the last call, but I don't either know a reason for that nor can I imagine why it is not None when calling `Destroy` and setting `Valid` on it, but not afterwards...

Concerning number 1:

In the log I see NM picks up the connection, configures it, and fires up dhclient. dhclient failes saying that bnep0 would be down and then the connection gets disconnected. Unfortunately I cannot really see a reason there why that happens. Could you have a look at your kernel syslog if and at which points you see state changes of bnep0?

Number 2 is not an issue I guess. :wink: 

Concerning number 3:

Well, is it available? You're trying the obexd branch, right? In Ubuntu you will need the package obexd-server, but unfortunately in a higher version than available (>= 0.47). You can find it on snapshots.debian.org: http://snapshot.debian.org/package/obexd/ The version requirement also holds for obexd-client, so I'm a little bit surprised sending works four you. What do you have installed?

cschramm, 2015-04-22:
> I'll look into the history how this was handled before. Guess we just should not list any service for a FakeDevice.

Yes, that's it. Please try https://github.com/blueman-project/blueman/commit/e9ef91fa1a27f824a66e1878f2ad0f9d397b3e83.

infirit, 2015-04-22:
>Yes, that's it. Please try e9ef91f.

That takes care of 4 and 5. I do not see any None errors but.. now I see these  https://gist.github.com/infirit/21aa8e578d85d316a98a

>Concerning number 1

Focusing on NM for now, this is all I see in syslog, https://gist.github.com/infirit/0cff7e473cbe00e0b839

>Concerning number 3
>You're trying the obexd branch, right

I am using master not the obexd branch and I hopefully did not misunderstand that master still uses obex-data-server. Installing obexd server and removing obex-data-server makes no difference, it still warns about obexd.

Gnome is not working very well either and I can not get a network up so it may very well not be blueman. Ignore this for now until I can get it working in at least __one__ desktop vm :smile:. I'll see how the latest fedora (bluez5) works.

cschramm, 2015-04-22:
> I am using master not the obexd branch and I hopefully did not misunderstand that master still uses obex-data-server. Installing obexd server and removing obex-data-server makes no difference, it still warns about obexd.

Sorry, I got that wrong. Sending a file **to** the **phone** of course means blueman acts as a client.

The current state of the master branch is obexd >= 0.47 for sending (client) and obex-data-server for receiving (server) files. So to send to the phone you need the obexd-client package, but with the mentioned version difficulty on Ubuntu.

The obexd branch is meant to replace the rest of obex-data-server with the obexd-server part.

infirit, 2015-04-22:
le sigh, so here is the problem. I expected to get a notification on the phone that device X wanted access to the mobile network. This message does not show up in 99.9% of the time for me and saw it only twice in ~2 days worth of poking. I am not sure if this is something on the phone or on bluez/gnu+linux.

The good news is, that now that I enabled this unconditionally for __all__ devices things work under NM + bluez4 on xubuntu 14.04 and blueman :relieved:. I also tried this with NM bluez5 in fedora 21 but with gnome-bluetooth and his also worked.

infirit, 2015-04-22:
Good, on fedora with blueman I can connect to the phone's mobile network as well :relieved:.

>now I see these https://gist.github.com/infirit/21aa8e578d85d316a98a

They look like more FakeDevice but they only show up when I remove the device and appear to be  harmless otherwise.

infirit, 2015-04-22:
Ok, here is something interesting.

When I put the below in the ``__del__`` function I see the output many times except on delete where we get the error. What also surprised me is how often ``__del__`` is called but this may be my lack of knowledge of blueman :wink:.

```python
stack = inspect.stack()
stack_print = (stack[1][1], stack[1][2], stack[1][3])
print("** object: %s" % self)
print("** Fake %s" % self.Fake)
print("** Calling module: %s line: %s function: %s " % stack_print)
```

blueman-applet log:
```
_________
UpdatePowerState (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:159)
off False 
foff False 
on True 
current state True 
new state True 
_________
check (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:119)
callbacks done 
_________
set_adapter_state (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/PowerManager.py:90)
True 
blueman-manager version 1.99.alpha2 starting
Stale PID, overwriting
_________
on_adapter_property_changed (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Devices dbus.Array([], signature=dbus.Signature('o'), variant_level=1) 
_________
initialize (/usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py:144)
rebuilding menu 
** object: <Device object at 0x7fa21cf2cbe0 (blueman+main+Device+Device at 0x1f3a000)>
** Fake True
** Calling module: /usr/lib/python2.7/dist-packages/blueman/plugins/applet/RecentConns.py line: 345 function: get_device 
_________
__del__ (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:72)
deleting device None 
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:99)
invalidating device None 
Exception AttributeError: "'NoneType' object has no attribute 'Signals'" in <bound method Device.__del__ of <Device object at 0x7fa21cf2cbe0 (blueman+main+Device+Device at 0x1f3a000)>> ignored
_________
Destroy (/usr/lib/python2.7/dist-packages/blueman/main/Device.py:99)
invalidating device /org/bluez/3209/hci0/dev_6C_83_36_B9_C6_D6 
```

infirit, 2015-04-22:
I think I found the problem, in RecentConns we import __two__ Device classes. One from ``blueman.main.Device`` as ``Device`` and a second one from ``blueman.bluez.Device`` as ``BluezDevice``.

Device is only used once and when I change that line to BluezDevice the error goes away.

```diff
diff --git a/blueman/plugins/applet/RecentConns.py b/blueman/plugins/applet/RecentConns.py
index 3ab8597..8e8d1f4 100644
--- a/blueman/plugins/applet/RecentConns.py
+++ b/blueman/plugins/applet/RecentConns.py
@@ -340,7 +340,7 @@ class RecentConns(AppletPlugin, Gtk.Menu):
         except:
             raise AdapterNotFound
         try:
-            return Device(adapter.find_device(item["address"]))
+            return BluezDevice(adapter.find_device(item["address"]))
         except:
             raise DeviceNotFound
 ```

cschramm, 2015-04-23:
main.Device is an ugly wrapper for bluez.Device which I definitely want to take care of in #254.

In RecentConns a main.Device object is needed as it provides a signal for when the device has been removed. I am not sure why that change leaves you with new errors. I would not expect it to fix the issues and I would even expect more.

The underlying problem are disconnect signal handlers that try to get some property from dbus, but at that point the dbus object is already gone. You're seeing this at three different places: The services (AudioSource in your case), the SerialManager plugin and the GameControllerWakelock plugin. While the services code and GameControllerWakelock are new, this SerialManager code is more than five years old:

````python
    def on_device_property_changed(self, key, value, path):
        if key == "Connected" and not value:
            d = Device(path)
            self.terminate_all_scripts(d.Address)
````

I don't know how that may have worked and what changed...

I'd expect the same issue with BlueZ 5 as well, by the way. If you don't seem them there, that might be a hint how to improve it.

cschramm, 2015-04-23:
I pushed the FakeDevice services fix to master.

infirit, 2015-04-24:
>I am not sure why that change leaves you with new errors.

It does not, it leaves me with less but they are part of the same problem you are describing. More below :smile: 

>I'd expect the same issue with BlueZ 5 as well, by the way. If you don't seem them there, that might be a hint how to improve it.

Ah, I never looked at what happens with bluez5, it does show the same TraceBacks from dbus and the Device module. So it is not specific to bluez 4 which I was testing under 14.04.

I was trying to make sure bluez4 was working properly nd now that I see that the remaining problems are the same under bluez5 there is nothing left in this issue to take care of so closing.

closed 2015-04-24 by infirit