blueman-project/blueman #268: gvfs does not support obexftp anymore, at least bluez5

infirit, 2015-04-21:
https://bugzilla.gnome.org/show_bug.cgi?id=729945

The question is, should blueman still allow the browse command if none of the file managers that use gvfs will work? Dolphin with bluedevil is supposed to work but how many folks will use those with blueman?

cschramm, 2015-04-22:
They have bug report ([740882](https://bugzilla.gnome.org/show_bug.cgi?id=740882)) for porting to BlueZ 5, but it does not look like they will, so removing makes sense, yes, at least for blueman 2.1 where I'll remove BlueZ 4 support.

According to their NEWS file they removed the code in 1.23.92 which is pretty new (a month ago). I wonder if we should keep it for blueman 2.0, so that folks using BlueZ 4 and gvfs < 1.23.92 are still able to use it, but the question is how likely it is to see a distribution that still has BlueZ 4 and no recent gvfs. It looks like Ubuntu 15.04 will keep its BlueZ 4, but already has gvfs 1.24, so there's probably no audience for that...

infirit, 2015-04-22:
The only distro's out there that could use gvfs obexftp would be wheezy and ubuntu < 15.04 which is still a considerable part of the gnu+linux users. Although bluez4 is still in gentoo obexftp is disabled on gvfs and the rest of the distro's moved to bluez5 some time ago.

So how about this gets mentioned in the release anouncement/notes and dependancies/readme?

cschramm, 2015-04-26:
I've added that information to [our new wiki page on integrations](https://github.com/blueman-project/blueman/wiki/Integrations) I wanted to start for a while now. :smiley:

commit 44fda8a5 (2015-05-28): Drop browse feature

Closes #268
commit 73227f63 (2015-06-09): Drop browse feature

Closes #268
closed 2015-06-09 by cschramm