blueman-project/blueman #270: blueman can't find my earphones, ubuntu studio latest version

davem360, 2015-04-23:
it looks like every thing is all right but it can't find them, I already tried with two earphones. I'm new in linux hopefully you guys can help me with a graphic friendly solution.

infirit, 2015-04-24:
Normally one needs to put the earphones in a visible state that it can be seen by blueman. Only then will blueman be able to pair and connect to it. You do not say which ones you tried so there is very little we can do.

davem360, 2015-04-25:
Thank you very much for your quick reply. I've being using bt earphones for many years so I'm used to make them work in any os... I know some times can be tricky. This is the first time I can't and I'm pretty sure is the very striped down ubuntu Studio distro.  In a way I love the performance, but is not too friendly, even the audio output is not automatic as in SUSE, so I'm guessing is the same case for bluetooth audio devices. They used to work just fine with SUSE and windows 7 on that laptop with any of my BT devices (dell earphones). I would love to find a way to update it or reinstall some of it related sw components... I also have issues with Kdenlive because it crashes often, they sugest to update it, but ubuntu studio should be already updated and tested (it doesn't event let me update kdenlive)... Well that is why I chosed it, looking for stability and no extra updating and fixing work.  I would love to hear sugestions. It would be cool to try another distro for audio and video production. One for real ready to work, adobe is cool but I love linux, I would hate having to go back to windows or spending in a high performance Apple. Cheers!

fabrizo, 2015-05-20:
buon giorno da italia

cschramm, 2015-06-10:
Closing as no info was provided at all.

closed 2015-06-10 by cschramm